zigpy/zigpy-znp #1: [SUGGESTION] Team up with Electrolama to beta-test the "Zig-a-zig-ah!" CC2652 adapter

Hedda, 2020-03-09:
@Adminiuga & @puddly Can I suggest that you try to reach out to Omer Kilic a.k.a. @omerk at Electrolama about teaming up to beta-test his "Zig-a-zig-ah!" CC2652 adapter together with zigpy-znp for ZHA in Home Assistant?  

https://electrolama.com/projects/zig-a-zig-ah/

https://github.com/electrolama/zig-a-zig-ah

As I understand both zig-a-zig-ah and zigpy-znp are only spare-time hobby projects but still I think that teaming up might give both projects some positive recognition in the Home Assistant community which might attract more developers and willing alpha/beta-testers from the community to come help out with both these projects which in the long run could benefit all users wanting ZHA to become more mature and working with more hardware than today.

https://www.home-assistant.io/components/zha/

Would also be nice to see collaboration with the guys working on zigpy-cc as an alternative

https://github.com/sanyatuning/zigpy-cc/

Hedda, 2020-03-11:
@Adminiuga Checkout related https://github.com/electrolama/zig-a-zig-ah/issues/5 where @omerk of electrolama.com express a willingness to collaborate and could provide some zig-a-zig-ah adapter prototypes to a few zigpy and ZHA developers to help the development of TI CC2652 support. You can send him an e-mail if you are interested.

Hedda, 2020-03-23:
@Adminiuga & @puddly not sure if you guys already sent Omer a mail regarding getting a zig-a-zig-ah (TI CC2652 USB-stick) prototype, but FYI, Koenkk from developed zigbee-shepherd just wrote that he has received a zig-a-zig-ah prototype from Omer and it seems to work great so far with zigbee2mqtt. Read his comments about it here: https://github.com/Koenkk/zigbee2mqtt/issues/1429#issuecomment-602485214

Adminiuga, 2020-03-23:
The board definitely looks nice, no doubts about it, but ATM the problem is not in hardware, but a simple lack of time on my part.