zigpy/zigpy-znp #3: Tag release versions and make PyPi release(s) of zigpy-znp?

Hedda, 2020-03-09:
A friendly reminder to tag an initial release version of zigpy-znp and make PyPi release of zigpy-znp

@Adminiuga I guess that you might not yet be ready to make your first release of this zigpy-znp library but I just wanted to add a friendly reminder here under issues for tracking purposes only, so there is no rush.

An initial PyPi release version tag (like for example v0.0.1) and then continues versions/releases of newer version if and when code is added or changed will make it much easier for projects like Home Assistant and Linux OS distributions like Hass.io and Raspbian or others to update and make use of each new versions of this library.

Version tagging and PyPi release(s) is something that is already available for the zigpy library as well as similar radio libraries that it depends on such as; bellows, zigpy-deconz, zigpy-xbee, zigpy-zigate, and zigpy-cc. My thinking is that the more people are testing this early on then more developers will eventually also get interested and will be able to help you out with the development of zigpy-znp

https://github.com/zigpy/zigpy/releases
https://github.com/zigpy/bellows/releases
https://github.com/zigpy/zigpy-xbee/releases
https://github.com/zigpy/zigpy-deconz/releases
https://github.com/doudz/zigpy-zigate/releases
https://github.com/sanyatuning/zigpy-cc/releases

https://pypi.org/project/zigpy
https://pypi.org/project/bellows
https://pypi.org/project/zigpy-xbee
https://pypi.org/project/zigpy-deconz
https://pypi.org/project/zigpy-zigate
https://pypi.org/project/zigpy-cc

Off-topic but have you considerd maybe move this zigpy-znp repository to a GitHub repo under the zigpy organization @ https://github.com/zigpy/ so that more than one person has access to all your zigpy related projects for collaboration as well as having a greater chance of getting noticed by new developers who might be willing to help out?

Hedda, 2020-03-31:
FYI, I have submitted PR #8 for updated README.md file and also copied the section for PyPI links but did not add a link as a project zigpy-znp does not yet look to be reserved by @Adminiuga or @puddly ?

# Releases via PyPI

Tagged versions will also be released via PyPI

- https://pypi.org/project/zigpy-znp/
    - https://pypi.org/project/zigpy-znp/#history
    - https://pypi.org/project/zigpy-znp/#files

puddly, 2020-03-31:
One is now up.

closed 2020-03-31 by Adminiuga