zigpy/zigpy-znp #4: [REQUEST] Automagically auto-detect port-path and discover radio-type?

Hedda, 2020-03-09:
This is a feature request that I'm not exactly sure if it belongs to the Home Assistant core, the ZHA integration for Home Assistant, to zigpy, to each and every radio library for zigpy, or in many of all those mentioned in order to be achieved. Anyway, my main point with this feature request is this:

Is it possible to add some kind of "auto-detect" scanning option for automatic discovery/detection of port-path and radio-type (with recommended select) to achieve an even more plug-and-play experience of the initial install/configuration of the ZHA integration in Home Assistant or a zigpy implementation in any other software for that matter? An option for a so-called "next next next" installation process if you will.

From an end-users point-of-view the ZHA integration implementation of zigpy I think that it would be absolutely awesome if the installation process could automagically detect exactly which type of Zigbee adapter(s) you have plugged in and which USB-port(s) that the Zigbee adapter(s) is plugged into. Is this possible?

I think nicest would probably be if the user would be greeted by an option for "auto-detect and recommend Zigbee adapter(s)" that you could then simply click for automagic plug-and-play configuration of your Zigbee stick hardware instead of having to manually type the full port-path and the radio-type as you have to do today in ZHA.

Such an automagical auto-detect port-path and discover radio-type would make the ZHA integration component in Home Assistant even easier to install and configure for first-time ZHA users by removing the need for end-user to manually configure port-path and radio-type.

I believe that a plug-and-play oriented hardware configuration could be the last piece that is missing to make the ZHA  in Home Assistant a much more user-friendly initial installation and configuration process experience.

Hedda, 2020-03-20:
If I understand it correctly that both https://github.com/zigpy/zigpy-cc and https://github.com/doudz/zigpy-zigate have now implemented automatic detection of ort-path but not necessarily used the same code for their solutions?

puddly, 2020-03-30:
Somewhat-automatic port detection exists but I do not have enough devices to thoroughly test it:

https://github.com/zha-ng/zigpy-znp/blob/805d43014b753f82dea4aae28aece768f3abcafa/zigpy_znp/uart.py#L135-L158

puddly, 2020-04-04:
The `probe` method was implemented and tested in 102745eadce1167c4a3f390da60c0ecb6e7f9fd0, which will tell you if a serial port is for a usable radio. Radio auto detection, if implemented, probably belongs in the Home Assistant ZHA component.

closed 2020-04-04 by puddly