zigpy/zigpy-znp #5: [REQUEST] Capability to handle serial-port disconnects and reconnects

Hedda, 2020-03-09:
It is possible to implement some type of handling of serial port disconnects and reconnects?

I think that you can test this by unplugging the Zigbee coordinator adapter and then plug it back in after a minute to two.

Point is that zigpy-znp as well as all other radio libraries for zigpy should somehow always be able to reconnect and recover connection after reconnecting.

My main reason for that is when using ser2net and socat to use serial device over the network.

Problem is if the network temporarily goes down then it can not reconnect to recover the connection. 

This is a relatively common networked setup if don't want the radio close to Home Assistant, see: 

https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

A real-world scenario would be if you have Home Assistant and running on a virtual machine on your NAS with UPS-backup-power-supply but have your Zigbee coordinator adapter installed somewhere else in your house on a Raspberry Pi without UPS-backup-power-supply then a power-outage will cause the Raspberry Pi to reboot and maybe also any network switch that you have in between.

I think that preferable it should be able to try to recover at least once every 60-seconds and that recovery should still be possible after as much as 10-minutes (or more) because that is reasonably how long a reboot can take after a short power outage.

Another problem scenario is if the user accidentally unplugs the Zigbee USB adapter when running, as seen in related issue: 

https://github.com/sanyatuning/zigpy-cc/issues/15

References:

https://community.home-assistant.io/t/add-network-protocol-support-to-zigbee-and-zwave-integrations/136056

https://github.com/Danielhiversen/pyRFXtrx/pull/85

https://github.com/zigpy/bellows/issues/183

https://github.com/zigpy/zigpy/issues/297

Hedda, 2020-03-20:
If I understand it correctly that both https://github.com/zigpy/zigpy-cc (and perhaps also  https://github.com/doudz/zigpy-zigate ) have now implemented automatic handling of serial-port disconnects and reconnects but not necessarily used the same code for their solutions?

puddly, 2020-04-04:
Working and tested as of c81fdab20603c44dc4be9ce53afcab9cd14bf8e8.

closed 2020-04-04 by puddly