zigpy/zigpy-znp #6: [SUGGESTION] Migrate zigpy-znp repository to upstream zigpy organization?

Hedda, 2020-03-09:
@Adminiuga can I suggest that you please consider having your zigpy-znp repository here on GitHub moved to the zigpy organization @ https://github.com/zigpy/ so that the mainstream copy of this repo will stay there as its upstream instead and owned by the upstream zigpy organization? 

That way all developers that are members in the zigpy organization can have write access to one or more projects that it own for collaboration as a community? 

This would also help new developers take notice of this zigpy-znp project.

Hope that you could all come to an agreement that will make everyone happy with continuing with working on the zigpy-znp repository together without getting offended by any ownership conflicts.

Adminiuga, 2020-03-09:
Unfortunately I do not have access to zigpy organization on GitHub. I'd like to move all projects under one "roof", but currently don't have the privileges required to do so.

Hedda, 2020-03-10:
Can maybe GitHub support can help if no one can contact @rcloran about this so that he can give more people has administrator privileges to the zigpy organization @ https://github.com/zigpy/ or?

Hedda, 2020-03-20:
Time for zigpy-znp repo migration to zigpy organization now that you are co-owner/co-admin?

Move repo from https://github.com/zha-ng/zigpy-znp to https://github.com/zigpy/zigpy-znp

Adminiuga, 2020-03-20:
zigpy-znp is in early stages and is not even working, so it is not ready to be publicly released. I pushed it there just to "capture" the name, but other than that there's a lot of work to be done.

Neustradamus, 2020-04-01:
@Adminiuga: It is possible to see for the transfer?

puddly, 2020-04-01:
@Neustradamus: I only started using this module a few days ago with my CC26X2R1. With a single tester, I don't think zigpy-znp is ready for anything just yet 😄. Maybe once I bring code coverage closer to 100% and get it running smoothly with my network.

Neustradamus, 2020-04-01:
You can move without problems, redirections will be here...

Hedda, 2020-04-01:
@Neustradamus I believe that the issue is probably that for now, they would prefer the stealth approach so that they don't expose zigpy-znp to discovery by normal-users when it not ready for general testing, ...and that way they also avoid getting requests from annoyed people like myself ;)

IMHO once ready for testing by advanced-users, it would benefit to move the project as it has then hopefully the possibility to not only attract advanced-users but also more developers that can help :)

Adminiuga, 2020-04-01:
Yep, I don't have issues moving this to zigpy in near the future once it is more or less ready for general use.

Hedda, 2020-09-07:
@Adminiuga & @puddly I just noticed that zigpy-znp (v0.1.1) will be supported soon by the upcoming Home Assistant so will that library be considered to be included into zigpy organization and as such have its repository moved to https://github.com/zigpy or?

https://github.com/home-assistant/core/pull/39700

https://github.com/zha-ng/zigpy-znp

PS: @puddly & @Adminiuga Congratulations on the milestone of adding zigpy-znp support to Home Assistant ZHA integration!

basnijholt, 2020-10-26:
Perhaps it makes sense to move this repo now?

puddly, 2020-10-28:
I've created a PR for this https://github.com/zha-ng/zigpy-znp/pull/45. Hopefully Travis CI and Coveralls will accept the new URLs without any problems.

Will a redirect be created from https://github.com/zha-ng/zigpy-znp/ to https://github.com/zigpy/zigpy-znp/ ?

Adminiuga, 2020-10-28:
yes, GH is really good with redirects. So old URL would lead to new location anyway

closed 2020-10-28 by puddly

Neustradamus, 2020-10-28:
Thanks all for your changes :)

Maybe, you can see for other repositories too?
- https://github.com/zha-ng

You can show your profile in public (by default it is not):
- https://github.com/orgs/zigpy/people

Hedda, 2020-11-02:
> You can show your profile in public (by default it is not):
> 
> * https://github.com/orgs/zigpy/people

Yes no people are currently listed zigpy org https://github.com/orgs/zigpy/people but are in https://github.com/orgs/zha-ng/people