zigpy/zigpy-znp #11: [REQUEST] Allow setting TX power on modules with built-in Power Amplifier (PA/LNA) frontend

Hedda, 2020-04-14:
Please consider allowing setting TX power on modules with built-in Power Amplifier (PA/LNA).

Backstory: Some TI CC modules feature a built-in Power Amplifier (PA/LNA) / powered RF-amp frontend, and the power that goes to that RF-amplifier should be controllable via software. For example; the default TX power for the CC1352P and CC2652P is 5 dBm, however, these chips support up to a max of 20 dBm, if you allow the user to control this then they could potentially get better reception if the power to amplification is increased. 

See https://github.com/Koenkk/zigbee2mqtt/issues/2253 from Koenkk of Zigbee2mqtt which the same idea and already implemented it as a proof-of-concept to allows setting the transmit power of the adapter for future versions which works for all adapters with Power Amplifiers, including CC2531, CC2530, CC1352P-2 and CC26X2R1, (in his POC allowed values are -22 dBm as the min to 19 dBm as the max, default is 0 except for the CC2530 + CC2591 = 4 and CC2530 + CC2592 = 19).

TI CC modules/chips which features an integrated or external RF-amplifier (PA/LNA) as I understand:
- CC1352P (integrated Power Amplifier), example: LAUNCHXL-CC1352P
- CC2652P (integrated Power Amplifier) 
- CC2530 + CC2591 (where CC2591 is an external Power Amplifier chip)
- CC2530 + CC2592 (where CC2592 is an external Power Amplifier chip)
- CC2531 (integrated amplifier?)
- CC2538 + CC2591 (where CC2591 is an external Power Amplifier chip)
- CC2538 + CC2592 (where CC2592 is an external Power Amplifier chip)

PS: Depending on the frequency band, some dBm values may also not be legal outdoors and/or indoors depending on your country regulations.

Hedda, 2020-04-14:
In the course of my further research I came across an interesting hint within the TI Forum, where an "undocumented" interface of the ZNP is pointed out:

https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/p/443496/1593040#1593040

If you use ZNP, you can use the undocumented MT_SYS_STACK_TUNE command to the ZNP with the data for setting the Tx transmit power. It is an SREQ with 2 bytes of data, so

02 21 0F 00 XX where:

02 = 2 bytes of data

21 = SREQ + MT_RPC_SYS_SYS

0F = MT_SYS_STACK_TUNE

00 = STK_TX_PWR

XX = a valid power setting from ZMAC.h:

typedef enum
{
TX_PWR_MINUS_22 = -22,
TX_PWR_MINUS_21,
TX_PWR_MINUS_20,
TX_PWR_MINUS_19,
TX_PWR_MINUS_18,
TX_PWR_MINUS_17,
TX_PWR_MINUS_16,
TX_PWR_MINUS_15,
TX_PWR_MINUS_14,
TX_PWR_MINUS_13,
TX_PWR_MINUS_12,
TX_PWR_MINUS_11,
TX_PWR_MINUS_10,
TX_PWR_MINUS_9,
TX_PWR_MINUS_8,
TX_PWR_MINUS_7,
TX_PWR_MINUS_6,
TX_PWR_MINUS_5,
TX_PWR_MINUS_4,
TX_PWR_MINUS_3,
TX_PWR_MINUS_2,
TX_PWR_MINUS_1,
TX_PWR_ZERO,
TX_PWR_PLUS_1,
TX_PWR_PLUS_2,
TX_PWR_PLUS_3,
TX_PWR_PLUS_4,
TX_PWR_PLUS_5,
TX_PWR_PLUS_6,
TX_PWR_PLUS_7,
TX_PWR_PLUS_8,
TX_PWR_PLUS_9,
TX_PWR_PLUS_10,
TX_PWR_PLUS_11,
TX_PWR_PLUS_12,
TX_PWR_PLUS_13,
TX_PWR_PLUS_14,
TX_PWR_PLUS_15,
TX_PWR_PLUS_16,
TX_PWR_PLUS_17,
TX_PWR_PLUS_18,
TX_PWR_PLUS_19
} ZMacTransmitPower_t; // The transmit power in units of -1 dBm.

puddly, 2020-04-14:
Implemented in 3f6968d7af53e15d893d92240136bf93ef42f7ed. It'd be best to expose this as a startup configuration option, though.

closed 2020-04-14 by puddly

puddly, 2020-04-16:
@Adminiuga thanks. Is there a reason for why the device schema is passed to both the API and the UART? There are a few configuration options that the API accesses that would essentially have to be passed from the Application to the API to get around the schema structure, which I'm "fixing" by just extending the device schema: https://github.com/zha-ng/zigpy-znp/blob/a6be2f8b2a801ee61df7871f467828d419bb9424/zigpy_znp/config.py#L25-L41

Also, will this schema be replacing the `form_network` method's arguments as well?

Adminiuga, 2020-04-16:
let's follow up in https://github.com/zigpy/zigpy/issues/381

puddly, 2020-05-13:
@Hedda if you're using zigpy-znp, you can change the TX power like this:

```yaml
zigpy:
  device:
    path: '/dev/serial/by-id/...'

  zigpy_znp:
    tx_power: 19
```

This needs to be documented in the README eventually.

Adminiuga, 2020-05-14:
This was changed slightly but gist remained the same. The new configuration for ZHA in `configuration.yaml` is

```yaml
zha:
  zigpy_config:
    znp_config:
      tx_power: 19
```

Essentially `zigpy_config:` section of `zha:`  is passed to ControllerApplication.new

Hedda, 2020-05-14:
Submitted PR for README.md update https://github.com/zha-ng/zigpy-znp/pull/21

(I do not own a CC1352 or CC2652 based adapter/board yet so I am not using zigpy-znp myself).