zigpy/zigpy-znp #15: Status.unknown_0xE9

Adminiuga, 2020-05-10:
Getting in the logs:
```
020-05-10 14:45:32 WARNING (MainThread) [zigpy_znp.types.named] Unhandled Status value: Status.unknown_0xE9
```

Latest `zigpy` version include NWK Status and MAC Statuses, so it would be nice to "chain" those here, as 0xE9 corresponds to [MAC_NO_ACK](https://github.com/zigpy/zigpy/blob/9a96093f89237efbffee2bb8c28a11f12f2a161d/zigpy/types/named.py#L323)

Similar feature in `zigpy_deconz` was implemented in https://github.com/zigpy/zigpy-deconz/pull/109

puddly, 2020-05-13:
Z-Stack has an enum identical to the one in zigpy so this is totally doable. The problem here is that I have no idea what the true status "type" actually is. All of the documented Z-Stack commands just have a `Status` response that could be one of a couple different types with `0x00` indicating success.

Do you happen to have the context for this status value? This warning is logged during the creation of a command instance.

Adminiuga, 2020-05-13:
Yep, same with deCONZ and I see it from time to time in ZCL and ZDO (zdo more often) replies too. 
Each have their own Status with well defined codes, but often they will pass the status from the lower levels APS, NWK or MAC being the lowest.  This is often particular implementation specific.

puddly, 2020-05-16:
I'll revisit this in the future. For now the simplest approach is to just extend the `Status` enum with all the possible values.

commit 5cdcafef (2020-05-16): Add all known Z-Stack status values (fixes #15)
closed 2020-05-16 by puddly