zigpy/zigpy-znp #16: After permitting new joins for 60s, network remains opened

Adminiuga, 2020-05-11:
Here's log. Removing the device, but it immediately joins back:
```
2020-05-10 22:06:17 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.LeaveInd.Callback(NWK=0xaa8f, IEEE=7c:b0:3e:aa:00:a5:e4:7f, Request=<Bool.false: 0>, Remove=<Bool.true: 1>, Rejoin=<Bool.false: 0>)
2020-05-10 22:06:17 INFO (MainThread) [zigpy_znp.zigbee.application] ZDO device left: ZDO.LeaveInd.Callback(NWK=0xaa8f, IEEE=7c:b0:3e:aa:00:a5:e4:7f, Request=<Bool.false: 0>, Remove=<Bool.true: 1>, Rejoin=<Bool.false: 0>)
2020-05-10 22:06:17 INFO (MainThread) [zigpy.application] Device 0xaa8f (7c:b0:3e:aa:00:a5:e4:7f) left the network
2020-05-10 22:06:20 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.TCDevInd.Callback(SrcNwk=0x66d6, SrcIEEE=7c:b0:3e:aa:00:a5:e4:7f, ParentNwk=0x0000)
2020-05-10 22:06:20 INFO (MainThread) [zigpy_znp.zigbee.application] ZDO device join: ZDO.TCDevInd.Callback(SrcNwk=0x66d6, SrcIEEE=7c:b0:3e:aa:00:a5:e4:7f, ParentNwk=0x0000)
2020-05-10 22:06:20 INFO (MainThread) [zigpy.application] Device 0x66d6 (7c:b0:3e:aa:00:a5:e4:7f) joined the network
2020-05-10 22:06:20 DEBUG (MainThread) [zigpy.application] Device 7c:b0:3e:aa:00:a5:e4:7f changed id (0xaa8f => 0x66d6)
2020-05-10 22:06:20 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy/device.py", line 77, in group_membership_scan
    await ep.group_membership_scan()
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy/endpoint.py", line 149, in group_membership_scan
    res = await self.groups.get_membership([])
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy/device.py", line 188, in request
    use_ieee=use_ieee,
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy_znp/zigbee/application.py", line 608, in request
    data=data,
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy_znp/zigbee/application.py", line 566, in _send_request
    partial=True, Endpoint=dst_ep, TSN=sequence
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy_znp/api.py", line 333, in request_callback_rsp
    await response
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy_znp/api.py", line 317, in request
    f"Expected SRSP response {partial_response}, got {response}", response
zigpy_znp.exceptions.InvalidCommandResponse: Expected SRSP response AF.DataRequestExt.Rsp(Status=<Status.Success: 0>), got AF.DataRequestExt.Rsp(Status=<Status.InvalidParameter: 2>)
2020-05-10 22:06:20 DEBUG (MainThread) [zigpy.device] Canceling old initialize call
2020-05-10 22:06:20 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x66d6), DstEndpoint=3, DstPanId=0x0000, SrcEndpoint=3, ClusterId=4, TSN=72, Options=<TransmitOptions.RouteDiscovery: 32>, Radius=30, Data=b'\x01\x48\x02\x00')
2020-05-10 22:06:20 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x66d6), DstEndpoint=3, DstPanId=0x0000, SrcEndpoint=3, ClusterId=0, TSN=73, Options=<TransmitOptions.RouteDiscovery: 32>, Radius=30, Data=b'\x00\x49\x00\x04\x00\x05\x00')
2020-05-10 22:06:20 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.InvalidParameter: 2>)
2020-05-10 22:06:20 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.InvalidParameter: 2>)
2020-05-10 22:06:20 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.EndDeviceAnnceInd.Callback(Src=0x66d6, NWK=0x66d6, IEEE=7c:b0:3e:aa:00:a5:e4:7f, Capabilities=<MACCapabilities.AllocateShortAddrDuringAssocNeeded|RXWhenIdle|MainsPowered|Router: 142>)
2020-05-10 22:06:20 INFO (MainThread) [zigpy_znp.zigbee.application] ZDO device announce: ZDO.EndDeviceAnnceInd.Callback(Src=0x66d6, NWK=0x66d6, IEEE=7c:b0:3e:aa:00:a5:e4:7f, Capabilities=<MACCapabilities.AllocateShortAddrDuringAssocNeeded|RXWhenIdle|MainsPowered|Router: 142>)
2020-05-10 22:06:20 INFO (MainThread) [zigpy.application] Device 0x66d6 (7c:b0:3e:aa:00:a5:e4:7f) joined the network
2020-05-10 22:06:20 DEBUG (MainThread) [zigpy.application] Skip initialization for existing device 7c:b0:3e:aa:00:a5:e4:7f
2020-05-10 22:06:20 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy/device.py", line 77, in group_membership_scan
    await ep.group_membership_scan()
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy/endpoint.py", line 149, in group_membership_scan
    res = await self.groups.get_membership([])
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy/device.py", line 188, in request
    use_ieee=use_ieee,
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy_znp/zigbee/application.py", line 608, in request
    data=data,
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy_znp/zigbee/application.py", line 566, in _send_request
    partial=True, Endpoint=dst_ep, TSN=sequence
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy_znp/api.py", line 333, in request_callback_rsp
    await response
  File "/home/lex/.venv/hass-zigpy/lib/python3.7/site-packages/zigpy_znp/api.py", line 317, in request
    f"Expected SRSP response {partial_response}, got {response}", response
zigpy_znp.exceptions.InvalidCommandResponse: Expected SRSP response AF.DataRequestExt.Rsp(Status=<Status.Success: 0>), got AF.DataRequestExt.Rsp(Status=<Status.InvalidParameter: 2>)
2020-05-10 22:06:20 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x66d6), DstEndpoint=3, DstPanId=0x0000, SrcEndpoint=3, ClusterId=4, TSN=74, Options=<TransmitOptions.RouteDiscovery: 32>, Radius=30, Data=b'\x01\x4A\x02\x00')
2020-05-10 22:06:20 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.InvalidParameter: 2>)
2020-05-10 22:06:27 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xfab6:1:0x0b04]: async_update
```

puddly, 2020-05-12:
Both `ZDO.TCDevInd` and `ZDO.EndDeviceAnnceInd` are received, so the device indeed did re-join the network. zigbee-herdsman and zigpy-znp both use the  `ZDO.TCDevInd` message as an indication of a device joining the network, while zigpy-cc uses the `ZDO.EndDeviceAnnceInd` message. 

I can't find what exactly the `TCDevInd` actually *is* in the ZDO spec so I'm not too sure what this means. I'll have to test this out on my own network with a sniffer to see what's actually happening.

Adminiuga, 2020-05-12:
`ZDO.EndDeviceAnnceInd` doesn't sound right, ~~although I don't know what it exactly does indicate in ZNP~~ seems to "This callback indicates the ZDO End Device Announce".

From  the docs `ZDO_TC_DEV_IND`  seems to be the right choice.  What's strange, the joined device is not an end  device, but a router. Why there's even the `EndDevAnnceInd`

so from  the  logs, Does `EndDevAnnceInd` also triggers `handle_join()`?
```
2020-05-10 22:06:20 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.EndDeviceAnnceInd.Callback(Src=0x66d6, NWK=0x66d6, IEEE=7c:b0:3e:aa:00:a5:e4:7f, Capabilities=<MACCapabilities.AllocateShortAddrDuringAssocNeeded|RXWhenIdle|MainsPowered|Router: 142>)
2020-05-10 22:06:20 INFO (MainThread) [zigpy_znp.zigbee.application] ZDO device announce: ZDO.EndDeviceAnnceInd.Callback(Src=0x66d6, NWK=0x66d6, IEEE=7c:b0:3e:aa:00:a5:e4:7f, Capabilities=<MACCapabilities.AllocateShortAddrDuringAssocNeeded|RXWhenIdle|MainsPowered|Router: 142>)
2020-05-10 22:06:20 INFO (MainThread) [zigpy.application] Device 0x66d6 (7c:b0:3e:aa:00:a5:e4:7f) joined the network
2020-05-10 22:06:20 DEBUG (MainThread) [zigpy.application] Skip initialization for existing device 7c:b0:3e:aa:00:a5:e4:7f
```

I would assume `ZDO.TCDevInd` is generated based on received ZDO Device_Annce message, ZDO `request id == 0x0013`

puddly, 2020-05-12:
Yes, it appears they both both call `handle_join` in zigpy-znp.

Looking more closely at the Z-Stack source code, it says that `EndDevAnnceInd` is called to "Handle a End Device Announce Descriptor request".

Adminiuga, 2020-05-12:
hrm,  "an End Device Announce Descriptor request"  doesn't ring a bell. Is it in result to a ZDO Parent Device Annce request?

puddly, 2020-05-12:
Just checked with a sniffer:

 - `ZDO.EndDeviceAnnceInd.Callback` is sent when a *Device Announcement* (0x0013)  ZDO message is received. For me, every time an IKEA light is screwed in or the physical switch is turned back on the light sends this message.
 - `ZDO.TCDevInd.Callback` is sent when a *Association Request* (0x01) or *Association Response* (0x02) IEEE 802.15.4 frame is received (the logging timing resolution is not high enough to tell which one but I think it's the latter).

puddly, 2020-05-12:
I believe the solution to this problem would be translating the `ZDO.EndDeviceAnnceInd` into a proper ZDO response and letting zigpy fire the `device_announce` event. Calling `handle_join` is a mistake.

Adminiuga, 2020-05-12:
Well, in this case either would work, but we should call handle_join only for one of those. Atm other libs rely on zdo device annce to trigger handle_join
Since it is an unsolicited message, you can put any tsn

puddly, 2020-05-15:
Potential fixes I've recently pushed:

 - rewriting `ZDO.EndDeviceAnnceInd` to a `ZDOCmd.Device_annce`
 - calling `handle_join` only when a `ZDO.TCDevInd.Callback` is received
 - resetting Z-Stack on startup

I think `Z-Stack` not being reset after a HA restarts was the original problem.

Adminiuga, 2020-05-18:
FYI, I've raised similar issue with CC2531 Router firmware (long time ago). After opening network, they remained open even after timer expiration. I've got new firmware which fixed the issue, but I don't know what was changed in the sources.

puddly, 2020-05-18:
I spent a few hours on and off trying to reproduce this bug with my hardware and I couldn't find up anything unexpected with my sniffer.

Just to be safe, I'll add a timer to call `permit_ncp(0)` after the permit duration is over.

Adminiuga, 2020-05-18:
I want to finish a few things with zha climate and will get back to testing this.

puddly, 2020-05-29:
If you had this problem occur after you flashed the zzh and formed the network, this might be fixed by 4f32b244208ed3747fe1292bcdd90e2a3f2d94a0.

closed 2020-07-25 by Adminiuga