zigpy/zigpy-znp #19: [REQUEST] Feature for "backup and restore" to migrate from one Zigbee coordinator HW adapter to other hardware?

Hedda, 2020-05-11:
This idea came from developer discussion between @Adminiuga & @puddly in https://github.com/zha-ng/zigpy-znp/issues/10

Would it be possible to script a "backup and restore" migration feature/function built into zigpy radio libraries that allow users to relatively easily migrate from one Zigbee coordinator hardware adapter to a other Zigbee coordinator hardware adapter if use same radio library and maybe even if they are not using the exact same hardware MCU chip model/module?

Migrations such as these could become common as new more powerful adapters become available:

- LAUNCHXL-CC26X2R1 development board -> Zig-a-zig-ah (ZZH) adapter
- LAUNCHXL-CC1352P-2 development board -> Zig-a-zig-ah (ZZH) adapter
- LAUNCHXL-CC26X2R1 development board -> LAUNCHXL-CC1352P-2 development board
- Any CC2652R based adapter hardware -> Any CC2652P based adapter hardware
- Any CC2652R based adapter hardware -> Any CC1352P based adapter hardware

Would be a more awesome feature if could also migrate from older TI chips to newer TI chips:

- Any CC2530 based adapter hardware -> Any CC2652 or CC1352P based adapter hardware
- Any CC2531 based adapter hardware -> Any CC2652 or CC1352P based adapter hardware
- Any CC2538 based adapter hardware -> Any CC2652 or CC1352P based adapter hardware

Would really be an awesome feature to have if could be made user-friendly for integration like ZHA!

puddly, 2020-05-13:
This should be possible between all Z-Stack 3/4 radios by just running my script from the linked issue.

The limiting factors for all other migrations will probably be whether or not your old network was Zigbee PRO, if there is a way to extract all of the required information from non-volatile storage, and what you need to do to the extracted to get it to be compatible with each radio.

I'll add my migration script to this package when I finish working through the other more important issues. Cross-vendor migrations are definitely cool but I don't really want to spend the time to work on it 😅

Hedda, 2020-05-14:
Cool! Do you think that your script could then be implemented as a UI feature in Home Assistant ZHA?

puddly, 2020-05-29:
I've implemented backup and restore as runnable modules in 42b14f961823417e59a88367db71b95041653969, see the `README` for instructions.

closed 2020-05-29 by puddly