zigpy/zigpy-znp #20: [REUQEST] Allow option to disable and enable LED/LEDs on adapter or board

Hedda, 2020-05-13:
Copy request from zigpy-cc https://github.com/zigpy/zigpy-cc/issues/42 and https://github.com/zigpy/zigpy/issues/384

Feature suggestion for optional settings to disable LED on CC2531 USB adapter/dongle/stick.

Option to configure setting for disabling and enabling LED/LEDs on the Zigbee radio adapter/dongle.

Requested for CC2531 dongle by Yevgenium in https://github.com/zigpy/zigpy-cc/issues/42

I understand this is dependent on https://github.com/zigpy/zigpy/issues/381 as another example of an optional extension?

puddly, 2020-05-13:
I don't think this is possible without customizing Z-Stack.

Adminiuga, 2020-05-13:
And only would be applicable to one board 🤷 I'd rather have that dev time spent for anything with a better ROI

puddly, 2020-05-13:
Agreed. Just open the clamshell (if your adapter has one) and stick on a little piece of electrical tape.

Also, zigpy-znp doesn't support older radios at the moment because I don't have any older radios to test with (yet).

Hedda, 2020-05-14:
FYI, was possible in zigpy-cc so @sanyatuning set LED to always disabled except for when pairing, see:

https://github.com/zigpy/zigpy-cc/commit/38556957ce468c195fe0593f001d3f336181bb5e

https://github.com/zigpy/zigpy-cc/issues/42

But you are of course both right, this is not critical, just a nice to have and not a need to have ;)

puddly, 2020-05-14:
Hmm, I did see this command earlier but I just thought it let you toggle the LEDs momentarily. If it indeed works on the dev kits, you should be able to send a `c.Util.LEDControl(Laded=0xFF, On=False)` command (though fixing the typos and omitted information pulled from the docs would make the command make more sense).

I'll re-open this issue, since it indeed seems to be possible.

commit f8409140 (2020-06-29): Implement LED control (fixes #20)
closed 2020-06-29 by puddly