zigpy/zigpy-znp #22: How to test

tube0013, 2020-05-28:
I just got a zzh board, and was wondering if I could help test out zigpy-znp with it in a test ha box I have. Are there any instructions on using zigpy-znp to set up a network vs the default zigpy-cc?

puddly, 2020-05-28:
I don't believe there is a way to do it without modifying Home Assistant's `zha` component. The good news is that the change is a single line in `homeassistant/components/zha/core/const.py`:

```diff
diff --git a/const.py b/const.py
index 8a99a8a..4e55c17 100644
--- a/const.py
+++ b/const.py
@@ -169,6 +169,7 @@ POWER_BATTERY_OR_UNKNOWN = "Battery or Unknown"
 class RadioType(enum.Enum):
     """Possible options for radio type."""

+    znp = ("ZNP", __import__("zigpy_znp.zigbee.application"))
     ezsp = (
         "ESZP: HUSBZB-1, Elelabs, Telegesis, Silabs EmberZNet protocol",
         bellows.zigbee.application.ControllerApplication,
```

tube0013, 2020-05-28:
thanks I can handle that. I can close this or keep it open if things change?

puddly, 2020-05-28:
I've not tested the above code or zigpy-znp with Home Assistant so let me know if it works.

As an alternative to editing the HA source code, you *might* also be able to create a custom component that monkey patches that enum in `zha`, allowing you to perform updates and such without having this config get overwritten. I'll try it out when I have some free time and post installation instructions in this project's `README` if it works.

tube0013, 2020-05-28:
Very true, I've done that before. I will give it a shot shortly and see how it goes.

tube0013, 2020-05-28:
So I tried this as a custom component...  I ended up adding:

"zigpy-znp==0.0.4" to the manifest.json

and for testing in const.py I added and import  at the end of the imports:

```
import zigpy_znp.zigbee.application
```

 and to make things simpler just changed the zigpy-cc line in const.py to:

```
    znp = (
        "ZNP: CC2531, CC2530, CC2652R, CC1352 etc, Texas Instruments ZNP protocol",
        zigpy_znp.zigbee.application.ControllerApplication,
    )
```
Auto detection fails, I can point to the serial port, and I seem to have better luck by unplugging/plugging back in the stick right before submitting.  the network seems to get formed but the process to add just hangs.  I can close it and the zha panel shows in integrations but I never get the zha config panel. 

Not a pressing issue, just wanted to record what I tried.

log from adding the integration until it just hangs spinning

```
2020-05-28 18:39:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Probing /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
2020-05-28 18:39:22 DEBUG (MainThread) [zigpy_znp.uart] Connecting to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 at 115200 baud
2020-05-28 18:39:22 DEBUG (MainThread) [zigpy_znp.uart] Opened /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 serial port
2020-05-28 18:39:22 DEBUG (MainThread) [zigpy_znp.uart] Connected to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 at 115200 baud
2020-05-28 18:39:22 DEBUG (MainThread) [zigpy_znp.api] Testing connection to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
2020-05-28 18:39:22 DEBUG (MainThread) [zigpy_znp.api] Sending request: Sys.Ping.Req()
2020-05-28 18:39:22 DEBUG (MainThread) [zigpy_znp.api] Received command: Sys.Ping.Rsp(Capabilities=<MTCapabilities.CAP_APP_CNF|CAP_GP|CAP_UTIL|CAP_ZDO|CAP_AF|CAP_SYS: 1625>)
2020-05-28 18:39:22 DEBUG (MainThread) [zigpy_znp.api] Connected to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 at 115200 baud
2020-05-28 18:39:22 DEBUG (MainThread) [zigpy_znp.api] We were disconnected from /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0: None
2020-05-28 18:39:22 DEBUG (MainThread) [zigpy_znp.uart] Closing /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 serial port
2020-05-28 18:39:22 INFO (MainThread) [homeassistant.setup] Setting up zha
2020-05-28 18:39:22 INFO (MainThread) [homeassistant.setup] Setup of domain zha took 0.0 seconds.
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy.appdb] Loading application state from /config/zigbee.db
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.uart] Connecting to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 at 115200 baud
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.uart] Opened /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 serial port
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.uart] Connected to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 at 115200 baud
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Testing connection to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Sending request: Sys.Ping.Req()
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Received command: Sys.Ping.Rsp(Capabilities=<MTCapabilities.CAP_APP_CNF|CAP_GP|CAP_UTIL|CAP_ZDO|CAP_AF|CAP_SYS: 1625>)
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Connected to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 at 115200 baud
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Sending request: Sys.OSALNVWrite.Req(Id=<NwkNvIds.CONCENTRATOR_ENABLE: 50>, Offset=0, Value=b'\x01')
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Received command: Sys.OSALNVWrite.Rsp(Status=<Status.SUCCESS: 0>)
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Sending request: Sys.OSALNVWrite.Req(Id=<NwkNvIds.CONCENTRATOR_DISCOVERY: 51>, Offset=0, Value=b'\x78')
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Received command: Sys.OSALNVWrite.Rsp(Status=<Status.SUCCESS: 0>)
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Sending request: Sys.OSALNVWrite.Req(Id=<NwkNvIds.CONCENTRATOR_RC: 54>, Offset=0, Value=b'\x01')
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Received command: Sys.OSALNVWrite.Rsp(Status=<Status.SUCCESS: 0>)
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Sending request: Sys.OSALNVWrite.Req(Id=<NwkNvIds.SRC_RTG_EXPIRY_TIME: 56>, Offset=0, Value=b'\xFF')
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Received command: Sys.OSALNVWrite.Rsp(Status=<Status.SUCCESS: 0>)
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Sending request: Sys.OSALNVWrite.Req(Id=<NwkNvIds.NWK_CHILD_AGE_ENABLE: 62>, Offset=0, Value=b'\x00')
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Received command: Sys.OSALNVWrite.Rsp(Status=<Status.SUCCESS: 0>)
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Sending request: Sys.ResetReq.Req(Type=<ResetType.Soft: 1>)
2020-05-28 18:39:23 DEBUG (MainThread) [zigpy_znp.api] Request has no response, not waiting for one.
2020-05-28 18:39:26 DEBUG (MainThread) [zigpy_znp.api] Received command: Sys.ResetInd.Callback(Reason=<ResetReason.PowerUp: 0>, TransportRev=2, ProductId=1, MajorRel=2, MinorRel=7, MaintRel=1)
2020-05-28 18:39:26 DEBUG (MainThread) [zigpy_znp.api] Sending request: Sys.OSALNVRead.Req(Id=<NwkNvIds.HAS_CONFIGURED_ZSTACK3: 96>, Offset=0)
2020-05-28 18:39:26 DEBUG (MainThread) [zigpy_znp.api] Received command: Sys.OSALNVRead.Rsp(Status=<Status.INVALID_PARAMETER: 2>, Value=b'')
[... cut ...]
2020-05-28 18:39:31 WARNING (MainThread) [zigpy_znp.api] Received an unhandled command: AppConfig.BDBCommissioningNotification.Callback(Status=<BDBCommissioningStatus.NoNetwork: 2>, Mode=<BDBCommissioningMode.NONE: 0>, RemainingModes=<BDBCommissioningMode.NONE: 0>)
2020-05-28 18:39:32 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.StateChangeInd.Callback(State=<DeviceState.DiscoveringPANs: 2>)
2020-05-28 18:39:32 WARNING (MainThread) [zigpy_znp.api] Received an unhandled command: ZDO.StateChangeInd.Callback(State=<DeviceState.DiscoveringPANs: 2>)

```

puddly, 2020-05-28:
Thanks for the info, I appreciate the testing.

The ZZH isn't auto detected at the moment because it just shows up as a generic CH340 USB serial adapter, which for me makes them indistinguishable from every other CH340 device out there, like most cheap Arduinos. I'll have to check if there's a better way.

As for the network formation failure, I'll have to check it out myself again. There's some magic combination of requests to get it to happen reliably, I guess I just haven't run into it yet.

Are you in a dense Zigbee environment? I haven't run into that specific status yet.

tube0013, 2020-05-28:
Yes, the area I'm testing it has a bunch of other zigbee devices.. have about 8 other zigbee devices from my production environment in the general area.

puddly, 2020-05-28:
All of the network details have been written to the adapter so it should work fine now with zigpy-cc and zigbee2mqtt. It might even just start working if you re-plug it and restart HA.

puddly, 2020-05-29:
I've tested again with a newly flashed zzh and indeed there were a few bugs introduced recently that prevented new networks from being formed. Thanks for letting me know.

1. I've added installation instructions and a configurable custom component that will inject zigpy-znp into `zha` without requiring any modifications of the Home Assistant source code.
2. NVRAM backup and restoration is also working so if you want to try again but already have a network formed, follow the instructions in the README for performing a backup and then re-flash the zzh using `cc2538-bsl.py`. I've tested it with my network of 50 devices and it starts up again just fine after a restore is performed (it's normal for the first request or two to fail as the routing table is being rebuilt).

tube0013, 2020-05-30:
I reflashed the zzh, set up the custom_component to inject znp into zha, and set up the custom dep for the python package as I'm using supervised and was able to set up a network and join a couple sensors to test. I did have trouble joining a konke temp/humidity sensor - I have several of these in my prod network so not sure if it is the sensor or what I'll keep testing and let you know if I have any issues/feedback.

Adminiuga, 2020-05-30:
Are you using channel 11,25 or 20?

tube0013, 2020-05-30:
15

puddly, 2020-05-30:
Did you use a sniffer to verify? The `channel` network configuration option is not actually utilized because Z-Stack doesn't expose the exact channel it's using, just the mask.

tube0013, 2020-05-30:
no I will check that and likely the issue give Konke's channel "feature" - probably tonight.

puddly, 2020-05-30:
The default channel mask is 15, 20, and 25, which is compatible with Konke. Is it joining the network but not being discovered properly? You should be able to see all of that in the debug logs.

puddly, 2020-06-08:
I've added installation instructions to the `README` and fixed the network formation issue so I'm going to close this issue. Feel free to open a new one if you encounter any more problems.

closed 2020-06-08 by puddly