zigpy/zigpy-znp #24: [REQUEST] Ability to upgrade Texas Instrument coordinator firmware via zigpy-znp and ZHA GUI?

Hedda, 2020-06-08:
Requesting this now as @puddly posted in https://github.com/zha-ng/zigpy-znp/issues/14 saying that he implemented the serial bootloader protocol ([f67e8a4](https://github.com/zha-ng/zigpy-znp/commit/f67e8a48f9a0f9bcf400a6229e2d14b82c3aae70)) used by the Windows utility mentioned in [Koenkk/zigbee2mqtt#320](https://github.com/Koenkk/zigbee2mqtt/issues/320) and was able to upgrade flash his already pre-flashed Texas Instruments CC adapter with a newer firmware without using any external tools, standalone to flash an adapter (in his case he tested it with a CC2531 adapter from ITead which is shipped with an older firmware)

> ```shell
> ip install zigpy-znp
> $ python -m zigpy_znp.tools.flash_read /dev/serial/by-id/radio -o flash.bin
> $ python -m zigpy_znp.tools.flash_write /dev/serial/by-id/radio -i flash.bin
> ```

Suggest considering adding ability to upgrade Texas Instrument coordinator firmware via zigpy-znp. So not just backup and restore but to update firmware if you buy an adapter that has old firmware.

Best would probably be if users could have this ability if only they first manually download and rename firmware binary image file in a specific format to a predetermined file-name and then place that correctly formatted/named file in a specific directory before being able to issue an "upgrade coordinator firmware" command.

Even more awesome would then be to also add the ability to utilize that ability from HA ZHA GUI.

If so it would probably be best to first let the user manually download and unpack the firmware themselves so leave the responsibility to them to make sure it is in the correct format and then having a button in the ZHA UI for "upgrade coordinator firmware".

More user-friendly would then, of course, be if ZHA let you enter a URL source location for automatic download of firmware, like for example https://github.com/Koenkk/Z-Stack-firmware/

Hedda, 2020-06-08:
Cross-posted this request to https://github.com/zigpy/zigpy-cc/issues/53 to discuss the ZHA GUI part.

puddly, 2020-06-08:
I renamed the commands to `flash_read`/`flash_write` because that's what they do, you can use the latter to write the coordinator firmware `.bin` file.

As for a UI, I think I've made it user friendly enough for now. There is an image size check to prevent the wrong files from being flashed and I've almost got the CRC working.

Hedda, 2020-06-08:
@puddly Does `flash_read` and `flash_write` support flashing TI CC13x2 and CC26x2 adapters too?

Maybe OK to ask @omerk from @electrolama if could test with his zig-a-zig-ah and zoe adapters?

- https://electrolama.com
  - https://github.com/electrolama/zig-a-zig-ah
  - https://github.com/electrolama/zoe

I don't have a TI CC13x2/CC26x2 adapter yet but I have placed an order for a zzh from Electrolama.

puddly, 2020-06-08:
The ZZH uses a different serial bootloader but you can flash it with `cc2538-bsl.py`. This is just for the CC2531.

Hedda, 2020-06-08:
Ah ok, could that be implemented into zigpy-znp as well?

puddly, 2020-06-08:
Why? `cc2538-bsl.py` is open source.

I'm not very familiar with the bootloader interfaces and only added this to zigpy-znp because that bootloader interface is implemented as a set of MT commands for the CC2531.

The bootloader for the other chips seems to use a completely different protocol. There's already a cross-platform tool to do it both graphically and via the command line.

closed 2020-06-08 by puddly

Hedda, 2020-06-09:
Guess I just through it is more likely to ever be implemented in UI for Home Assistant's ZHA if it was already implemented in a radio library for zigpy