zigpy/zigpy-znp #28: [SUGGESTION] Suggest zigpy-znp and ZHA devs team-up with slaesh to test his CC2652RB stick as another reference adapter

Hedda, 2020-08-10:
Suggest zigpy-znp + zigpy-cc and ZHA devs team-up with @slaesh to test his new CC2652RB stick as another reference adapter.

@slaesh I believe that you already donated adapters to the Zigbee2mqtt project so wondering if you would also be willing to donate some of your CC2652RB stick adapters to @sanyatuning & @Adminiuga & @puddly & @dmulcahey for [zigpy/zigpy-cc](https://github.com/zigpy/zigpy-cc) and [https://github.com/zha-ng/zigpy-znp](https://github.com/zha-ng/zigpy-znp) development as used with zigpy in [Home Assistant's ZHA integration component for its native Zigbee support](https://www.home-assistant.io/integrations/zha/)?

FYI; @slaesh has developed his new CC2652RB stick as an open source hardware designed similar to the CC2652R stick by Electrolama/omerk as a another friendly competition option, and as noted, slaesh stick is based on CC2652RB instead of CC2652R. His adapter is since recently listed as a supported reference adapter by Zigbee2MQTT on their project page.

https://github.com/slaesh/cc2652-stick

and

https://slae.sh/projects/cc2652/

Like omerk, slaesh is also based in Europe and is selling the finished product (unflashed) via Tindie 

https://www.tindie.com/products/slaesh/cc2652-zigbee-coordinator-or-openthread-router/

slaesh, 2020-08-11:
thanks @Hedda !
Sure, sounds great. Thanks for your request.

Currently stuck during a business trip.

Maybe write me for more details https://t.me/slaesh

Hedda, 2020-08-11:
> thanks @Hedda !
> Sure, sounds great. Thanks for your request.
> 
> Currently stuck during a business trip.
> 
> Maybe write me for more details https://t.me/slaesh

@slaesh That is very nice of you! Could you maybe post that in https://github.com/zigpy/zigpy-cc/issues/63 and https://github.com/zha-ng/zigpy-znp/issues/28 telling @sanyatuning & @Adminiuga & @puddly & @dmulcahey that they can each e-mail their own postal addresses to you if they would like to get you? 

PS: I don't have their contact information or any personal connection to them myself, I am really just a fan of zigpy and ZHA ;)

Hedda, 2020-09-02:
@slaesh Back from your business trip?

pipiche38, 2020-09-03:
hello, can you help me and let me know if that is the proper way to start the ControllerApp ?

```
        self.zigpyApp = await ControllerApplication.new(
            config=ControllerApplication.SCHEMA({
                "database_path": /tmp/zigpy.db,
                "device": {
                    "config_for_port_path":"/dev/ttyUSB0"
                }
            }),
            auto_form=True,
            start_radio=True,
        )
```

Do I have to do more , is serial line the only way to get access to the hardware ? Is there any other parameters I should take care ?

puddly, 2020-09-03:
@pipiche38 I believe I answered your question within the [Domoticz-Zigpy repo](https://github.com/pipiche38/Domoticz-Zigpy/issues/10#issuecomment-686606027).

In the future, please open a new issue or tag me as @puddly within an issue opened in your own project (which is how I was notified of your question to begin with 😃). When you comment on an existing issue, everybody involved will get notified, which may include people who aren't involved.

Adminiuga, 2020-09-03:
basic zigpy schema is defined in https://github.com/zigpy/zigpy/blob/dev/zigpy/config/__init__.py, but some `RadioLibs` may change/extend `SCHEMA_DEVICE` to have some addional options like `baudrate` or `flow_control` check sources for other radios.
It is not `"config_for_port_path"` but `"path"`, better use constants 
 defined in the radio lib or https://github.com/zigpy/zigpy/blob/64e4f731d34e3ed0aa75cb1146cb79ea1314c91a/zigpy/config/__init__.py#L23

pipiche38, 2020-09-03:
> @pipiche38 I believe I answered your question within the [Domoticz-Zigpy repo](https://github.com/pipiche38/Domoticz-Zigpy/issues/10#issuecomment-686606027).
> 
> In the future, please open a new issue or tag me as @puddly within an issue opened in your own project (which is how I was notified of your question to begin with 😃). When you comment on an existing issue, everybody involved will get notified, which may include people who aren't related to your specific problem.

Sorry for that. I'll do it from now.

Thanks @Adminiuga for the information.

puddly, 2020-09-14:
Thanks for the suggestion. As much as I like growing my CC2652R collection, I don't think testing is needed for an adapter that is functionally identical to the ZZH with the exception of a much better USB serial chip.

@slaesh has sent me all the information I need to autodetect the serial port, although that functionality isn't being used by HA at the moment.

closed 2020-09-14 by puddly

manouallou, 2021-02-28:
Hello everyone,

Can you please confirm to a non tech-savvy guy like me (and perhaps others) that slaesh's usb coordinator will work with Home Assistant and ZHA? I am currently using a CC2531 from Sonoff as my coordinator and I am looking to upgrade to a better one since I am facing intermittent connection problems with the CC2531.

Hedda, 2021-03-01:
@manouallou  Yes CC2652**RB** is supported https://github.com/zigpy/zigpy-znp/blob/dev/README.md#hardware-requirements

It needs Z-Stack 3 coordinator firmware https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

Home Assistant ZHA depends on zigpy-znp https://www.home-assistant.io/integrations/zha/#known-working-zigbee-radio-modules

PS: Off-topic; just be aware that several others are working on CC2652**P** based coordinators which should offer even greater range.

manouallou, 2021-04-02:
Hey @Hedda thank you for the information. Really helpful! 

I am still expecting my CC2652RB stick from Slaesh to test it out. I have paid since February 27th (today is April 3rd) and the dude is not replying to any of my emails or Telegram, nothing. He is seriously got some problems with his communication and his production/distribution!

Maybe it is a sign to cancel the CC2652RB and go for a CC2652P based coordinator as you suggested which indeed looks really interesting. Problem is I cannot easily find any available online.