zigpy/zigpy-znp #29: [REQUEST] Zigbee 3.0 and/or Zigbee PRO network forming support in zigpy-znp?

Hedda, 2020-08-10:
This is a request/suggestion to make zigpy-znp support Zigbee 3.0 and/or Zigbee PRO network forming?

This directly relates to Zigbee 3.0 network supported requested for zigpy in https://github.com/zigpy/zigpy/issues/360 and https://github.com/zigpy/bellows/issues/292

That is, make it possible for zigpy-znp with or without zigpy to form a Zigbee 3.0 network allowing newer devices to use a Zigbee 3.0 profile as an option instead of ZHA 1.2 profile in a Zigbee Home Automation 1.2 network if and when newer devices support Zigbee 3.0 profiles? 

If I understand it correctly, some of not all newer Zigbee Stacks support Zigbee network forming which allow a Zigbee network formed which support multiple profiles at the same time, allowing Zigbee 3.0 devices to connect using a Zigbee 3.0 profile, Zigbee Light Link 1.x devices can connect using a ZLL 1.0 profile, and Zigbee Home Automation 1.2 devices can connect using a ZHA 1.2 profile, or?

I know that bellows radio library just very recently got support for the later EZSP API versions used Silicon Labs EmberZNet Zigbee Stack firmware that supports Zigbee 3.0 and the zigpy-znp radio library supports the newer Texas Instruments chips use Z-Stack 3.0 Zigbee Stack firmware which all do support Zigbee 3.0 standard. 

https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2018/04/24/how_to_form_zigbee3-zrzB

https://www.silabs.com/documents/public/application-notes/an1117-migrating-zigbee-profiles-to-z30.pdf

https://www.ti.com/lit/swra615

PS: Previous versions of Silicon Labs firmware was also Zigbee PRO compatible, and the core of the Zigbee 3.0 standard is based on the Zigbee PRO 2015 specification, while the core of the Zigbee 3.1 standard will be based on the based on the Zigbee PRO 2017 specification.

puddly, 2020-08-10:
I believe that there are no changes to network formation, you just need to supply the device's install code to the coordinator before it joins the network.

This won't be worked on for a while because I don't have any Zigbee 3.0 devices to test with (although I think I heard somewhere that IKEA might be pushing OTA updates to make their end-devices Zigbee 3.0 compliant?). For reference, here is how bellows implements it: https://github.com/zigpy/bellows/blob/5874e59e14554479c76a297c35d8821fcd5dd033/bellows/zigbee/application.py#L541-L564

Hedda, 2020-08-11:
Most added Zigbee 3.0 support via OTA updates so you probably have many more Zigbee 3.0 capable devices than you know ;)

All new devices from a while back must be Zigbee 3.0 compatible to be added on Zigbee Alliance list of their certified devices:

https://zigbeealliance.org/zigbee_products/

Just some examples of very popular and commonly available devices include those from manufactures/brands such as;  GE, HEIMAN, IKEA, Innr, LEDVANCE, Leedarson, Legrand, Linkind, Müller-Licht, Namron, Paul Neuhaus, Paulmann, Philips (Hue), Schneider Electric, Sengled, Samsung (SmartThings), Sunricher, Sylvania, TaHoma, TuYa, Ubisys, and even Xiaomi (Aqara).

https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Aqara
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=GE
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=HEIMAN
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=hue
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=IKEA
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Innr
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=LEDVANCE
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Leedarson
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Legrand
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Linkind
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=M%C3%BCller
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Namron
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Neuhaus
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Paulmann
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Schneider
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Sengled
https://zigbeealliance.org/zigbee_products/page/2/?product_type=certified_product&se=SmartThings
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Sunricher
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Sylvania
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=TaHoma
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Ubisys

As you can see on that list many popular devices that have already been available to buy for a while are now Zigbee 3.0 certified.

Zigbee Alliance is pushing very hard for Zigbee 3.0 certification and devices need to be independently tested to be listed there.

PS: I also read that ITead is aiming to get its latest Sonoff branded Zigbee sensors (Silabs based) certified for Zigbee 3.0 now.

puddly, 2020-08-12:
Most of my IKEA devices are listed and are running the latest firmware but I don't have any devices with any install codes to actually try out. Maybe I'll get some of the newer Aqara sensors or something...

Adminiuga, 2020-08-12:
the new Aqara US switches (with or without neutral) which are ZB3 come with install code.

Hedda, 2020-08-13:
> the new Aqara US switches (with or without neutral) which are ZB3 come with install code.

If that is true for the new Aqara "S2" switches then maybe we can assume that is same with all these new "S2" labelled devices?

https://zigbeealliance.org/zigbee_products/wall-switch-with-neutral-single-rocker-s2/
https://zigbeealliance.org/zigbee_products/wall-switch-with-neutral-double-rocker-s2/
https://zigbeealliance.org/zigbee_products/occupancy-sensor-s2/
https://zigbeealliance.org/zigbee_products/smart-plug-s2/
https://zigbeealliance.org/zigbee_products/wall-outlet-s2/

And if that holds true for those then wondering if is same for the new "T1" labelled devices also certified as Zigbee 3.0 as well?

https://zigbeealliance.org/zigbee_products/wireless-mini-switch-t1/
https://zigbeealliance.org/zigbee_products/door-and-window-sensor-t1/
https://zigbeealliance.org/zigbee_products/water-leak-sensor-t1/

By the way, I see that they in addition have a new EU version of the Aqara Smart Plug coming out (already available in Russia):

https://www.aqara.com/en/smart_outlet.html

puddly, 2020-09-24:
`ControllerApplication.permit_with_key` is implemented and works with Z-Stack 3 devices. I've tested and confirmed the network key is not sent in plaintext with a Consciot bulb that uses install codes.

commit 2dffbc70 (2020-09-24): Implement and test support for install codes (fixes #29)
closed 2020-09-24 by puddly