zigpy/zigpy-znp #31: Change Channel

sshaikh, 2020-09-11:
I am using Zstack3 on a CC2531, with zigpy-znp via the HA beta.

I have successfully run the channel scan as per the readme, but can't figure out how to set this. Z2M has a config parameter in the yaml, for instance.

I am only using the ZHA integration and have not edited the HA config file for is (as I believe all configuration should be done via the UI). Should I add config for channel? If so, where?

puddly, 2020-09-11:
You'll have to re-form your network but you can also change it in `configuration.yaml` (https://github.com/zigpy/zigpy/issues/154#issuecomment-635484220):

```yaml
zha:
  zigpy_config:
    network:
      channel: 15
```

puddly, 2020-09-11:
The ZHA user interface doesn't expose many of these options (yet!) so to re-form your network, shut down HA, run the following command within your virtual environment, and re-start HA:

```console
(venv) $ python -m zigpy_znp.tools.nvram_reset /dev/serial/by-id/your-cc2531
```

closed 2020-09-11 by puddly

Adminiuga, 2020-09-11:
@puddly wanna try doing something like this? https://github.com/Adminiuga/zha_custom/blob/f5ef6a259d3b38d90ad7531480aa2a9eccf6d21c/ezsp.py#L33

Ch_mask is actually the new channel. Send a broadcast to all devices about channel change the actually set the channel on radio.  Probably not going to work for xiaomis, but at least could avoid rejoin ing all routers

puddly, 2020-09-11:
I'll have to check the spec more closely.

The worst case scenario that I'd like to support is a network that includes both routers and RFDs, but some routers are critical and cannot drop off the network without severing communication with another part of the network. Imagine a coordinator and a bulb that are too far apart to communicate directly, except through another intermediate bulb. If the broadcast reaches only the intermediate bulb, that distant bulb will never migrate and cannot receive another broadcast.

I think the most foolproof way to migrate the network to a new channel would be to traverse the network tree (e.g. from `relays` attribute updates) and iteratively send unicast messages to migrate nodes that have no children.

Adminiuga, 2020-09-11:
Broadcast should be re-,relayed by a few nodes. If there's only one node, then yeah, most likely going to drop.
Try sending unicast, but if the device switches immediately and that one was an intermediary device, then the comms to the rest of the network is going to be severed.

Also, if the device misses the update, it should try rejoining scanning all channels (or channels per channel mask) and it should hear beacons from the devices that switched successfuly. I swear I thought I've seen Konke working on channel 18 after a change from ch 25

puddly, 2020-09-11:
@Adminiuga a while ago I tried just switching the channel of my entire network (probably in a terrible way, by directly modifying the internal network IB struct in NVRAM) and none of my IKEA bulbs ever tried to re-join, despite me switching from one channel in the mask to another, power cycling the bulbs more than a couple of times, and waiting for almost an hour. Perhaps I didn't look at my topology closely enough and maybe the bulbs that were next to my coordinator weren't joined directly to it but through a parent in the same room?

Once I finish the unit test rewrites and the Z-Stack Home 1.2 integration I'll form a new network to play around with this stuff.

MattWestb, 2020-09-13:
I was testing changing channel in bellows in the beta with zha_custom.

Service:
zha_custom.execute
Service Data:
command: ezsp_set_channel
command_data: 25

Waiting some minutes and restarting and looking in the HA log and the NCP is living on channel 25 (before 15).
3 IKEA GU10 WS (ZB3) have moving and working.
2 HOMA LED Drives has not and need reseted so it can being rejoined (Losing group membership).
Then having the HOMAs rejoined / paired the IKEA motion sensors (Old and New) is rejoining also the E1743 and the 5 button remote (I don't know if it was open the network for joining that was doing the end devices).

In the end it was going better than i was expected only the HOMAs (chinese ZB3)  was making trouble and the other was keeping all bindings and groupe :-))))

wes1993, 2021-07-20:
> I am using Zstack3 on a CC2531, with zigpy-znp via the HA beta.
>
> I have successfully run the channel scan as per the readme, but can't figure out how to set this. Z2M has a config parameter in the [...]

Hello, how you have do the channels scan?