zigpy/zigpy-znp #32: Migrate asynctest to stdlib's AsyncMock

Gamester17, 2020-09-14:
Same as https://github.com/zigpy/bellows/issues/339 and https://github.com/zigpy/zigpy/issues/475 as reported by @fabaff of Home Assistant / hass.io :

_**Asynctest doesn't work with Python 3.8 and above. It's the same issue as for zigpy (zigpy/zigpy#475).**_

puddly, 2020-09-14:
This is already being done: https://github.com/zha-ng/zigpy-znp/blob/54b21cfbeae58ed7275a4de58291ab8d143010f9/tests/test_application.py#L6-L10

I can also check if making the dependency install only if Python 3.7 is in use will work (e.g. `'asynctest; python_version < "3.8.0"'`).

puddly, 2020-09-20:
Both changes have been implemented. Thanks for bringing this up.

closed 2020-09-20 by puddly