zigpy/zigpy-znp #44: Should you backup NVRAM before firmware update?

MarhyCZ, 2020-10-28:
Hi! 
I finally completely switched to using the ZHA with this library and I must say, the setup of all devices was much more smooth than when I was doing it with deconz. The attribute reporting like Cover position with Fyrtur blinds was setup immediately, group binding, device binding, Everyting worked right away. Really good piece of sw! 

I am not so sure from the documentation if I need to do NVRAM backup/restore when upgrading Koenk Z-Stack firmware. (So no migration from different radios)
The zigbee2mqtt does a automatic coordinator backup from time to time, but I think the zigpy does not. 

So should the right firmware upgrade procedure when using ZHA/zigpy-znp be: Backup NVRAM, reflash new firmware(I am using the https://github.com/JelmerT/cc2538-bsl ),and then restore nvram? 

I just want to avoid nicely spent 2-3 hours of repairing devices haha.

puddly, 2020-10-28:
Running `cc2538-bsl.py` with `-evw` will indeed erase everything, including the NVRAM. Running it without `-e` doesn't seem to work and leaves my device unbootable until reflashed with the mass erase option.

When restoring NVRAM your network will have a bunch of routing errors for a bit because zigpy-znp at the moment doesn't proactively send route discovery messages like zigbee2mqtt does. It'll take a few minutes for everything to sync back up.

MarhyCZ, 2020-10-28:
Ok, understood, and the NVRAM backup/restore is mandatory, so without it, it would not work without recreating network? (the zha/zigpy does not cache the settings that are needed like pan_id etc.?). Sorry for asking and not finding it by myself.

I also see that your library has a flash.write tool. So can I just use that and feed it the .hex firmware file from Koenkk? If that handles the upgrade more gracefully, maybe I can put a commit to a README.md with info about fw upgrade through your library? (Because I am not sure if the users actually know about this, or if its intentional that way.)

However thanks for a really fast reply!

puddly, 2020-10-28:
Not a problem, I'll update the documentation to make this stuff more clear.

1. That's correct. Upgrading the firmware wipes all of your settings and because zigpy/ZHA assumes your device persists all of your settings, your network will be completely gone if you don't back it up before upgrading.
2. `flash_read/write` only works with the CC2531's bootloader, unfortunately. I should probably rename it. You'll have to use `cc2538-bsl.py` for sticks using the CC2652R/RB.

MarhyCZ, 2020-10-28:
Now its totally clear!
I almost didn't realize this and pushed the enter for the fw upgrade without even backing it up..

So big thanks for all the info! Hope you are staying safe.

commit ea29537d (2020-11-10): Clarify the need for NVRAM backup/restore during upgrades (fixes #44)
closed 2020-11-10 by puddly