zigpy/zigpy-znp #47: [REQUEST] ZGP (Zigbee Green Power) with zigpy-znp and Texas Instruments Z-Stack 3

Hedda, 2020-11-12:
@puddly Requesting support for ZGP (Zigbee Green Power) using zigpy-znp with Texas Instruments Z-Stack 3 Zigbee coordinator.

FYI, @zoic21 is trying to implement ZGP (Zigbee Green Power) device support into zigpy, however he wrote that he is currently developing it using deCONZ (ConBee II) and Silicon Labs Zigbee coordinator and not a TI coordinator. See -> https://github.com/zigpy/zigpy/pull/519

PS: zoic21 is currently using EnOcean PTM 215ZE (PTM215ZE) based ZGP switch as reference but it should maybe be noted that it looks like the newer EnOcean PTM 216Z (PTM216Z) might be more common use, please see -> https://github.com/zigpy/zha-device-handlers/issues/478

PPS: There is more info about ZGP (Zigbee Green Power) and devices supporting it in the zigpy request -> https://github.com/zigpy/zigpy/issues/341

puddly, 2020-11-12:
I can add the required endpoint but I have no way to actually test this out. Do you know of any cheaper ZGP devices? $30 for just a module is a little steep.

Hedda, 2020-11-18:
There are links to a number of ZGP devices in https://github.com/zigpy/zigpy/issues/341 but unfortunately there looks to be no inexpensive ones as of yet.

Maybe find a used one or might be possible to one on sale during next weeks Black Week and Black Friday or Cyber Monday?

Regardless it would probably be most useful to get a remote control model or a wireless switch model instead of only a module.

> Most common examples of Green Powered devices for home automation are rocker wall switches:
> 
> * https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=green+power
>   
>   * https://runlesswire.com/`click_philips_hue`/
>   * http://illumra.com/products/wireless-light-switches/zigbee-switch
>   * https://www.signify.com/global/our-company/news/press-releases/2018/20180830-new-switch-and-new-luminaire-partners-offer-more-choice-in-design-and-control-of-your-philips-hue-smart-lights
>   * https://www.cnet.com/news/this-boring-switch-might-be-the-best-philips-hue-gadget-yet/#ftag%3DCAD590a51e
>   * https://community.smartthings.com/t/theyre-here-new-green-energy-battery-free-wall-switches-for-hue-bridge-illumra-busch-jaeger-etc/135280
>   * https://zigbee.blakadder.com/Philips_9290011152X.html
>   * https://new.abb.com/low-voltage/products/residential-product/friends-of-hue-smart-switch
>   * https://www.enocean.com/en/products/enocean_modules_24ghz/ptm-216z/
>   * https://www.enocean.com/en/products/enocean_modules_24ghz/ptm-215ze/
>   * https://www.enocean.com/en/products/enocean_modules_24ghz/easyfit-single-double-rocker-wall-switch-for-zigbee-ewssz-ewsdz/
>   * https://www.enocean.com/en/products/enocean_modules_24ghz/easyfit-single-double-rocker-wall-switch-for-zigbee-ewsszg-ewsdzg/
>   * https://www.enocean.com/en/products/enocean_modules_24ghz/

Ratief, 2020-12-08:
I would be willing to buy you a ZGP device if it would help get this working. Personally I love the Hue Taps and they are the last device that I need to migrate away from my bridge.

Kazymire, 2021-02-15:
yes i am very interesting in zgpd too, now i using cc usb dongle with z2m, but i see that with conbbe II, we can see gpd trame. but not with cc dongle.
As Ratief, it's the last fonction i need, to migrate on zigpy
Best regard

Hedda, 2021-02-15:
FYI, zoic21 have made progress with ZGP for zigpy and is requesting regression testing of his new pull request in https://github.com/zigpy/zigpy/pull/656

> Can someone can test with this code if all it's ok ? No side effect. If it I think we can merge this first step (after I fix coveralls).

That is, he is asking people do functional and non-functional tests to ensure that his patch does not break any existing functions. If it does not break anything obvious in zigpy then it could possible be merged 'as-is' even if ZGP might be be fully production ready.

sepahewe, 2021-03-04:
@puddly Would you still be helped by a device? If so, PM me on twitter (see my profile) and I'll send you one.

omerk, 2021-03-10:
@puddly Happy to contribute to the R&D budget for this exploration, can also send a few of the CC2652R/P sticks if you need spares/test devices?

Hedda, 2021-03-24:
> Happy to contribute to the R&D budget for this exploration, can also send a few of the CC2652R/P sticks if you need spares/test devices?

@zoic21 @sarakha63 Do you have Texas Instruments Z-Stack 3 ZNP coordinators to test with your https://github.com/zigpy/zigpy/pull/656 PR for zigpy?

zoic21, 2021-03-24:
No we only focus for now on deconz and EZSP but after integration on this type of key we can manage integreation on ZNP.

Hedda, 2026-10-31:
Bump this request now ZGP (Zigbee Green Power) frame RX and packet forwarding has been added to bellows, zigpy-deconz, and ziggurat:

* https://github.com/zigpy/bellows/pull/767

* https://github.com/zigpy/zigpy-deconz/pull/279

  * https://github.com/zigpy/zigpy-deconz/pull/281

* https://github.com/zigpy/ziggurat/pull/69

Texas Instruments CC2652P is known among the most commonly used Zigbee Coordinaor adapter, (second only to EFR32MG21/EFR32MG24).

puddly, 2026-10-31:
I tested this a few weeks ago and I found no way to enable GP reception in Z-Stack. It just isn't compiled in. If you want Green Power support, use either https://github.com/zigpy/bellows or https://github.com/zigpy/zigpy-ziggurat.

closed 2026-10-31 by puddly

Hedda, 2026-10-31:
> I tested this a few weeks ago and I found no way to enable GP reception in Z-Stack

Hmm, I am pretty sure I read that Zigbee2MQTT does support Zigbee Green Power devices via the z-stack adapter in zigbee-herdsman?

* https://github.com/Koenkk/zigbee-herdsman/tree/master/src/adapter/z-stack

Their documentation does at least not specifcially mention that zStack based (Texas Instruments) adapters have any such limitations?

* https://www.zigbee2mqtt.io/guide/adapters/

  * https://www.zigbee2mqtt.io/guide/adapters/zstack.html

* https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices

  * https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices

You can also find Koenkk's original implementation announcement here from 2020 (back when Z2M more or less only supported Z-Stack): 

* https://github.com/Koenkk/zigbee2mqtt/issues/3322

From the discussion here it at least sounds as if they got Zigbee Green Power support working with the zstack adapter in zigbee-herdsman.

puddly, 2026-10-31:
You can use ZGP even if the coordinator itself doesn't support receiving GPDFs directly: you just need to have a Zigbee 3.0 router on the network, which will tunnel the frames encapsulated in a ZCL cluster command. This issue is tracking GP reception support for TI radios and as far as I can tell, it just isn't possible to receive GP frames with a TI coordinator using the Z2M firmware builds, you need a router to proxy it.

The only two radios that can properly handle Green Power are SiLabs (via bellows) and Ziggurat. deCONZ has partial support but has debouncing issues with unencrypted frames.

Hedda, 2026-11-01:
> The only two radios that can properly handle Green Power are SiLabs (via bellows) and Ziggurat. deCONZ has partial support but has debouncing issues with unencrypted frames.

Should that then maybe be mentioned under limitations in ZHA docs? 

* https://www.home-assistant.io/integrations/zha#limitations

Noticed that next release will removed ZGP section under limitation:

* https://next.home-assistant.io/integrations/zha#limitations

Perhaps add something new under ZHA limitations like example this:

### _**ZGP (Zigbee Green Power) limitation on some radios**_

_**The only two types of radio adapters that can handle ZGP (Zigbee Green Power) direct reception today in ZHA are Silicon Labs (via bellows) and Ziggurat. The deCONZ based adapters do also have partial support for ZGP but have issues with unencrypted frames. You can still use ZGP with other radios if you have a Zigbee 3.0 router on your network which will proxy the ZGP packages.**__

Probably a good idea to be clear about that, especially if not planned.

Hedda, 2026-11-01:
@puddly I have now promted AI (Claude Opus 5.5 Max) to dig into this request for gaps in zigpy-znp ZGP support and analyze zstack adapter in their zigbee-herdsman + research Zigbee2MQTT's previous discussions, while comparing it to your implementations in bellows, zigpy-deconz and zigpy-ziggurat. Got this result back copied below with suggestions on what is possibly missing in zigpy-znp as well as a patch:

### TL;DR

- Z-Stack ZNP firmware can't pass the GPDFs it hears to the host. TI said so when Koenkk asked, and nothing has changed in the current SDK or in Koenkk's patches.
- zigbee-herdsman gets Green Power on Z-Stack only from GP Notification and GP Commissioning Notification commands tunneled by GP proxies. It never uses the MT_GP subsystem.
- zigpy-znp needs to join the GP group, tag GP-cluster frames with the GP profile, and refuse aliased sends. Patch and description below.
- Direct reception at the znp coordinator, as bellows, zigpy-deconz and ziggurat now do it, would need a new ZNP firmware first.

Koenkk never got Z-Stack to receive Green Power (GP) frames either. zigbee-herdsman’s Z-Stack adapter only sees GP frames that a GP-capable router (a “proxy”) has already received and repackaged as ordinary Zigbee messages. The ZNP firmware never hands the host a GP frame the coordinator heard itself. zigpy can now take that same router path, but zigpy-znp still needs three small fixes for the router path to work.

Download patch for zigpy-znp to fix router path:

[zigpy-znp-green-power-proxy-path.patch](https://github.com/user-attachments/files/34246760/zigpy-znp-green-power-proxy-path.patch)

#### What zigbee-herdsman really does on Z-Stack
- **Two setup steps, nothing else firmware-specific.** It registers endpoint 242 with the GP profile 0xA1E0 (`endpoints.ts`). It also adds endpoint 242 to the GP group 0x0B84 with `extFindGroup`/`extAddGroup` (`manager.ts:101`). Both came in Koenkk’s original “Green Power support” commit, April 2020.
- **No use of TI’s GP host interface.** It never uses the MT_GP subsystem; there’s one secReq definition that nothing calls. Commissioning mode, notification parsing and pairing all live in the adapter-agnostic `greenPower.ts`, which works on messages forwarded by routers.
- **TI told Koenkk the ZNP can’t forward GP frames.** In 2022 he asked TI how to get Hue Tap commissioning frames out of a CC26x2 ZNP. The ZNP project defines `DISABLE_GREENPOWER_BASIC_PROXY`, which removes the GP task (`gp_event_loop`) from the firmware. Without that task, the GP receive layer never reaches `MT_GPDataInd`, and removing the flag breaks the build. TI’s conclusion was that the SDK 6.30 ZNP will not forward GP messages to any host.
- **Nothing has changed since.**
  - The current SDK (lpf2‑8.33) still gates `gp_event_loop`/`gp_Init` behind that flag in `zstackstartup.c`.
  - Koenkk’s Z-Stack 3.x.0 firmware patch doesn’t touch GP. -> https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0
  - His CC2530/CC2531 3.0.x builds add `#define DISABLE_GREENPOWER_BASIC_PROXY` explicitly. -> https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x

So puddly’s conclusion matches TI’s own answer. The ZHA docs for 2026.11 say the same thing: direct GP for `ziggurat`, `ezsp` (bellows) and `deconz`, while znp only works through routers.

#### What zigpy-znp needs for the router path

zigpy 2.4 already registers endpoint 242 through `register_endpoints()`, and `permit()` already tells routers to start GP commissioning. The gaps are in zigpy-znp:

1. **Join the GP group.**
    - After commissioning, zigpy calls `subscribe_to_multicast_group(0x0B84, endpoint_id=242`), but zigpy-znp doesn’t implement it, so nothing happens.
    - Routers send button presses to group 0x0B84, and stock Z-Stack drops group messages when no endpoint is a member.
    - The fix is `ZDO.ExtAddGroup(242, 0x0B84)`, accepting “already exists” as success, plus `ExtRemoveGroup` for unsubscribing. This is the same thing herdsman does.
2. **Infer the GP profile.**
    - Z-Stack’s incoming-message callback (`AF.IncomingMsg`) has no profile field, so zigpy-znp takes the profile from the destination endpoint.
    - Koenkk’s firmware delivers group messages for groups nobody joined to endpoint 1, so a GP notification arrives labelled with the HA profile.
    - zigpy only recognises forwarded GP frames when the profile is 0xA1E0, so it misses them and logs “Unknown device”.
    - The fix: anything from endpoint 242 on cluster 0x0021 gets profile 0xA1E0, whatever the destination endpoint.
3. **Refuse alias sends.**
    - zigpy announces each GP device’s short address by sending as that address (an “alias”), and falls back if the radio raises `NotImplementedError`.
    - Z-Stack’s send commands have no alias parameter, and zigpy-znp silently ignores the alias instead of raising.
    - The fix is to raise `NotImplementedError`, as zigpy-deconz does.
4. **Housekeeping.**
    - Require `zigpy>=2.4.0`.
    - The current test suite collapses on zigpy 2.4.1: 137 failed, against 167 passing on 2.3.0. Startup now sends a GP commissioning broadcast that the fake ZNP server doesn’t answer, and later message sequence numbers shift.

`zigpy-znp-green-power-proxy-path.patch` implements all four, against zigpy-znp `dev` (926626c):

- The full suite passes on zigpy 2.4.1 (412 tests).
- 12 of the 16 new GP tests fail on current dev, so they catch the bugs above.
[... cut ...] o it sends as itself with the alias's fixed TSN 0. Fix: raise `NotImplementedError`, as [zigpy-deconz does](https://github.com/zigpy/zigpy-deconz/blob/0a6de2196e0cd100dfc7c7f1e587f3e8c819086b/zigpy_deconz/zigbee/application.py#L603-L605).
4. **Housekeeping.** Require `zigpy>=2.4.0`. Against zigpy 2.4.1, 137 of the 167 tests in `tests/application` currently fail (all pass on 2.3.0). Startup now broadcasts GP Proxy Commissioning Mode, which the fake ZNP server doesn't answer, and later TSNs shift.