zigpy/zigpy-znp #51: 100% CPU Load while requests are active (eg toggling lights)

dumpfheimer, 2020-12-14:
Hi!

I noticed that my CPU load goes up to 100% while I am toggling multiple lights (almost happens every time I toggle living room lights - ~15 lights). I shortly had a very stable network with all the lights responding extremely fast: then the CPU load was normal.

I tried debugging asyncio but I did not get any messages.
Is there anything I can do to find out what the cause is?

puddly, 2020-12-14:
zigpy-znp does not do any processing at all. It just asynchronously sends/receives messages over serial.

Make sure your Home Assistant database isn't too big. Are you running the latest HA beta? If so, try individually downgrading the following packages and see if anything changes:

 - `pyserial-asyncio==0.4`  (currently `0.5`)
 - `pyserial==3.4`  (currently `3.5`)
 - `zigpy-znp==0.2.2`  (currently `0.3.0`)

You will probably have to run Home Assistant with `--skip-pip` so it doesn't upgrade them.

Adminiuga, 2020-12-14:
Oopsie, wrong button...

dumpfheimer, 2020-12-16:
Okay, I found out what caused this: ZHA Map

I was looking at "ZHA Map" to check link qualities but as it turns out it drastically reduces performance.
My guess is that it refreshes all devices each time lqi is updated or some similar chain reaction.

As soon as I close ZHA Map things go MUCH quicker.

closed 2020-12-16 by dumpfheimer