zigpy/zigpy-znp #54: ZDOCmd.Unbind_req not implemented (needed to unbind device from coordinator)

TheJulianJES, 2021-01-06:
After https://github.com/home-assistant/core/pull/42854 was merged, Home Assistant can now unbind a device from the Zigbee coordinator. This is required for a Philips Hue Dimmer Switch (RWL021) as they can only be bound to one group/device because of the binding table size being 1.
When trying this using a ZZH with zigpy-znp, the following is displayed in the (debug) logs:
```
2021-01-06 19:51:31 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted a ZDO request: dst_addr=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x577A), dst_ep=0, src_ep=0, cluster=ZDOCmd.Unbind_req, sequence=203, options=TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST, radius=30, data=b'\xcb\xb7\xd6t\x08\x01\x88\x17\x00\x01\x08\x00\x03\xf5\xf3\xa5\xfe\xff\xcc\xcc\xcc\x01'
2021-01-06 19:51:31 ERROR (MainThread) [zigpy_znp.zigbee.application] ZDO converter for cluster ZDOCmd.Unbind_req has not been implemented! Please open a GitHub issue and attach a debug log: https://github.com/zigpy/zigpy-znp/issues/new
2021-01-06 19:51:31 DEBUG (MainThread) [zigpy.zdo] [0x577a:zdo] cluster: 6 Unbind_req --> [cc:cc:cc:ff:fe:a5:f3:f5] failed: No ZDO converter
2021-01-06 19:51:31 DEBUG (MainThread) [zigpy.zdo] [0x577a:zdo] cluster: 8 Unbind_req --> [cc:cc:cc:ff:fe:a5:f3:f5] failed: No ZDO converter
2021-01-06 19:51:31 INFO (MainThread) [homeassistant.components.zha.api] Devices un-bound: source_ieee: [00:17:88:01:08:74:d6:b7] target_ieee: [cc:cc:cc:ff:fe:a5:f3:f5]
```
It looks like the ``ZDOCmd.Unbind_req`` "command" isn't implemented in https://github.com/zigpy/zigpy-znp/blob/dev/zigpy_znp/zigbee/zdo_converters.py.

puddly, 2021-01-06:
I'll add that to the next release. Thanks!

MattWestb, 2021-01-06:
@TheJulianJES If you is having one groupe 0x0000 (the default coordinator group if ZHA have making it) you can trying unbinding it from the dimmer switch and if its working you is losing the event reporting if not the new groupe is having the coordinator added.
It was working most of the time with EZSP before the fix was made.

TheJulianJES, 2021-01-06:
Tried calling ``reset_all`` on the Groups cluster of the dimmer switch. It also didn't seem to work.

MattWestb, 2021-01-06:
I think the coordinator is bonded with device NWK 0x0000 and not with group binding ala Philips HUE. 
In tasmota it  was possible binding group but was not working until sending unbind device 0x0000 then it was start working.

TheJulianJES, 2021-01-06:
I tried simply adding adding this to zdo_converters:
```
    ZDOCmd.Unbind_req: (
        (
            lambda addr, SrcAddress, SrcEndpoint, ClusterID, DstAddress: (
                c.ZDO.UnBindReq.Req(
                    Dst=addr.address,
                    Src=SrcAddress,
                    SrcEndpoint=SrcEndpoint,
                    ClusterId=ClusterID,
                    Address=DstAddress,
                    DstEndpoint=X, # What should be put here?
                )
            )
        ),
```
But zigpy calls this without "giving" the ``DstEndpoint``. However, it seems to be required by zdo.py: ``UnBindReq``.
Should DstEndpoint=SrcEndpoint?

commit 62de24da (2021-01-06): Add ZDO converter for `ZDOCmd.Unbind_req` (fixes #54)
closed 2021-01-06 by puddly