zigpy/zigpy-znp #62: [PSA] Recommend upgrading Z-Stack 3 Coordinator firmware based on Texas Instruments Z-Stack 4.40.00.44 SDK?

Hedda, 2021-02-01:
Please see discussion and copy of PSA about critical bug and new firmware recommendation in https://github.com/zigpy/zigpy/discussions/655

Maybe update readme to recommend upgrade all Z-Stack 3 Coordinators to firmware based on TI Z-Stack SDK 4.40.00.44 or later?

The "develop" tree for Koenkk's Z-Stack-firmware repo now has updated firmware images:

https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

MattWestb, 2021-02-01:
Best is putting on matrix in the Zigpy Wiki https://github.com/zigpy/zigpy/wiki/Coordinator-Firmware-Updates#instructions-for-znp-radios like i have doing for the EZSP coordinators that is known and on the zigpy and zigpy-znp referring to the wiki for the recommended / working firmware in the Wiki.

puddly, 2021-02-01:
Yes, it is very much recommended. I have been reliably able to trigger Z-Stack crashes on older firmwares in about 3-4 horus that are impossible to reset remotely on both the LAUNCHXL boards and on the ZZH. Slaesh's stick should in theory be able to recover from any lockup with the unreleased reconnect watchdog in the `dev` branch. New firmware has been running for over 36 hours now without issues.

Hedda, 2021-02-02:
https://github.com/zigpy/zigpy-znp/blob/dev/README.md

# Hardware requirements
USB-adapters, GPIO-modules, and development-boards running TI's Z-Stack are supported. Reference hardware for this project includes:

 - (**STABLE**) [TI LAUNCHXL-CC26X2R1](https://www.ti.com/tool/LAUNCHXL-CC26X2R1) running [Z-Stack 3 firmware (based on version 4.40.00.44)](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin). You can flash `CC2652R_20210120.hex` using [TI's UNIFLASH](https://www.ti.com/tool/download/UNIFLASH).
 - (**STABLE**) [Electrolama zzh CC2652R](https://electrolama.com/projects/zig-a-zig-ah/) and [Slaesh CC2652R](https://slae.sh/projects/cc2652/) sticks running [Z-Stack 3 firmware (based on version 4.40.00.44)](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin). You can flash `CC2652R_20210120.hex` or `CC2652RB_20210120.hex` respectively using [cc2538-bsl](https://github.com/JelmerT/cc2538-bsl).
 - (**BETA**) CC2531 running [Z-Stack 3.0.1](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.0.x/bin/CC2531_20190425.zip). You can flash `CC2531ZNP-without-SBL.bin` to your stick directly with `zigpy_znp`: `python -m zigpy_znp.tools.flash_write -i /path/to/CC2531ZNP-without-SBL.bin /dev/serial/by-id/YOUR-CC2531` if your stick already has a serial bootloader.
 - (**ALPHA**) CC2531 running [Z-Stack Home 1.2](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20190608.zip). You can flash `CC2531ZNP-Prod.bin` to your stick directly with `zigpy_znp`: `python -m zigpy_znp.tools.flash_write -i /path/to/CC2531ZNP-Prod.bin /dev/serial/by-id/YOUR-CC2531` if your stick already has a serial bootloader.

closed 2021-03-15 by puddly