zigpy/zigpy-znp #67: Autodetect serial ports that has compatible adapter connected to it?

Hedda, 2021-03-01:
Would it be practical for zigpy-znp to have a probe method that scan all available serial ports in order to auto-detect port-path?

puddly posted in https://github.com/zigpy/zigpy-znp/issues/28 slaesh provided information autodetecting at least his adapters:

> @slaesh has sent me all the information I need to autodetect the serial port, although that functionality isn't being used by HA at the moment.

slaesh adapter does however not use the same type of USB-to-serial converter chip as others such as example Electrorama's ZZH.

> I don't think testing is needed for an adapter that is functionally identical to the ZZH with the exception of a much better USB serial chip.

Perhaps that belongs in either the ZHA integration application level, discovery and/or Home Assistant Supervisor level instead?

HA already has several different methods to discover network services but no probes to discover Serial and/or USB adapters?

https://www.home-assistant.io/integrations/discovery/

https://developers.home-assistant.io/docs/supervisor/

https://www.home-assistant.io/integrations/zha/

PS: It would also be very good to be able to probe and discover if an adapter is already "in use" by any other application or not.

puddly, 2021-03-01:
I had autodetection implemented in the past but it was not useful. This is currently done by the `ControllerApplication.probe` classmethod, which returns `True` or `False` when passed a radio configuration with a serial port. It is used by ZHA to autodetect the radio library when you select a serial port in the dropdown.

puddly, 2021-03-01:
For reference, here is the old implementation: https://github.com/zigpy/zigpy-znp/commit/61f8ea7933852b134bbae4d6659261bc5448b2a1

The ZZH using a generic CH340 chip makes it indistinguishable from every other cheap Arduino or dev kit. I think it's up to the upstream application using a radio library to figure out autodetection if it wants to go that route instead of presenting a dropdown of serial ports with model and manufacturer info, which seems pretty user-friendly in my opinion.

closed 2021-03-01 by puddly