zigpy/zigpy-znp #70: Migration CC2531 to CC2652R (zzh) inside LXC

chrismast, 2021-04-10:
He, first of all, great tool.
I was just about to migrate from my CC2531 to a new zzh stick and in time saw your tool that could simplify the migration a lot.
Currently, I am running zigbee2mqtt inside a Proxmox LXC and passed-through my USB devices.
Both show up under lsusb in /dev/ttyACM0 (CC2531) and /dev/ttyUSB0 (CC2652R).
I can also do the initial backup of the CC2531 and the backup file is generated using the following command:

> sudo python3 -m zigpy_znp.tools.network_backup /dev/ttyACM0 -o network_backup.json

Now when I try to restore the backup config to /dev/ttyUSB0 I get an error message though as following (with -v enabled):

> sudo python3 -m zigpy_znp.tools.network_restore /dev/ttyUSB0 -i network_backup.json -v

[image]

Any advice on how to fix this?

puddly, 2021-04-10:
Hmm. Your CC2652R seems to be getting stuck during network formation. Does this happen after you unplug/plug in the stick and try the restore again?

chrismast, 2021-04-10:
> Hmm. Your CC2652R seems to be getting stuck during network formation. Does this happen after you unplug/plug in the stick and try the restore again?

Just tried it as per your comment, did the following:

1) unplug the CC2652R
2) restarted the LXC
3) did a `chmod o+rw /dev/ttyUSB0` on the Proxmox host to set permissions
4) run above command again, same error though

puddly, 2021-04-10:
If it's a noisy RF environment, the stick could be refusing to form a network. I've seen it happen with CC2531s more than the zzh but anything is possible with TI's Z-Stack. Try unscrewing the antenna and try the restore once more?

If that still doesn't work, check over your CC2531 backup. If your network key looks like `01030507090b0d0f00020406080a0c0d`, you don't really have any secret info in the backup other than the IEEE addresses of the devices on your network so if you don't mind posting the backup JSON here, I can just run the restoration locally and upload an NVRAM state that you can restore directly to your stick via `python -m zigpy_znp.tools.nvram_write -i nvram_backup.json /dev/...`. It bypasses network formation and should "just work".

chrismast, 2021-04-10:
> If it's a noisy RF environment, the stick could be refusing to form a network. I've seen it happen with CC2531s more than the zzh but anything is possible with TI's Z-Stack. Try unscrewing the antenna and try the restore once more?
> 
> If that still doesn't work, check over your CC2531 backup. If your network key is `01030507090b0d0f00020406080a0c0d`, you don't really have any secret info in the backup other than the IEEE addresses of the devices on your network (they're only unique, not really "secret") so if you don't mind posting the backup JSON here, I can just run the restoration locally and upload an NVRAM state that you can restore directly to your stick via `python -m zigpy_znp.tools.nvram_write -i nvram_backup.json /dev/...`. It bypasses network formation and should "just work".
> 
> In the future I'm hoping to have zigpy-znp directly set the network up in NVRAM without asking TI's Z-Stack to form it so this won't be an issue. But until then...

That seemed to have done the trick, I unscrewed the antenna and the restore did run through all the way without error.
Adjusted my configuration.yaml, started zigbee2mqtt and did some mqtt commands, all working so far.

Thanks for the help! Will revert if anything else comes up.

closed 2021-04-10 by chrismast

puddly, 2021-04-10:
Glad that worked, I'll make the error message a little more descriptive.