zigpy/zigpy-znp #72: CC2538 being seen as CC2531 and most of the time offline in Home Assistant

glcos, 2021-05-22:
CC2538 module is being seen as CC2531 in Home Assistant and is most of the time showed as offline even though is seems to work fine.
This is what I see in Home Assistant log:

```
2021-05-22 21:15:46 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](CC2531, Z-Stack 3.0.1/3.0.2): Attempting to checkin with device - missed checkins: 1
2021-05-22 21:15:46 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](CC2531, Z-Stack 3.0.1/3.0.2): does not have a mandatory basic cluster
2021-05-22 21:17:04 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](CC2531, Z-Stack 3.0.1/3.0.2): Attempting to checkin with device - missed checkins: 2
2021-05-22 21:17:04 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](CC2531, Z-Stack 3.0.1/3.0.2): does not have a mandatory basic cluster
```

zigpy-znp version 0.5.1

MattWestb, 2021-05-22:
ZHA/zigpy and the radio libs is not knowing the "real ID" of the device only the "family" used by the Zigbee stack protocol.
Its the same for deCONZ is not knowing if its one RapBee I or one ConBee II that the system is "talking" with only that is using the protocol "deCONZ".

I think all radios is virtual devices and is not behaving as normal one and is only updating its online status with some hours interval so if looking in last seen they can being off line most of the time.
I have reading that doing o re scan of network topography is making the last seen and is making it being flagged on line.
Some user have setting the scan interval very short for going around the offline problem.

puddly, 2021-05-22:
The name issue is cosmetic but I will see if I can find a way to distinguish the CC2538 from the CC2531 in the next release. Z-Stack doesn't actually provide any way to query device or really any stack information and I don't have a CC2538 to test with but if you want to help out, can you enable [ZHA debug logging](https://www.home-assistant.io/integrations/zha/#debug-logging) and post all of the lines containing `zigpy_znp` from your `home-assistant.log` file? Your randomly-generated network key will be in the log so you can redact it if you want.

As for the "offline" status, I noticed this when testing with a Silicon Labs coordinator as well. Seems like just a frontend glitch, as zigpy-znp would be logging tons of errors if the coordinator was ever "offline"". @dmulcahey, is there anything special the coordinator `Device` object needs to have other than a node descriptor for ZHA to always show it as online? Maybe set `device.last_seen = time.time()` after successful requests? Or give it that missing `Basic` cluster, which would allow ZHA to poll it periodically?

glcos, 2021-05-22:
This is what I could capture from the log, hope it helps.

```
2021-05-22 22:49:11 DEBUG (MainThread) [zigpy_znp.uart] Connecting to /dev/ttyAMA0 at 115200 baud
2021-05-22 22:49:11 DEBUG (MainThread) [zigpy_znp.uart] Opened /dev/ttyAMA0 serial port
2021-05-22 22:49:11 DEBUG (MainThread) [zigpy_znp.uart] Toggling RTS/CTS to skip CC2652R bootloader
2021-05-22 22:49:12 DEBUG (MainThread) [zigpy_znp.uart] Connected to /dev/ttyAMA0 at 115200 baud
2021-05-22 22:49:12 DEBUG (MainThread) [zigpy_znp.api] Waiting 1s before sending anything
2021-05-22 22:49:13 DEBUG (MainThread) [zigpy_znp.api] Sending bootloader skip byte
2021-05-22 22:49:13 DEBUG (MainThread) [zigpy_znp.api] Waiting 1s or until a reset indication is received
2021-05-22 22:49:14 DEBUG (MainThread) [zigpy_znp.api] Testing connection to /dev/ttyAMA0
2021-05-22 22:49:14 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2021-05-22 22:49:14 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.CAP_APP_CNF|CAP_GP|CAP_APP|CAP_UTIL|CAP_SAPI|CAP_ZDO|CAP_AF|CAP_SYS: 1913>)
2021-05-22 22:49:14 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.NVLength.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=0)
2021-05-22 22:49:14 DEBUG (MainThread) [zigpy_znp.api] Received command: RPCError.CommandNotRecognized.Rsp(ErrorCode=<ErrorCode.InvalidCommandId: 2>, RequestHeader=CommandHeader(id=0x32, subsystem=Subsystem.SYS, type=CommandType.SREQ))
2021-05-22 22:49:14 DEBUG (MainThread) [zigpy_znp.api] Detected Z-Stack 3.0
2021-05-22 22:49:14 DEBUG (MainThread) [zigpy_znp.api] Connected to /dev/ttyAMA0 at 115200 baud
2021-05-22 22:49:14 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVLength.Req(Id=<OsalNvIds.HAS_CONFIGURED_ZSTACK3: 96>)
2021-05-22 22:49:14 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVLength.Rsp(ItemLen=1)
2021-05-22 22:49:14 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVReadExt.Req(Id=<OsalNvIds.HAS_CONFIGURED_ZSTACK3: 96>, Offset=0)
2021-05-22 22:49:14 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVReadExt.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x55')
2021-05-22 22:49:14 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.ResetReq.Req(Type=<ResetType.Soft: 1>)
2021-05-22 22:49:14 DEBUG (MainThread) [zigpy_znp.api] Request has no response, not waiting for one.
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.ResetInd.Callback(Reason=<ResetReason.PowerUp: 0>, TransportRev=2, ProductId=2, MajorRel=2, MinorRel=7, MaintRel=2)
2021-05-22 22:49:15 INFO (MainThread) [zigpy_znp.zigbee.application] ZNP is already configured, not forming a new network
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVLength.Req(Id=<OsalNvIds.LOGICAL_TYPE: 135>)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVLength.Rsp(ItemLen=1)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVReadExt.Req(Id=<OsalNvIds.LOGICAL_TYPE: 135>, Offset=0)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVReadExt.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x00')
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVLength.Req(Id=<OsalNvIds.CONCENTRATOR_ENABLE: 50>)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVLength.Rsp(ItemLen=1)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVReadExt.Req(Id=<OsalNvIds.CONCENTRATOR_ENABLE: 50>, Offset=0)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVReadExt.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x01')
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVLength.Req(Id=<OsalNvIds.CONCENTRATOR_DISCOVERY: 51>)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVLength.Rsp(ItemLen=1)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVReadExt.Req(Id=<OsalNvIds.CONCENTRATOR_DISCOVERY: 51>, Offset=0)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVReadExt.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x78')
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVLength.Req(Id=<OsalNvIds.CONCENTRATOR_RC: 54>)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVLength.Rsp(ItemLen=1)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVReadExt.Req(Id=<OsalNvIds.CONCENTRATOR_RC: 54>, Offset=0)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVReadExt.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x01')
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVLength.Req(Id=<OsalNvIds.SRC_RTG_EXPIRY_TIME: 56>)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVLength.Rsp(ItemLen=1)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVReadExt.Req(Id=<OsalNvIds.SRC_RTG_EXPIRY_TIME: 56>, Offset=0)
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVReadExt.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\xFF')
2021-05-22 22:49:15 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVLength.Req(Id=<OsalNvIds.NWK_CHILD_AGE_ENABLE: 62>)
[... cut ...] ius=30, SourceRoute=[], Data=b'\x00\x02\x00\x00\x00')
2021-05-22 22:49:23 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestSrcRtg.Rsp(Status=<Status.SUCCESS: 0>)
2021-05-22 22:49:23 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.MAC_NO_ACK: 233>, Endpoint=1, TSN=2)
2021-05-22 22:49:23 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.MAC_NO_ACK: 233>), retry attempt 5 of 5
```

glcos, 2021-06-02:
Upgraded application.py file to latest version and now the coordinator is showing CC2538, Great job!
Can't wait for the latest release.

glcos, 2021-06-08:
I've been monitoring my system for a while, it's working fine, however I still see this warning:

```
2021-06-08 07:52:53 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](CC2538, Z-Stack 3.0.x (build 20201010)): Attempting to checkin with device - missed checkins: 1
2021-06-08 07:52:53 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](CC2538, Z-Stack 3.0.x (build 20201010)): does not have a mandatory basic cluster
2021-06-08 07:54:08 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](CC2538, Z-Stack 3.0.x (build 20201010)): Attempting to checkin with device - missed checkins: 2
2021-06-08 07:54:08 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](CC2538, Z-Stack 3.0.x (build 20201010)): does not have a mandatory basic cluster
```

Could it be something related to the firmware I have flashed to the coordinator ?

puddly, 2021-06-08:
This is related to the ZHA issue I mentioned earlier, where ZHA is treating the coordinator like a normal device and it attempts to poll it to make sure it's still alive after the coordinator doesn't "check in" for a while. I don't think this is a zigpy-znp problem but a ZHA problem. The coordinator should always show "online" in the UI (or that status should be removed for it).

puddly, 2021-10-30:
Fixed by https://github.com/zigpy/zigpy-znp/pull/93.

closed 2021-10-30 by puddly