zigpy/zigpy-znp #75: Unhandled Commands Warnings in Home Assistant Core

fraserp, 2021-06-20:
I've recently set up a new clean instance of Home Assistant with an Electrolama zzh! (TI CC2652R) plugged in to a RPi4 but am receiving a fairly solid stream of unhandled command warnings (~1 per minute):

```
Logger: zigpy_znp.api
Source: /usr/local/lib/python3.8/site-packages/zigpy_znp/api.py:395
First occurred: 10:30:35 (27 occurrences)
Last logged: 10:54:44

Received an unhandled command: ZDO.ParentAnnceRsp.Callback(Src=0xE6FA, Status=<Status.SUCCESS: 0>, ChildInfo=[00:12:4b:00:1f:54:22:77, 00:12:4b:00:1f:54:4c:80])
Received an unhandled command: ZDO.ConcentratorInd.Callback(NWK=0xF0DD, IEEE=00:1e:5e:09:02:13:2a:28, PktCost=3)
Received an unhandled command: ZDO.ConcentratorInd.Callback(NWK=0xF0DD, IEEE=00:1e:5e:09:02:13:2a:28, PktCost=1)
Received an unhandled command: ZDO.MgmtNWKUpdateNotify.Callback(Src=0xE6FA, Status=<Status.SUCCESS: 0>, ScannedChannels=<Channels.ALL_CHANNELS: 134215680>, TotalTransmissions=24, TransmissionFailures=7, EnergyValues=[167, 183, 162, 160, 170, 162, 173, 166, 174, 162, 164, 194, 182, 181, 169, 183])
```

Please advise if this is a problem with the configuration, or 'expected' behaviour?

puddly, 2021-06-20:
Out of curiosity, what brands do you have on your network? I get maybe *one* warning per week, never more, so seeing them this frequently is unexpected but harmless.

These messages are not used by zigpy or zigpy-znp so in the next release I will just silence these warnings.

MattWestb, 2021-06-20:
`ParentAnnce` is one IKEA killer but its looks like they is looking on that problems if reading on reddit (need EZSP 6.7.7.0 for fixing it or patching lower version).

I was loosing 3 GU10 WS and one plug this week on IKEA GW that was working with the remotes but was off line in the GW and coming back nicely after re power them.

fraserp, 2021-06-20:
Updated, as its been running all afternoon...

```
Logger: zigpy_znp.api
Source: /usr/local/lib/python3.8/site-packages/zigpy_znp/api.py:395
First occurred: 10:30:35 (564 occurrences)
Last logged: 19:50:34

Received an unhandled command: ZDO.ConcentratorInd.Callback(NWK=0xF0DD, IEEE=00:1e:5e:09:02:13:2a:28, PktCost=3)
Received an unhandled command: ZDO.ConcentratorInd.Callback(NWK=0xF0DD, IEEE=00:1e:5e:09:02:13:2a:28, PktCost=1)
Received an unhandled command: ZDO.MgmtNWKUpdateNotify.Callback(Src=0xE6FA, Status=<Status.SUCCESS: 0>, ScannedChannels=<Channels.ALL_CHANNELS: 134215680>, TotalTransmissions=24, TransmissionFailures=7, EnergyValues=[167, 183, 162, 160, 170, 162, 173, 166, 174, 162, 164, 194, 182, 181, 169, 183])
Received an unhandled command: ZDO.MgmtNWKUpdateNotify.Callback(Src=0xF0DD, Status=<Status.SUCCESS: 0>, ScannedChannels=<Channels.ALL_CHANNELS: 134215680>, TotalTransmissions=31, TransmissionFailures=8, EnergyValues=[170, 169, 172, 169, 177, 171, 173, 174, 170, 170, 209, 179, 200, 204, 169, 169])
Received an unhandled command: ZDO.ConcentratorInd.Callback(NWK=0xF0DD, IEEE=00:1e:5e:09:02:13:2a:28, PktCost=5)
```

00:1e:5e:09:02:13:2a:28 = Computime/Salus SP600 Smart Socket - this one is the one thats generating a warning a minute, the others are much less frequent
00:12:4b:00:1f:54:22:77 & 00:12:4b:00:1f:54:4c:80= eWeLink/Sonoff TH01 Temp & Humidity Sensor (I have 4 of these total)

Broadly I have:
1 Aqara water leak sensor
1 Ikea bulb
1 ikea switch
1 Salus Sp600 socket
5 ewelink sockets (bot al are inactive currently, so offline)
4 sonoff temp & humidity sensors
2 sonoff door sensors
1 sonoff button
1 sonoff motion sensor
1 sonoff switch
plus the zzh coordinator

MattWestb, 2021-06-20:
@puddly Is ZNP doing energy scan then the network it up and running ??
That is not allowed in the later ZB3 specs then its blocking / disconnecting part of the network and can making (source)router not working or the mesh is collapsing.

But i like that  and also moving from one bad channel then its possible (but that is not working with Xiaomi devices).

Back to the spamming:
- ZDO.ConcentratorInd.Callback = (source)routing feedback = OK also if being many in short time (shall being hidden in normal logging and over 500 in 10 hours = 1 / minute).
- ZDO.ParentAnnceRsp.Callback = one device have coming back to the network after re power or reboot = OK is happens some time in the week if many time in the day its something very wrong.
- ZDO.MgmtNWKUpdateNotify.Callback if its energy scan its making problems with the routing = not OK

closed 2021-06-21 by puddly

puddly, 2021-06-21:
> Is ZNP doing energy scan then the network it up and running ??

zigpy-znp definitely isn't but I'll have to set a sniffer up to check if Z-Stack itself is. I don't see any reason why it would need to.

fraserp, 2021-06-21:
Thanks for the quick responses  :)

commit 6ce31d72 (2021-06-21): Do not log warnings for frequently unhandled commands (fixes #75)