zigpy/zigpy-znp #77: Is there an zigpy-znp API

alan-rpi, 2021-07-10:
Hi. Looks very useful. Is there a published API so that zigpy-znp can be used by a python program to communicate with a CC2531 running as a client/router to send data via zigbee to a coordinator  -> zigbee2mqtt -> mqtt broker -> recipient. Is this feasible?

I would imagine APIs would be needed to 1) set the zigbee identifier, 2) send a topic and message, 3) receive topics and messages. 

Thanks, Alan

MattWestb, 2021-07-10:
Its depends wot for continuations you like sending from your client. 

It gets some DIY CC-253X firmware that can sending status to / from GIPOs or UART communication to / from the coordinator.

I think (without knowing) that is gets no "MQTT over Zigbee" protocol in Zigbee standard but can being implanted _Commercial Building Automation_ profile in Zigbee 3.

Large data streams (then around 56 kb) is not recommended then the underplaying network is limited to around 200 kb then its needed more bandwidth its better with BT mesh or WiFi.

puddly, 2021-07-10:
> Is there an zigpy-znp API

There's no autogenerated documentation but just about everything is completely documented in the source and the API is pretty stable at this point.

zigpy-znp implements the complete [Z-Stack Monitor and Test API](https://github.com/zigpy/zigpy-znp/blob/dev/docs/Z-Stack%20Monitor%20and%20Test%20API.pdf) and includes helper functions to make working with it much easier. Anything you can do with that, you can do with the `zigpy_znp.api:ZNP` class. `zigpy_znp.zigbee.application:ControllerApplication` handles high-level network operations like formation, channel changing, sending data with retries, etc.

For some examples, take a look at the `tools` folder: https://github.com/zigpy/zigpy-znp/blob/dev/zigpy_znp/tools

>  Is there a published API so that zigpy-znp can be used by a python program to communicate with a CC2531 running as a client/router to send data via zigbee to a coordinator

Joining an existing Zigbee as a router isn't something I have ever tried to do but it's definitely possible (I think even with unmodified coordinator firmware). You can also send inter-PAN data, but this would require modifying Z2M for it to work.

alan-rpi, 2021-07-10:
Thanks guys for a very quick and helpful response. 

I have tried to use Raspi with Python -> Mosquitto -> zibbee2MQTT -> CC2531 but could not get the CC2531 to work as a router (no serial port visible to zibbee2MQTT ). I managed (accidentally!) to flash a CC2531 with co-ordinator firmware and it did "work" (serial port visible) and it was searching for devices but all my devices were already paired with my main zigbee co-ordinator.

I've been reading Q&As on the TI site and setting PAN ids seems a possibility (zibbee2MQTT has a PAN configuration setting). I could then pair a particular device with this 2nd zigbee network but I've yet to find an off the shelf Zigbee Bridge to link the 2 networks. 

I would prefer a single network - let's hope I can solve the Koenkk router firmware problem. 

Meanwhile I will look at zigpy tools (a quick glance shows it's a bit technical for my currently level of understanding). I've seen some other python frameworks but they rely on Koenkk' firmware.

Alan

puddly, 2021-07-10:
I believe the only serial port the router firmware exposes is just for debugging output with the debug build, you can't control the router itself aside from resetting it (what else would there be to control?).

What are you trying to accomplish that would require sending data from a second computer with a CC2531 to another Zigbee network that you also control? If they're both on the same Zigbee network would it not be easier to send the data some other way (or if you're using Z2M, just publish to the appropriate MQTT topics)?

alan-rpi, 2021-07-11:
Hi puddly
>What are you trying to accomplish that would require sending data from a second computer with a CC2531 to another Zigbee network that you also control? 
Yes
>If they're both on the same Zigbee network would it not be easier to send the data some other way (or if you're using Z2M, just publish to the appropriate MQTT topics)?

The 2nd computer (a Pi Zero) will not be connected to the internet as it is located in the electricity supply cabinet and it has connections with the 3 meters (PV gen, consumption, purchase) and is only protected with a 100A master fuse!

My plan (in hope, as I got no feedback on the Z2M forum) was to go (Pi Zero) Python App -> Mosquitto -> Z2M -> CC2531 (end device or router) -> Zigbee network -> Zigbee Immersion Switch. 

At the moment the CC2531 Router firmware is not working for me. Sees a bit overkill - a Python App -> CC2531 -> Zigbee Immersion Switch would be more lightweight. Schematic below.

I would also like to do (Pi Zero) Python App -> Mosquitto -> Z2M -> CC2531 -> Zigbee Network -> existing Pi4 with CC2531 (Coordinator) -> Z2M -> Mosquitto -> a) InfluxDB and Grafana b) Python app with Alerts sent to Email/SMS/Telegram.  On the Pi4 I'm running Domoticz.

I would have thought there would be plenty of examples of people using a small microprocessor (e.g. Pi Zero or Pico) and python (or microPython) to build their own internet sensor that would be a end device, but I've yet to find any examples.

Thanks for your interest.
Alan
[image]
I've written most of the PiZero code using a Pi simulator running on a Windows PC. Screen shot below shows in red box the simulated user interface for the system and in blue the input panel to simulate the 3 electricity meters.
[image]

MattWestb, 2021-07-11:
Then only sending "normal sensor data" i think its easiest using one firmware that can sending text strings and you is getting MQTT messages in the other end with Z2M like the  Russian have doing in there DIY multi sensor firmware for TI CC-253X.
https://ptvo.info/zigbee-switch-configurable-firmware-v2-210/

From the read.me in the firmware package: 
```
UART
=========================

The UART interface is designed to send text commands between an external device (e.g. Arduino) and a host. 

The maximum command size is 127 characters. Every command line should ends with CR (0x0D). If you need 
to send a number to a host, you should convert it to text.

UART uses P02 and P03 pins (3.3V). You should enable UART on one of output pins in the configuration.

UART parameters: 9600 8N1. You cannot change these parameters.
```

Going with IT-CC253X i cant recommending then its one very obsolete hardware and its not supporting  Zigbee 3.

If going with one more modern hardware like Silabs EFR32MG you can designing one complete Zigbee device as you like and using BAC profile for your device  and putting  in all cluster (sensor types) you like and can being supported of all normal open Zigbee systems but its costing 100 $ for one license plus hardware and time but then you is making on "real Zigbee sensor".

Its also possible making it as one Zigbee Green power device but then you must doing much more software work and need extra licenses for the encryption and so on.

alan-rpi, 2021-07-11:
HaHa! My first attempt over a year ago used PTVO and the Counter function but I could not get it to work - I tried a simple push button to make the pulses. I could get the internal functions like temperature to work. I send a load of info to the developer and all he said was "It works for me". 

I still have the CC2530 (with the amplifier). I'll look at the UART/Text option. .

The other option that keeps coming up is XBee but I guess that means 2 of them at £20 UK (c$20) each. I could flash 1 with End Device code and the other with Co-ordinator code with a different PAN. I have another Pi 0 in production that could be the receiver and then send MQTT messages over an Ethernet cable link to the Pi4. 

Many thanks

Hedda, 2021-11-03:
@alan-rpi You might also be interested in the new somewhat realted discussion here -> https://github.com/zigpy/zigpy/issues/842

As well as be aware of zigpy-cli -> https://github.com/zigpy/zigpy-cli