zigpy/zigpy-znp #81: CC2538 + CC2592 set TX Power

wes1993, 2021-07-22:
Hello to everyone,
i have found a BUG i think because i can't set the TX power of my CC2538 + CC2592 dongle.
The log here:
```
2021-07-22 23:12:29 DEBUG (MainThread) [zigpy_znp.nvram] Read NVRAM["LEGACY"][0x008f] = <Bool.true: 1>
2021-07-22 23:12:29 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.SetTxPower.Req(TXPower=19)
2021-07-22 23:12:29 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.SetTxPower.Rsp(Status=<Status.MAC_MEM_ERROR: 19>)
2021-07-22 23:12:29 DEBUG (MainThread) [zigpy_znp.uart] Closing serial port
2021-07-22 23:12:29 ERROR (MainThread) [zigpy.application] Couldn't start application
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 74, in new
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 197, in startup
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 263, in _startup
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 532, in request
zigpy_znp.exceptions.InvalidCommandResponse: Expected SRSP response SYS.SetTxPower.Rsp(Status=<Status.SUCCESS: 0>), got SYS.SetTxPower.Rsp(Status=<Status.MAC_MEM_ERROR: 19>)
```

I hope someone could help
Best Regards
Stefano

puddly, 2021-07-23:
Hmm. It looks like in only newer Z-Stack versions the return value is either `SUCCESS` or `INVALID_PARAMETER`. In older Z-Stack releases, the return value is the closest valid power level to the supplied one.

puddly, 2021-07-23:
Can you test out #82? You can follow the [**Testing `dev`** instructions](https://github.com/zigpy/zigpy-znp#testing-dev-with-home-assistant-core) in the README but use the following URL to install the correct branch:

```
git+https://github.com/puddly/zigpy-znp.git@puddly/cc2538-tx-power
```

wes1993, 2021-07-23:
@puddly Hello and thanks  a lot fo your reply,
Sorry for my delay in reply but i'm in italy :-( 
Works well now:
```
2021-07-23 10:15:47 DEBUG (MainThread) [zigpy_znp.nvram] Read NVRAM["LEGACY"][0x008f] = <Bool.true: 1>
2021-07-23 10:15:47 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.SetTxPower.Req(TXPower=19)
2021-07-23 10:15:47 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.SetTxPower.Rsp(StatusOrPower=19)
```

Another question, based on the datasheet is correct that the range is what is inside this image?:
https://user-images.githubusercontent.com/919914/77421979-cf386100-6ddd-11ea-9d30-b17e2627892a.png

For example i could chose 22 or 21.5 etc?

Best Regards
Stefano

puddly, 2021-07-23:
Here's what's in the Z-Stack source code:

```C
#if defined HAL_PA_LNA_CC2592 || defined MAC_RUNTIME_CC2592
const uint8 CODE macRadioDefsTxPwrCC2592[] =
{
  21,  /* tramsmit power level of the first entry */
  (uint8)(int8)10, /* transmit power level of the last entry */
  /*  21 dBm */   0xF4,   /* characterized as 21.47 dBm */
  /*  20 dBm */   0xD4,   /* characterized as 20.26 dBm */
  /*  19 dBm */   0xC0,   /* characterized as 19.31 dBm */
  /*  18 dBm */   0xA5,   /* characterized as 18.21 dBm */
  /*  17 dBm */   0x95,   /* characterized as 17.14 dBm */
  /*  16 dBm */   0x82,   /* characterized as 16.27 dBm */
  /*  15 dBm */   0x72,   /* characterized as 15.15 dBm */
  /*  14 dBm */   0x63,   /* characterized as 14.12 dBm */
  /*  13 dBm */   0x61,   /* characterized as 13.24 dBm */
  /*  12 dBm */   0x55,   /* characterized as 11.93 dBm */
  /*  11 dBm */   0x50,   /* characterized as 10.90 dBm */
  /*  10 dBm */   0x49    /* characterized as 10.29 dBm */
};
#endif
```

If you provide a value of `22`, the radio will reduce it to `21` and zigpy-znp will (hopefully) print a warning.

puddly, 2021-07-23:
As for that logged exception, could you post the `zigpy_znp` debug log lines from `10:20:00` until `10:24:16`?

wes1993, 2021-07-23:
Unfortunately I can’t send you the log now because I’m out.
Another question, based on the datasheet the TXPower value are different from what is inside the source code, is this normal?

For example, 20 dBm is 0xC5 in the source code is 0xD4.
Best regards
Stefano

puddly, 2021-07-23:
Your firmware internally translates the numerical power level (`10` to `21`) into the appropriate register value (`0x49` to `0xF4`) so it's not something you really need to worry about or can directly control with the serial protocol (from what I understand).

closed 2021-07-23 by puddly

wes1993, 2021-07-24:
Thanks a lot Puddly for all you help and support!! I’m new to this world :(.

When I will come back to my home I will send you the log sorry but I’m out of home for 2 days

wes1993, 2021-07-25:
Hello again @puddly,
Here are the logs:
HomeAssistant 
```
Logger: homeassistant
Source: deps/lib/python3.9/site-packages/zigpy_znp/api.py:435
First occurred: 19:07:22 (3 occurrences)
Last logged: 19:07:48

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/deps/lib/python3.9/site-packages/zigpy_znp/api.py", line 435, in request_callback_rsp
    return await callback_rsp
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/deps/lib/python3.9/site-packages/zigpy_znp/api.py", line 435, in request_callback_rsp
    return await callback_rsp
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 55, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 92, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```
Zigpy-ZNP log:
```
2021-07-25 19:06:06 DEBUG (MainThread) [zigpy_znp.uart] Connecting to /dev/serial/by-id/usb-Texas_Instruments_CC2538_USB_CDC-if00 at 115200 baud
2021-07-25 19:06:06 DEBUG (MainThread) [zigpy_znp.uart] Opened /dev/serial/by-id/usb-Texas_Instruments_CC2538_USB_CDC-if00 serial port
2021-07-25 19:06:06 DEBUG (MainThread) [zigpy_znp.uart] Toggling RTS/CTS to skip CC2652R bootloader
2021-07-25 19:06:06 DEBUG (MainThread) [zigpy_znp.uart] Connected to /dev/serial/by-id/usb-Texas_Instruments_CC2538_USB_CDC-if00 at 115200 baud
2021-07-25 19:06:06 DEBUG (MainThread) [zigpy_znp.api] Waiting 1s before sending anything
2021-07-25 19:06:07 DEBUG (MainThread) [zigpy_znp.api] Sending bootloader skip byte
2021-07-25 19:06:07 DEBUG (MainThread) [zigpy_znp.api] Waiting 1s or until a reset indication is received
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Testing connection to /dev/serial/by-id/usb-Texas_Instruments_CC2538_USB_CDC-if00
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|APP|UTIL|SAPI|ZDO|AF|SYS: 1913>)
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVLength.Req(Id=<OsalNvIds.NWKKEY: 130>)
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVLength.Rsp(ItemLen=24)
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVReadExt.Req(Id=<OsalNvIds.NWKKEY: 130>, Offset=0)
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVReadExt.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x00\x01\x03\x05\x07\x09\x0B\x0D\x0F\x00\x02\x04\x06\x08\x0A\x0C\x0D\x03\x00\x20\xAD\xDF\x3F\x00')
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.nvram] Read NVRAM["LEGACY"][0x0082] = b'\x00\x01\x03\x05\x07\x09\x0B\x0D\x0F\x00\x02\x04\x06\x08\x0A\x0C\x0D\x03\x00\x20\xAD\xDF\x3F\x00'
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.nvram] Detected struct alignment: True
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.NVLength.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=0)
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Received command: RPCError.CommandNotRecognized.Rsp(ErrorCode=<ErrorCode.InvalidCommandId: 2>, RequestHeader=CommandHeader(id=0x32, subsystem=Subsystem.SYS, type=CommandType.SREQ))
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Detected Z-Stack 3.0
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Connected to /dev/serial/by-id/usb-Texas_Instruments_CC2538_USB_CDC-if00 at 115200 baud
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVLength.Req(Id=<OsalNvIds.HAS_CONFIGURED_ZSTACK3: 96>)
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVLength.Rsp(ItemLen=1)
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVReadExt.Req(Id=<OsalNvIds.HAS_CONFIGURED_ZSTACK3: 96>, Offset=0)
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVReadExt.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x55')
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.nvram] Read NVRAM["LEGACY"][0x0060] = 85
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.ResetReq.Req(Type=<ResetType.Soft: 1>)
2021-07-25 19:06:08 DEBUG (MainThread) [zigpy_znp.api] Request has no response, not waiting for one.
2021-07-25 19:06:11 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.ResetInd.Callback(Reason=<ResetReason.PowerUp: 0>, TransportRev=2, ProductId=2, MajorRel=2, MinorRel=7, MaintRel=2)
2021-07-25 19:06:11 INFO (MainThread) [zigpy_znp.zigbee.application] ZNP is already configured, not forming a new network
2021-07-25 19:06:11 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVLength.Req(Id=<OsalNvIds.LOGICAL_TYPE: 135>)
2021-07-25 19:06:11 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVLength.Rsp(ItemLen=1)
2021-07-25 19:06:11 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVReadExt.Req(Id=<OsalNvIds.LOGICAL_TYPE: 135>, Offset=0)
2021-07-25 19:06:11 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVReadExt.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x00')
2021-07-25 19:06:11 DEBUG (MainThread) [zigpy_znp.nvram] Read NVRAM["LEGACY"][0x0087] = <DeviceLogicalType.Coordinator: 0>
[... cut ...]
2021-07-25 19:07:19 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2021-07-25 19:07:19 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=81)
2021-07-25 19:07:19 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0x1CB1, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=73, SecurityUse=<Bool.false: 0>, TimeStamp=220451, TSN=0, Data=b'\x08\x

puddly, 2021-07-25:
It appears that close to 30 devices are unreachable at startup, with over 32 requests happening concurrently. Do these devices normally work? I suspect it's a timeout unexpectedly elapsing due to requests taking too long to send.

As for the database warning, you can ignore it, it's nothing to worry about. This is a bug that was fixed in zigpy.

wes1993, 2021-07-25:
From what i can see all devices works.
I don't know how can i test this devices.
DO you think that pair again everything solve this issue?

Best Regards
Stefano

PR #82 (2021-07-23): Properly handle older Z-Stack `SYS.SetTxPower` responses
Older Z-Stack releases reply to `SYS.SetTxPower.Req` with the set power level instead of a `SUCCESS` / `INVALID_PARAMETER` status code.

Fixes #81.