zigpy/zigpy-znp #83: Danalock v3 sometimes becomes unavailable after lock.lock / lock.unlock service calls

paulius2k, 2021-08-17:
Hello,

_sometimes_, after `lock.lock` / `lock.unlock` service calls,  my Danalock V3 becomes unavailable in the ZB network . 
Meanwhile the lock keeps on working fine using the Danalock iOS app, which communicates via Bluetooth.

I am using an Electrolama zzh! stick with Texas Instruments Z-Stack ZNP protocol: CC253x, CC26x2, CC13x2, Z-Stack 3.30+ (build 20210120).

The only resolution I found is to restart the lock (batteries out--in). It re-joins the network and operates fine until the next malfunction.

As an example, after this service call the lock became not reachable with `last_seen` being 22:46:22:
```
2021-08-17 22:46:17 DEBUG (MainThread) [zigpy.device] [0xfd59] Extending timeout for 0x13 request
2021-08-17 22:46:17 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xFD59), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=257, TSN=19, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x01\x13\x00')
2021-08-17 22:46:17 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2021-08-17 22:46:20 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.APS_NO_ACK: 183>, Endpoint=1, TSN=19)
2021-08-17 22:46:20 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.APS_NO_ACK: 183>), retry attempt 1 of 5
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xFD59), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=257, TSN=19, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x01\x13\x00')
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xFD59, Relays=[0xC6F4, 0x015B])
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xFD59, Relays=[0xC6F4, 0x015B])
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xFD59, Relays=[0xC6F4, 0x015B])
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xFD59, Relays=[0xC6F4, 0x015B])
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xFD59, Relays=[0xC6F4, 0x015B])
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=257, SrcAddr=0xFD59, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=48, SecurityUse=<Bool.false: 0>, TimeStamp=3997910, TSN=0, Data=b'\x09\x13\x00\x00', MacSrcAddr=0x015B, MsgResultRadius=27)
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy.zcl] [0xfd59:1:0x0101] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=19 command_id=0>
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=257, SrcAddr=0xFD59, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=51, SecurityUse=<Bool.false: 0>, TimeStamp=3998791, TSN=0, Data=b'\x09\x13\x00\x00', MacSrcAddr=0x015B, MsgResultRadius=27)
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=257, SrcAddr=0xFD59, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=51, SecurityUse=<Bool.false: 0>, TimeStamp=3999763, TSN=0, Data=b'\x09\x13\x00\x00', MacSrcAddr=0x015B, MsgResultRadius=27)
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy.zcl] [0xfd59:1:0x0101] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=19 command_id=0>
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy.device] [0xfd59] Invalid state on future for 0x13 seq -- probably duplicate response
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy.zcl] [0xfd59:1:0x0101] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=19 command_id=0>
2021-08-17 22:46:22 DEBUG (MainThread) [zigpy.device] [0xfd59] Invalid state on future for 0x13 seq -- probably duplicate response
2021-08-17 22:46:25 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.APS_NO_ACK: 183>, Endpoint=1, TSN=19)
2021-08-17 22:46:25 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.APS_NO_ACK: 183>), retry attempt 2 of 5
2021-08-17 22:46:25 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestSrcRtg.Req(DstAddr=0xFD59, DstEndpoint=1, SrcEndpoint=1, ClusterId=257, TSN=19, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, SourceRoute=[0x015B, 0xC6F4], Data=b'\x01\x13\x00')
2021-08-17 22:46:25 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestSrcRtg.Rsp(Status=<Status.SUCCESS: 0>)
2021-08-17 22:46:27 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2021-08-17 22:46:27 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
2021-08-17 22:46:28 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.APS_NO_ACK: 183>, Endpoint=1, TSN=19)
2021-08-17 22:46:29 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteDisc.Req(Dst=0xFD59, Options=<RouteDiscoveryOptions.UNICAST: 0>, Radius=30)
2021-08-17 22:46:29 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteDisc.Rsp(Status=<Status.SUCCESS: 0>)
2021-08-17 22:46:30 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.APS_NO_ACK: 183>), retry attempt 3 of 5
2021-08-17 22:46:30 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xFD59), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=257, TSN=19, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x01\x13\x00')
2021-08-17 22:46:30 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2021-08-17 22:46:34 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.APS_NO_ACK: 183>, Endpoint=1, TSN=19)
2021-08-17 22:46:34 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.APS_NO_ACK: 183>), retry attempt 4 of 5
2021-08-17 22:46:34 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.IEEE: 3>, address=58:8e:81:ff:fe:43:d3:4f), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=257, TSN=19, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x01\x13\x00')
2021-08-17 22:46:34 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2021-08-17 22:46:36 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xF142, Relays=[0x015B])
2021-08-17 22:46:36 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1026, SrcAddr=0xF142, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=48, SecurityUse=<Bool.false: 0>, TimeStamp=4879456, TSN=0, Data=b'\x18\x61\x0A\x00\x00\x29\x50\x06', MacSrcAddr=0x015B, MsgResultRadius=28)
2021-08-17 22:46:36 DEBUG (MainThread) [zigpy.zcl] [0xf142:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=97 command_id=Command.Report_Attributes>
2021-08-17 22:46:36 DEBUG (MainThread) [zigpy.zcl] [0xf142:1:0x0402] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=1616>)]]
[... cut ...]
2021-08-17 22:46:37 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.APS_NO_ACK: 183>), retry attempt 5 of 5
2021-08-17 22:46:38 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xFD59:1:0x0101]: command failed: 'lock_door' args: '()' kwargs '{}' exception: 'Request failed after 5 attempts: <Status.APS_NO_ACK: 183>'
2021-08-17 22:46:38 ERROR (MainThread) [homeassistant.components.zha.entity] lock.danalock_v3_door_lock: Error with lock_door: Request failed after 5 attempts: <Status.APS_NO_ACK: 183>
``` 

Could anybody help to find what is the matter here?  
Big thank you.

LaurisGale, 2021-12-18:
Absolutely same issue here with the same logs. I put the lock near the coordinator to see if it is the link quality. But after few lock/unlock actions from HA the lock simply stops responding with error <Status.APS_NO_ACK: 183>. Only battery out-in solution gets it back to network. Bluetooth, works fine all the time, though. 

- Home Assistant OS 7.0
- Version core-2021.12.3
- supervisor-2021.12.2
- Texas Instruments Z-Stack ZNP protocol CC2652R_coordinator_20210708

puddly, 2021-12-18:
Can you post the "Zigbee device signature" of your lock?

LaurisGale, 2021-12-18:
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4444, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x000a",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0020",
        "0x0101",
        "0x0b05",
        "0xfcff"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Danalock",
  "model": "V3-BTZBE",
  "class": "zigpy.device.Device"
}`

paulius2k, 2021-12-19:
Update from my side:
I have communicated with Danalock support back in September.
This is what they wrote:

> A quick fix if Danalock goes offline in the Zigbee setup is to remove one battery and re-insert. We are investigating why the lock goes offline and what can be done about it. 
> This issue is mainly seen in setup with Home Assistant and Deconz. Adding the lock to a commercial hub like e.g. Smartthings the lock does not show the same behavior.
> 
> 
> Conclusion: We will keep your lock for further testing and send you a complete new one. The new lock will most likely not solve any of the above issues, but we will continue to investigate. 

I asked that they replace my ZB version with a Z-wave one.
I have not experienced the abovementioned issue with the Z-wave version.

puddly, 2021-12-19:
The device reports `*is_receiver_on_when_idle=False` so it's going to be asleep 99% of the time and will only periodically wake up to poll its parent for new packets. If it does not, there's really nothing zigpy-znp can do about it.

If the lock is sometimes *crashing* when it's locked/unlocked, this seems like a bug with their firmware. It'll be difficult to diagnose this issue without a detailed Zigbee traffic capture of its communication with a "stable" hub in addition to the ZHA+zigpy-znp debug logs.

LaurisGale, 2021-12-19:
Thanks guys! This released me from a week of headache figuring out my setup and additional few worthless days of changing my coordinator to Conbee and trying both Zigbe2mqtt and ZHA on it that I was planning to do next week (having in mind all 30 ZB devices in my net).. 

Just wonder whether Danalock knows the sleep  along with the crash issues? 

Though i have paired the lock from scratch in the morning and left it for the whole day today without triggering it from HA. To my surprise I yet see it still reports the status, but I know if trigger it from HA or BT app couple of times, it will loose Zigbee network and I will need to pul the batteries again.

Still hope for a Christmas present from Danalock in a form of new FW, but that might end as yet another unfulfilled hope.. 

I'm on a Danalock FW 0.17.15 for V3-BTZBE, to keep this as a log for the chain.

puddly, 2021-12-19:
> I know if trigger it from [...] BT app couple of times, it will loose Zigbee network and I will need to pul the batteries again.

Do you mean that you can break the lock's Zigbee functionality by only using the manufacturer's app over Bluetooth?

LaurisGale, 2021-12-20:
> 

My mistake,indeed. Triggering the lock only from BT app does not crash the lock. It also reports zigbee status correctly and does no go offline for last 24hrs already.

paulius2k, 2022-03-28:
closing as not active

closed 2022-03-28 by paulius2k