zigpy/zigpy-znp #96: [Bug] Devices are not rediscovered after Zigbee Coordinator Power Outage

aleck011mi, 2021-11-08:
I have a network based Zigbee Coordinator (tubes Ethernet Coordinator https://www.tubeszb.com/product/cc2652_coordinator/1). Which works great with ZHA on Home Assistant. I can reboot or turn off Home Assistant for as long as I need and all devices will be there when it is turned back on. However, if power is lost to the coordinator for any significant period of time (over 5 minutes or so), almost certainly at least one of my 20 devices will drop offline. The problem is when I attempt to re-pair the device and ZHA seems to completely ignore the device despite being very close to the coordinator and in pairing mode. The only way I seem to be able to pair the device again is by factory resetting the coordinator, losing all my devices and causing me to need to repair them. After taking to tube, he recommended I open a bug report here: https://community.home-assistant.io/t/tubes-zb-coordinators-and-routers-was-zigbee-router-on-steroids/280896/445?u=aleck011mi

Thanks!

MattWestb, 2021-11-08:
This is the right place for the issue but it tricky one.

If looking on Z2M issues they is more that is having more or less the same problem: Problem pairing devices after some time or after restart of the TI coordinator but i dont have one (only CC-2531 for testing and sniffing) so i cant testing how is working but many user have doing it in Z2M.

I hope some devs can triggering the behavior and also finding one was fixing it but i think its in the firmware of the TI coodinator.

aleck011mi, 2021-11-08:
Thanks for the confirmation Matt. If there is any logs or other information I can provide you or anyone else, let me know which and how to get them and I'll be happy to do so.

puddly, 2021-11-08:
Hmm. I've had power outages last more than 48 hours and my network was fully usable within a few minutes of the coordinator and all other routers coming back online.

What exact devices do you have on your network?

MattWestb, 2021-11-15:
New firmware for new TI coordinator is in dev https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin.

aleck011mi, 2021-11-15:
Thanks Matt! I'll give it a try. I'm not seeing a changelog. Most of my devices are Xfinity security system devices: a keypad, door/window open sensors (XHS2-SE), and motion sensors. I do have a couple Aqara Water Leak sensors as well.

puddly, 2021-11-16:
> Most of my devices are Xfinity security system devices

Those seem like they would cause problems. Did the water leak sensors become unresponsive as well or just the Xfinity stuff?