zigpy/zigpy-znp #98: Conbee II as ZHA coordinator: can't make a backup

Barmatuhin, 2021-11-16:
The goal to transfer all zigbee devices from RPI with ConbeeII dongle (used as ZHA coordinator) to x86 host.
Try to backup devices with command

Steps are:
1. Disable ZHA integration in GUI
2. Go to SSH and get inside docker container - `docker exec -it homeassistant bash `
3. Give this command ` python -m zigpy_znp.tools.network_backup /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2456090-if00 -o network_backup.json`

And get timeout

> bash-5.1# python -m zigpy_znp.tools.network_backup /dev/serial/by-id/usb-dresden                                                  _elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2456090-if00 -o network_backup.jso                                                  n
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 395, in r                                                  equest
>     response = await response_future
> asyncio.exceptions.CancelledError
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.9/runpy.py", line 197, in _run_module_as_main
>     return _run_code(code, main_globals, None,
>   File "/usr/local/lib/python3.9/runpy.py", line 87, in _run_code
>     exec(code, run_globals)
>   File "/usr/local/lib/python3.9/site-packages/zigpy_znp/tools/network_backup.py                                                  ", line 104, in <module>
>     asyncio.run(main(sys.argv[1:]))  # pragma: no cover
>   File "/usr/local/lib/python3.9/asyncio/runners.py", line 44, in run
>     return loop.run_until_complete(main)
>   File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until                                                  _complete
>     return future.result()
>   File "/usr/local/lib/python3.9/site-packages/zigpy_znp/tools/network_backup.py                                                  ", line 95, in main
>     await znp.connect()
>   File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 116, in c                                                  onnect
>     self.capabilities = (await self.request(c.SYS.Ping.Req())).Capabilities
>   File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 395, in r                                                  equest
>     response = await response_future
>   File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line                                                   55, in __aexit__
>     self._do_exit(exc_type)
>   File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line                                                   92, in _do_exit
>     raise asyncio.TimeoutError
> asyncio.exceptions.TimeoutError

What i'm doing wrong? Any chance to solve it?

Adminiuga, 2021-11-16:
Zigpy-znp does backups of znp based radios only. It is not applicable to ConBee stick, because it is different. 
In theory you could use Deconz to write down the extended pan Id, pan Id, keys etc and create a manual backup to be imported to znp. ATM there are not utilities to do it automatically.

puddly, 2021-11-16:
If you're not migrating between two coordinator sticks, there's no need to perform a backup. The relevant network settings are stored on your stick.

If you're trying to migrate from a Conbee II to a TI coordinator (e.g. CC2652), follow the instructions in this PR: https://github.com/zigpy/zigpy-cli/pull/2

Adminiuga, 2021-11-16:
Actually what puddly said. I missed you are just changing the hosts, and not the sticks.

closed 2021-11-16 by puddly

Barmatuhin, 2021-11-16:
@puddly  is it works if i use Conbee II stick as ZHA coordinator? All devices and entities stored in zigbee.db, and if i copy configs and db from RPi's platform (OS, Supervisor, Docker) to HA Core on x86 all works?

> # Back up the old radio.      Note: this backup format will change in the future.
> zigpy -vv radio deconz /dev/serial/by-id/... backup deconz.pickle

[image]

puddly, 2021-11-16:
Yes. If you are using the same coordinator, there's no need to run anything or do anything else besides copying `/config/`.

Barmatuhin, 2021-11-16:
@puddly  - Thank you! Will try

Barmatuhin, 2021-11-16:
Sorry for offtopic, but after transfer zha.storage,  zigbee.db and merging configuration.yaml i can see Conbee II stick as ZHA coordinator, but all device names and entities reset names to factory names. Is it any lifehack to transfer device and entities names  to new host?