zigpy/zigpy-znp #99: [BUG] can't bind rwl021

Ulrar, 2021-11-18:
**Describe the bug**
I can't figure out a way to bind a hue dimmer switch (rwl021) to a bulb or a group using ZHA.
Binding works fine when using zigbee2mqtt so the hardware is capable.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair a rwl021
2. Try to unbind from controller
3. Try to bind it to a light or group
4. See it does nothing

**Expected behavior**
Being able to bind a switch to a light or group.

**Additional context**
I've seen other reports saying it has to be unbound from the controller first, which is also true when using zigbee2mqtt, but that does not help as the unbinding fails :

```
 2021-10-23 17:18:30 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.UnBindRsp.Callback(Src=0x22DE, Status=<Status.NO_ENTRY: 136>) 2021-10-23 17:18:30 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.UnBindRsp.Callback(Src=0x22DE, Status=<Status.NO_ENTRY: 136>) 2021-10-23 17:18:30 DEBUG (MainThread) [zigpy.zdo] [0x22de:zdo] cluster: 6 Unbind_req --> [00:12:4b:00:21:5f:f7:cf] failed: [0x22de:0:0x0022]: Message send failure 2021-10-23 17:18:30 DEBUG (MainThread) [zigpy.zdo] [0x22de:zdo] cluster: 8 Unbind_req --> [00:12:4b:00:21:5f:f7:cf] failed: [0x22de:0:0x0022]: Message send failure 
```

According to the specs `Status.NO_ENTRY` means the switch didn't find that binding, but it must have it since it does send the button presses to the controller.
Trying to bind it to something else does log a success, but doesn't actually work, the button presses still come to the controller afterwards.

Googling this it seems I'm not the only one with that issue, I've found other reddit threads about this where people just give up.
Unfortunately I have a lot of these switches, I'd rather not have to replace them, I hope I'm just doing something wrong.

puddly, 2021-11-18:
Hmm. I'm not able to replicate this locally with zigpy-znp 0.5.4 (or 0.6.1) by modifying an existing device to have those same endpoints, clusters, and profiles:

```SQL
INSERT INTO devices_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',57535,2);
INSERT INTO endpoints_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',1,49246,2096,1);
INSERT INTO endpoints_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',2,260,12,1);
INSERT INTO in_clusters_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',1,0);
INSERT INTO in_clusters_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',2,0);
INSERT INTO in_clusters_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',2,1);
INSERT INTO in_clusters_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',2,3);
INSERT INTO in_clusters_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',2,15);
INSERT INTO in_clusters_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',2,64512);
INSERT INTO node_descriptors_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',2,0,0,0,0,8,128,4107,89,63,0,63,0);
INSERT INTO out_clusters_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',1,3);
INSERT INTO out_clusters_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',1,4);
INSERT INTO out_clusters_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',1,6);
INSERT INTO out_clusters_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',1,8);
INSERT INTO out_clusters_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',1,5);
INSERT INTO out_clusters_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',2,25);
INSERT INTO out_clusters_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',1,0);
INSERT INTO attributes_cache_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',1,0,4,'Philips');
INSERT INTO attributes_cache_v7 VALUES('00:0b:57:ff:fe:2b:cf:1f',1,0,5,'RWL021');
```

The correct quirk loads (`zhaquirks.philips.rwlfirstgen.PhilipsRWLFirstGen`) but binding/unbinding always uses endpoint 1 on both the coordinator and the device.  Do I need to modify something in `.storage` as well?

---

What's could be choosing the incorrect endpoint?

 1. zigpy-znp defines [`ControllerApplication.get_dst_address(cluster)`](https://github.com/zigpy/zigpy-znp/blob/4ac608c82b6f48d304bab9a20bd648450bed54a7/zigpy_znp/zigbee/application.py#L409-L426), which picks the correct endpoint based on the cluster's profile and cluster ID. Right now, it only considers the profile (since there are only two endpoints).
     - Zigpy uses that method within [`zdo.bind(cluster)` and `zdo.unbind(cluster)`](https://github.com/zigpy/zigpy/blob/7c168c1c2ca2a338676e5956d2277af7c40971d6/zigpy/zdo/__init__.py#L97-L111). These two methods are defined identically so binding and unbinding should be choosing the correct endpoints.

 2. ZHA also performs binding and unbinding symmetrically in [`async_binding_operation`](https://github.com/home-assistant/core/blob/94bfa5272d2a52181f85095f88121dd963898ff0/homeassistant/components/zha/api.py#L840) but it skips the zigpy convenience functions that utilize `get_dst_address`.

In theory, even if the wrong choice is made, it should be made for both bind and unbind.

puddly, 2021-11-26:
Still no luck replicating it by joining a fake version of the device at runtime:

```Python
        device = self.add_device(ieee=t.EUI64.convert("00:17:88:01:08:0e:1c:97"), nwk=0x7978)
        device.status = zigpy.device.Status.ENDPOINTS_INIT
        device.node_desc = zdo_t.NodeDescriptor(
            logical_type=zdo_t.LogicalType.EndDevice,
            complex_descriptor_available=0,
            user_descriptor_available=0,
            reserved=0,
            aps_flags=0,
            frequency_band=zdo_t.NodeDescriptor.FrequencyBand.Freq2400MHz,
            mac_capability_flags=zdo_t.NodeDescriptor.MACCapabilityFlags.AllocateAddress,
            manufacturer_code=4107,
            maximum_buffer_size=89,
            maximum_incoming_transfer_size=63,
            server_mask=0,
            maximum_outgoing_transfer_size=63,
            descriptor_capability_field=zdo_t.NodeDescriptor.DescriptorCapability.NONE,
        )

        device.manufacturer = "Philips"
        device.model = "RWL021"

        ep1 = device.add_endpoint(1)
        ep1.status = zigpy.endpoint.Status.ZDO_INIT
        ep1.profile_id = 49246
        ep1.add_input_cluster(0x0000)
        ep1.add_output_cluster(0x0000)
        ep1.add_output_cluster(0x0003)
        ep1.add_output_cluster(0x0004)
        ep1.add_output_cluster(0x0005)
        ep1.add_output_cluster(0x0006)
        ep1.add_output_cluster(0x0008)

        ep2 = device.add_endpoint(2)
        ep2.status = zigpy.endpoint.Status.ZDO_INIT
        ep2.profile_id = 260
        ep2.add_input_cluster(0x0000)
        ep2.add_input_cluster(0x0001)
        ep2.add_input_cluster(0x0003)
        ep2.add_input_cluster(0x000F)
        ep2.add_input_cluster(0xFC00)
        ep2.add_output_cluster(0x0019)

        self.listener_event("device_joined", device)
        await asyncio.sleep(0.1)
        self.listener_event("device_initialized", device)
```

Binding/unbinding still happens between endpoint 1 on both devices:

```Python
2021-11-26 14:56:38 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.BindReq.Req(Dst=0x7978, Src=00:17:88:01:08:0e:1c:97, SrcEndpoint=1, ClusterId=6, Address=MultiAddress(addrmode=3, ieee=00:12:4b:00:1e:17:ef:a8, endpoint=1))
2021-11-26 14:56:38 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.BindReq.Req(Dst=0x7978, Src=00:17:88:01:08:0e:1c:97, SrcEndpoint=1, ClusterId=8, Address=MultiAddress(addrmode=3, ieee=00:12:4b:00:1e:17:ef:a8, endpoint=1))

2021-11-26 14:56:50 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.UnBindReq.Req(Dst=0x7978, Src=00:17:88:01:08:0e:1c:97, SrcEndpoint=1, ClusterId=6, Address=MultiAddress(addrmode=3, ieee=00:12:4b:00:1e:17:ef:a8, endpoint=1))
2021-11-26 14:56:50 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.UnBindReq.Req(Dst=0x7978, Src=00:17:88:01:08:0e:1c:97, SrcEndpoint=1, ClusterId=8, Address=MultiAddress(addrmode=3, ieee=00:12:4b:00:1e:17:ef:a8, endpoint=1))
```

Is this dimmer available for purchase in the United States?

Ulrar, 2021-11-26:
Not sure, I'm in Ireland.
But maybe I could just post you one, if that's what it takes ? they're not that expensive. Not sure on shipping

Adminiuga, 2021-11-26:
> ZHA also performs binding and unbinding symmetrically in async_binding_operation but it skips the zigpy convenience functions that utilize get_dst_address

And that's I believe is the cause. ZHA cannot use `Cluster.unbind()` method because that method is hardcoded to bind/unbind coordinator and ZHA needs to bind/unbind other devices, not just coordinator. Need to add an exception to use the `get_dst_address()`, when the target device is a coordinator.

puddly, 2021-11-26:
Is `get_dst_address` even necessary? zigpy-znp is the only library that implements it and the coordinator's `Device` object's endpoints and clusters (with profiles) are correct. In theory, ZHA has all the info it needs to select compatible endpoints.

Adminiuga, 2021-11-26:
That's because ZNP won't receive messages if the sender uses a different profile. EZSP, XBee and I think ConBee don't care. I'm not sure about zigate.

Adminiuga, 2021-11-26:
So essentially it was introduced specifically ZNP to allow more intelligent decision when choosing an endpoint. They really should patch the firmware so it ignores the profile and send everything to coordinator EP1.

puddly, 2021-11-26:
I think this has been addressed in the current Z-Stack patch: https://github.com/Koenkk/Z-Stack-firmware/blob/eb674e838cb030568bc483f44bba7abb4b728038/coordinator/Z-Stack_3.x.0/firmware.patch#L399

Will have to check on my network later.

Adminiuga, 2021-11-26:
Looks so indeed.

Ulrar, 2021-11-27:
Are you saying the bug is already fixed in the latest firmware ?
I'm fairly sure I'm already on the July firmware, but I could always reflash it to make sure, unless that patch isn't included in the July one.

puddly, 2021-11-27:
A different glitch but perhaps the workaround for it is causing your problem. Are you running `20210708`?

Ulrar, 2021-11-27:
I'm almost certain I am since in flashed it just before opening this bug report. But I'll flash it again tomorrow just to make sure, in case I used an older one somehow.

puddly, 2021-11-27:
The debug log will tell you the exact firmware revision right when the stick is start up. As does the device model info in ZHA.

puddly, 2021-11-27:
Flashing firmware is also going to erase network settings (if you do it) so make sure to perform a backup before doing so https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md#network-backup

Ulrar, 2021-11-27:
Ah, I hadn't noticed you could get the build from there.
So yes, I'm already on July : 

```
Texas Instruments CC1352/CC2652, Z-Stack 3.30+ (build 20210708)
```

Adminiuga, 2021-11-27:
> Are you saying the bug is already fixed in the latest firmware ?

not really, because it still binds EP2 on coordinator, but tries to unbind on EP1 on coordinator. 
But if you on the latest firmware with that patch included (dev branch?) and you can modify   https://github.com/zigpy/zigpy-znp/blob/cdd267cecd4f4dd963f4b8383de628d385f95dea/zigpy_znp/zigbee/application.py#L420-L424 to be `dst_addr.endpoint = 1`, then factory reset RWL022, re-pair and try un-binding again.

Ulrar, 2021-11-27:
I'm on the latest binary firmware in that repo, not sure whether or not the patch is included in it.

I'll give that a try tomorrow morning and report back, thanks !

Ulrar, 2021-11-27:
Just gave that a try with a switch there and after a reset, the unbind from controller does succeed.
I then bound it to a light group and it seems to be working as expected now, thanks for the workaround !

```
root@Homer:~# grep -ari 'ZDO.*Bind' home-assistant.log
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy.zdo] [0x022e:zdo] processing cluster: 6 Unbind_req --> [00:12:4b:00:21:5f:f7:cf]
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy.zdo] [0x022e:zdo] processing cluster: 8 Unbind_req --> [00:12:4b:00:21:5f:f7:cf]
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted a ZDO request: dst_addr=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x022E), dst_ep=0, src_ep=0, cluster=ZDOCmd.Unbind_req, sequence=58, options=TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST, radius=30, data=b':\n\r\x0e\x08\x01\x88\x17\x00\x01\x06\x00\x03\xcf\xf7_!\x00K\x12\x00\x01'
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted AP ZDO request ZDOCmd.Unbind_req({'SrcAddress': 00:17:88:01:08:0e:0d:0a, 'SrcEndpoint': 1, 'ClusterID': 6, 'DstAddress': MultiAddress(addrmode=3, ieee=00:12:4b:00:21:5f:f7:cf, endpoint=1)}) and replaced with ZDO.UnBindReq.Req(Dst=0x022E, Src=00:17:88:01:08:0e:0d:0a, SrcEndpoint=1, ClusterId=6, Address=MultiAddress(addrmode=3, ieee=00:12:4b:00:21:5f:f7:cf, endpoint=1))
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted a ZDO request: dst_addr=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x022E), dst_ep=0, src_ep=0, cluster=ZDOCmd.Unbind_req, sequence=59, options=TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST, radius=30, data=b';\n\r\x0e\x08\x01\x88\x17\x00\x01\x08\x00\x03\xcf\xf7_!\x00K\x12\x00\x01'
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted AP ZDO request ZDOCmd.Unbind_req({'SrcAddress': 00:17:88:01:08:0e:0d:0a, 'SrcEndpoint': 1, 'ClusterID': 8, 'DstAddress': MultiAddress(addrmode=3, ieee=00:12:4b:00:21:5f:f7:cf, endpoint=1)}) and replaced with ZDO.UnBindReq.Req(Dst=0x022E, Src=00:17:88:01:08:0e:0d:0a, SrcEndpoint=1, ClusterId=8, Address=MultiAddress(addrmode=3, ieee=00:12:4b:00:21:5f:f7:cf, endpoint=1))
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.UnBindReq.Req(Dst=0x022E, Src=00:17:88:01:08:0e:0d:0a, SrcEndpoint=1, ClusterId=6, Address=MultiAddress(addrmode=3, ieee=00:12:4b:00:21:5f:f7:cf, endpoint=1))
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.UnBindReq.Rsp(Status=<Status.SUCCESS: 0>)
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.UnBindReq.Req(Dst=0x022E, Src=00:17:88:01:08:0e:0d:0a, SrcEndpoint=1, ClusterId=8, Address=MultiAddress(addrmode=3, ieee=00:12:4b:00:21:5f:f7:cf, endpoint=1))
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.UnBindReq.Rsp(Status=<Status.SUCCESS: 0>)
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.UnBindRsp.Callback(Src=0x022E, Status=<Status.SUCCESS: 0>)
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.UnBindRsp.Callback(Src=0x022E, Status=<Status.SUCCESS: 0>)
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy.zdo] [0x022e:zdo] cluster: 6 Unbind_req --> [00:12:4b:00:21:5f:f7:cf] completed: [<Status.SUCCESS: 0>]
2021-11-27 11:26:54 DEBUG (MainThread) [zigpy.zdo] [0x022e:zdo] cluster: 8 Unbind_req --> [00:12:4b:00:21:5f:f7:cf] completed: [<Status.SUCCESS: 0>]
2021-11-27 11:27:15 DEBUG (MainThread) [zigpy.zdo] [0x022e:zdo] processing 0x022e: Bind_req 00:17:88:01:08:0e:0d:0a, ep: 1, cluster: 6 to group: 0x0003
2021-11-27 11:27:15 DEBUG (MainThread) [zigpy.zdo] [0x022e:zdo] processing 0x022e: Bind_req 00:17:88:01:08:0e:0d:0a, ep: 1, cluster: 8 to group: 0x0003
2021-11-27 11:27:16 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted a ZDO request: dst_addr=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x022E), dst_ep=0, src_ep=0, cluster=ZDOCmd.Bind_req, sequence=61, options=TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST, radius=30, data=b'=\n\r\x0e\x08\x01\x88\x17\x00\x01\x06\x00\x01\x03\x00'
2021-11-27 11:27:16 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted AP ZDO request ZDOCmd.Bind_req({'SrcAddress': 00:17:88:01:08:0e:0d:0a, 'SrcEndpoint': 1, 'ClusterID': 6, 'DstAddress': MultiAddress(addrmode=1, nwk=3)}) and replaced with ZDO.BindReq.Req(Dst=0x022E, Src=00:17:88:01:08:0e:0d:0a, SrcEndpoint=1, ClusterId=6, Address=MultiAddress(addrmode=1, nwk=3))
2021-11-27 11:27:16 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.BindReq.Req(Dst=0x022E, Src=00:17:88:01:08:0e:0d:0a, SrcEndpoint=1, ClusterId=6, Address=MultiAddress(addrmode=1, nwk=3))
2021-11-27 11:27:16 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted a ZDO request: dst_addr=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x022E), dst_ep=0, src_ep=0, cluster=ZDOCmd.Bind_req, sequence=62, options=TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST, radius=30, data=b'>\n\r\x0e\x08\x01\x88\x17\x00\x01\x08\x00\x01\x03\x00'
2021-11-27 11:27:16 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted AP ZDO request ZDOCmd.Bind_req({'SrcAddress': 00:17:88:01:08:0e:0d:0a, 'SrcEndpoint': 1, 'ClusterID': 8, 'DstAddress': MultiAddress(addrmode=1, nwk=3)}) and replaced with ZDO.BindReq.Req(Dst=0x022E, Src=00:17:88:01:08:0e:0d:0a, SrcEndpoint=1, ClusterId=8, Address=MultiAddress(addrmode=1, nwk=3))
[... cut ...]
2021-11-27 11:27:17 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.BindRsp.Callback(Src=0x022E, Status=<Status.SUCCESS: 0>)
2021-11-27 11:27:17 DEBUG (MainThread) [zigpy_znp.utils] Future already has a result set: <Future finished result=ZDO.BindRsp.C...s.SUCCESS: 0>)>
2021-11-27 11:27:17 DEBUG (MainThread) [zigpy.zdo] [0x022e:zdo] 0x022e: Bind_req 00:17:88:01:08:0e:0d:0a, ep: 1, cluster: 6 to group: 0x0003 completed: [<Status.SUCCESS: 0>]
2021-11-27 11:27:17 DEBUG (MainThread) [zigpy.zdo] [0x022e:zdo] 0x022e: Bind_req 00:17:88:01:08:0e:0d:0a, ep: 1, cluster: 8 to group: 0x0003 completed: [<Status.SUCCESS: 0>]
```

PR #105 (2021-12-02): Use coordinator endpoint 1 for Z-Stack >= 20210708
Fixes #99 by using endpoint 1 when Z-Stack build 20210708 or newer is used.
closed 2021-12-02 by puddly