zigpy/zigpy-znp #100: asyncio.exceptions.TimeoutError

wes1993, 2021-11-19:
Hello, I have this error in the log everytime i start HA.
Could someone help me?
```
2021-11-19 16:02:34 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xC5D1), DstEndpoint=1, DstP
anId=0x0000, SrcEndpoint=1, ClusterId=6, TSN=110, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=30, Data=b'\x18\x6E\x0B\x0A\x00')
2021-11-19 16:02:34 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2021-11-19 16:02:34 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=110)
2021-11-19 16:02:34 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=115)
2021-11-19 16:02:36 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 435, in request_callback_rsp
    return await callback_rsp
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 435, in request_callback_rsp
    return await callback_rsp
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 55, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 92, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
2021-11-19 16:02:36 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 435, in request_callback_rsp
    return await callback_rsp
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 435, in request_callback_rsp
    return await callback_rsp
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 55, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 92, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
2021-11-19 16:02:36 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 435, in request_callback_rsp
    return await callback_rsp
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 435, in request_callback_rsp
    return await callback_rsp
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 55, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 92, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
2021-11-19 16:02:44 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xB77F, Relays=[])
2021-11-19 16:02:44 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xB77F, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Boo
l.false: 0>, LQI=15, SecurityUse=<Bool.false: 0>, TimeStamp=246223, TSN=0, Data=b'\x08\x76\x0A\x00\x00\x10\x00', MacSrcAddr=0xB77F, MsgResultRadius=29)
2021-11-19 16:02:44 DEBUG (MainThread) [zigpy.zcl] [0xb77f:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=T
rue disable_default_response=False> manufacturer=None tsn=118 command_id=Command.Report_Attributes>
2021-11-19 16:02:44 DEBUG (MainThread) [zigpy.zcl] [0xb77f:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
2021-11-19 16:02:44 DEBUG (MainThread) [zigpy.zcl] [0xb77f:1:0x0006] Attribute report received: on_off=0
```
Thanks
Stefano

puddly, 2021-11-19:
Do you have any mains-powered devices that are offline? If it happens only on startup, it might be ZHA trying to check if they're online and failing.

wes1993, 2021-11-19:
Thanks a lot for your fast reply.
I don’t know, the power is in all devices but I don’t know why some devices sometimes lost the network probably for this?
Do you know if there is some problem with ikea Lamp/power outlet?

Thanks a lot 
Stefano