zigpy/zigpy-znp #108: a synopsis or template on how to use the zigpy-znp library

pitchoun38, 2021-12-08:
The zigpy-znp library looks very nice and I wonder if this could be use as a standalone library imported in a python3 code, without the all zigpy stack.

I got some hints from the interesting zigpy-cli project , and so far I'm able to open the communication with the Zigbee usb key. However, I'm struggling on  how to send zigbee raw command and how to receive them.

For receiving messages, I guess that I need to create a listener, but not sure on which API it should rely on.

For sending messages, that should I use the request(), broadcast() or mrequest() depending of unicast, briadcast or multi-cast ?
is those API standard for each zigpy-<radio> ?

pitchoun38, 2021-12-08:
This looks very interesting and especially that one [#709](https://github.com/zigpy/zigpy/discussions/709) thanks.

what is not clear for me is related to zipy-<radio module> versus zigby stack. In other terms, can I develop a python application relying only on the zigpy-<radio module> ?

What do you mean by "Received messages should be passed to handle_message." in [#709](https://github.com/zigpy/zigpy/discussions/709) ?

puddly, 2021-12-08:
> In other terms, can I develop a python application relying only on the zigpy- ?

Yes. Zigpy on its own just handles ZCL and ZDO, it doesn't talk to any radios and expects a `ControllerApplication` subclass to implement this.  You can use any library's `ControllerApplication` in the same way (`bellows.zigbee.application.ControllerApplication`, `zigpy_deconz.zigbee.application.ControllerApplication`, etc.).

zigpy-znp is dependent on zigpy and internally uses its structures, like `zigpy.device.Device`, so it's not possible to use standalone unless you want to directly talk to the Texas Instruments radio with `zigpy_znp.api.ZNP` and handle *everything* yourself.  Each radio library has a different *internal* interface for communicating with its Zigbee chip, which is why `ControllerApplication` exists.

> What do you mean by "Received messages should be passed to handle_message."

That linked document is explaining how to write a radio library to talk to a new USB coordinator, like zigpy-znp. It describes the internals. Zigpy implements [`handle_message`](https://github.com/zigpy/zigpy/blob/dev/zigpy/application.py#L196) but you can subclass `ControllerApplication` and replace all of this.

> In that case the all zigpy stack mecanism came, correct ?

What portion of zigpy are you trying to avoid?  If you configure it to use its SQLite database, it will store device NWK and IEEE addresses and when performing initialization, persist the node descriptor, endpoint descriptors, and model/manufacturer attribute values.  That's about it. Binding, setting up attribute reporting, and actual device control is up to your application.

Hedda, 2021-12-20:
Is this related to the Domoticz-Zigate implementation of zigpy-znp and zigate-znp radio libraries discussed here:

https://github.com/zigpy/zigpy/discussions/865

At least it sounds as pipiche38 what to achieve the same thing in his Zigate plugin for Domoticz because that already have its own ZCL and ZDO layers which is working so I believe he for now only wants to partly use zigpy radio libraires for hardware abstraction.

pitchoun38, 2022-01-07:
Thanks for the help

closed 2022-01-07 by pitchoun38