zigpy/zigpy-znp #115: List CC2530 and CC2531 under "Legacy support" and comment them with "not recommended"?

Hedda, 2021-12-20:
Suggest to list CC2530 and CC2531 in both zigpy-znp README.md and Home Assistant's ZHA integration in a "Legacy support" subsection under a compatible hardware and comment them with a "not recommended" explanation, what do you think?

Zigbee2MQTT community have recently added such recommendations with arguments that while CC2530 and CC2531 based Zigbee coordinator adapters still work in many scenarios they are no longer recommended by the Zigbee2MQTT community at all since they use outdated chips and older firmware. 

Thus might be a good idea for zigpy-znp and Home Assistant's ZHA integration to update compatibility lists in documentation by adding a clear recommendation that  CC2530 and CC2531 are not recommended in a "production" environment (since not powerful enough).

References:

https://www.zigbee2mqtt.io/guide/adapters/#not-recommended

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator#im-using-a-cc2530-or-cc2531-which-firmware-should-i-use

Note that Zigbee2MQTT community is now even listing CC2538 as "not recommended" in favour of newer CC26x2 and CC13x2 based adapters instead in order so that new users do not go out and blindly buy old CC2538 hardware as their first Zigbee coordinator just because it is listed as compatible hardware. 

PS: puddly also posted this comment in #48 about CC2538 around a year ago "The last Z-Stack update for it was released more than two years ago and it will not support the newer route discovery features that improve network stability because the version of Z-Stack that it's stuck with supposedly has glitches."

Hedda, 2022-01-20:
README.md for zigpy-znp added "not recommended" notes for CC2538 but maybe stronger worded discouragement is needed?

https://github.com/zigpy/zigpy-znp/blob/dev/README.md#hardware-requirements

CC2530 and CC2531 is compatible with zigpy-znp however not recommended due to slow old hardware and end-of-life firmware.

# Hardware requirements
USB-adapters, GPIO-modules, and development-boards flashed with TI's Z-Stack are compatible with zigpy-znp:

 - CC2652P/CC2652R/CC2652RB USB stick and dev board hardware
 - CC1352P/CC1352R USB stick and dev board hardware
 - CC2538 + CC2592 USB stick and dev board hardware (**not recommended, old hardware and end-of-life firmware**)
 - CC2531 USB stick hardware (**not recommended for Zigbee networks with more than 20 devices**)
 - CC2530 + CC2591/CC2592 USB stick hardware (**not recommended for Zigbee networks with more than 20 devices**)

Hedda, 2022-01-20:
FYI, submitted a PR for ZHA docs with a "not recommended" comment for all CC253x -> https://github.com/home-assistant/home-assistant.io/pull/21226

pipiche38, 2022-01-20:
Honestly for the CC2538/CC2592 I don't see what is the issue. if the Hardware is old - and of life this is not the zigpy-znp issue.
Now if the hardware because it is old do not support some features that are needed I would then understand.

Hedda, 2022-01-20:
> if the Hardware is old - and of life this is not the zigpy-znp issue.

While that is true, the problem is support and false bug reports as well as first-impression feedback/reviews from beginner users just starting out when the new end-users still keep buying CC2530/CC2531 based adapters as an impulse buy on a whim simply because of those listed as compatible the CC2530/CC2531 based ones are least expensive and most available worldwide.  

Those end-users are normally not met with positivity from more experienced community members who will usually tell them **_"buy a new adapter and do not use CC253x_**" which is throwing away time and money for those new users and a perhaps fair that they then go and tell their friends that the community have a bad attitude which is not friendly to new users.

Yes you could say that users who don't do research are lazy but that does not stop them from spreading rumours online that zigpy based implementations do not work properly, forgetting to mention that the reason might be the obsolete hardware they bought because it was cheap. 

Please understand that I am not trying to solve a technical problem here but instead solve an educational issue, as in trying to educate or scare new users to not buy CC2530/CC2531 so that they do not make a mistake due to them being uninformed which that will give them a bad experience that could lead to them not continuing to use zigpy based implementations or worse, bad talk zigpy based implementations to others so that end result is a smaller userbase, which perhaps leads to fewer developers.