zigpy/zigpy-znp #125: What happens if user set tx_power to 20 for CC2652P/CC1352P instead of 19 which looks to be current max?

Hedda, 2022-01-05:
I think a real-world could problem could be that many ZHA users will just assume that 20 will be the tx_power maximum that they can set even if the documentation says the range only goes to 19 because both Texas Instruments and Zigbee coordinator manufacturers (such as ITead/Sonoff) are have marketing material which state that CC2652P/CC1352P as having "+20 dBm" power amplification.

https://github.com/zigpy/zigpy-znp/blob/dev/README.md#configuration

```
      # Only if your stick has a built-in power amplifier (i.e. CC1352P and CC2592)
      # If set, must be between:
      #  * CC1352/2652:  -22 and 19
      #  * CC253x:       -22 and 22
      tx_power:  
```

For consistency might it not be a good idea to allow a tx_power maximum of 20 in zigpy-znp/ZHA configuration?

This might otherwise also confuse users coming from Zigbee2MQTT/Z2M to ZHA/zigpy-znp as it allow users to set transmit_power to 20:

https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#transmitter-power

```
experimental:
  # Optional: Transmit power setting in dBm (default: 5).
  # This will set the transmit power for devices that bring an inbuilt amplifier.
  # It can't go over the maximum of the respective hardware and might be limited
  # by firmware (for example to migrate heat, or by using an unsupported firmware).
  # For the CC2652R(B) this is 5 dBm, CC2652P/CC1352P-2 20 dBm.
  transmit_power: 5
```

@puddly  What happens if ZHA users try to set tx_power to 20 for CC2652P/CC1352P instead of 19 which looks to be current max?

As I understand it, the CC2538 + CC2592 TX Power maximum was 19, but the TX Power maximum for newer CC2652P/CC1352P is 20?

Did you already change the  tx_power maximum to 20 for CC2652P/CC1352P in zigpy-znp or is 19 still the maximum for all types?`

At least I think when you implemented tx_power setting in zigpy-znp it was for CC2538 + CC2592 only which has max tx_power 19?

https://github.com/zigpy/zigpy-znp/issues/81

mdeweerd, 2022-01-06:
The CC2530 ZNP stack compares the requested value to the maximum in the internal table and caps the value to the maximum.
I suppose this is the same for the newer stacks.

Hedda, 2022-01-10:
> The CC2530 ZNP stack compares the requested value to the maximum in the internal table and caps the value to the maximum. I suppose this is the same for the newer stacks.

Do you mean it is already possible to set tx_power to 20 for zigpy-znp (in HA's [ZHA](https://www.home-assistant.io/integrations/zha)) so it will accept and pass to ZNP stack which internally will use 20? Or will zigpy-znp reject setting if try to configure  tx_power to 20 instead of 19 with is mentioned as max?

Again, both zigpy-znp docs currently state that 19 is highest can set tx_power for CC1352/2652 (i.e. CC2652P/CC1352P):

https://github.com/zigpy/zigpy-znp/blob/dev/README.md#configuration

```
      # Only if your stick has a built-in power amplifier (i.e. CC1352P and CC2592)
      # If set, must be between:
      #  * CC1352/2652:  -22 and 19
      #  * CC253x:       -22 and 22
      tx_power:  
```

ZHA docs does not yet mention TX Power:

https://www.home-assistant.io/integrations/zha

mdeweerd, 2022-01-10:
> Do you mean it is already possible to set tx_power to 20 for zigpy-znp (in HA's [ZHA](https://www.home-assistant.io/integrations/zha)) so it will accept and pass to ZNP stack which internally will [...]

I do not know if zigpy-znp sets a limit, but I referred to the ZNP firmware itself.  The protocol has a "set_tx_power" command and if you exceed the value that is configured in the firmware it is capped to the maximum.  So if you ask for 20 and the firmware is set to 5 then it will be "5" .   The firmware is hard coded with the limits (which depend on all hardware involved, including a potential RF amplifier).

puddly, 2022-01-14:
The `set_tx_power` option forwards the value directly to the firmware via `c.SYS.SetTxPower.Req(TXPower=dbm)`.

I've confirmed that (reverse) LQI steadily increases as TX power increases and plateaus after reaching the firmware maximum. Setting it to `0`, unplugging the coordinator, then to `100`, unplugging the coordinator, and then letting it run does not cause the LQI to decrease, confirming that the firmware limits it.

closed 2022-01-14 by puddly