zigpy/zigpy-znp #127: await device.zdo.request(zdo_t.ZDOCmd.Mgmt_Rtg_req, idx) ultimately results in AssertionError

mdeweerd, 2022-01-10:
The call:
```python
await device.zdo.request(zdo_t.ZDOCmd.Mgmt_Rtg_req, idx)
```

results in:
```raw
  File "/config/custom_components/zha_custom/neighbours.py", line 195, in async_get_routes
    status, val = await device.zdo.request(zdo_t.ZDOCmd.Mgmt_Rtg_req, idx)
  File "/usr/local/lib/python3.9/site-packages/zigpy/device.py", line 265, in request
    result, msg = await self._application.request(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 453, in request
    return await self._send_request(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1483, in _send_request
    response = await self._send_request_raw(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1326, in _send_request_raw
    return await self._send_zdo_request(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1295, in _send_zdo_request
    self._receive_zdo_message(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1081, in _receive_zdo_message
    assert set(zdo_kwargs) == set(field_names)
```

puddly, 2022-01-10:
Try it with https://github.com/zigpy/zigpy-znp/pull/109

mdeweerd, 2022-01-11:
I see that you're refactoring a lot related to this - I can't set up a test setup for this.

I am closing this and will report again later if this still happens after the changes propagated to Home Assistant.

closed 2022-01-11 by mdeweerd