zigpy/zigpy-znp #129: self._znp.request_callback_rsp( AttributeError: 'NoneType' object has no attribute 'request_callback_rsp'

mdeweerd, 2022-01-17:
I got this on `ha core restart` with core-2021.12.9: 
```raw
2022-01-17 21:44:47 ERROR (MainThread) [zigpy.zcl] [0xefb3:14:0x0702] Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy/device.py", line 265, in request
    result, msg = await self._application.request(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 453, in request
    return await self._send_request(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1483, in _send_request
    response = await self._send_request_raw(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1369, in _send_request_raw
    self._znp.request_callback_rsp(
AttributeError: 'NoneType' object has no attribute 'request_callback_rsp'
```

puddly, 2022-01-17:
This is likely because pending requests were not cancelled or new ones were initiated after the serial port was closed.

mdeweerd, 2022-01-17:
This happened during the startup phase and the exceptions were reported as 'Task exception was never retrieved'.
I do not know if that applies to exceptions generated before startup (during the previous shutdown).

It doesn't look critical, I am mainly informing about this.

mdeweerd, 2023-04-27:
Still happening with other radio.
Only at startup, 1 year old & not critical -> closing.

```
2023-04-25 00:15:27.479 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/protocol.py", line 131, in command
    return await future
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/v8/__init__.py", line 51, in pre_permit
    await self.setPolicy(
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/protocol.py", line 130, in command
    async with asyncio_timeout(EZSP_CMD_TIMEOUT):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```

closed 2023-04-27 by mdeweerd