zigpy/zigpy-znp #130: AttributeError: 'NoneType' object has no attribute 'send' (self._uart.send(frame)) errors

deviantintegral, 2022-01-24:
I've upgraded to zigpy-znp 0.7.0 and zigpy 0.42.0 manually in my Home Assistant install. I'm seeing some new errors in the logs. `0xa6a6` is a Sengled E1C-NB7 power switch, and `0x88a1` is a Sengled Z01-A19NAE26 bulb.

> Logger: homeassistant
> Source: /usr/src/homeassistant/homeassistant/runner.py:98 
> First occurred: 3:56:04 AM (5 occurrences) 
> Last logged: 3:56:04 AM
> 
> Error doing job: Task exception was never retrieved
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 886, in request_callback_rsp
>     await self.request(request, **response_params)
>   File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 842, in request
>     self._uart.send(frame)
> AttributeError: 'NoneType' object has no attribute 'send'

> Logger: zigpy.zcl
> Source: /usr/local/lib/python3.9/site-packages/zigpy/util.py:293 
> First occurred: 3:56:04 AM (6 occurrences) 
> Last logged: 3:56:04 AM
> 
> [0xa6a6:1:0x0702] Traceback (most recent call last): File "/usr/local/lib/python3.9/site-packages/zigpy/device.py", line 265, in request result, msg = await self._application.request( File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 439, in request return await self._send_request( File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1401, in _send_request response = await self._send_request_raw( File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1286, in _send_request_raw response = await asyncio.shield( File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 886, in request_callback_rsp await self.request(request, **response_params) File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 842, in request self._uart.send(frame) AttributeError: 'NoneType' object has no attribute 'send'
> [0x88a1:1:0x0019] Traceback (most recent call last): File "/usr/local/lib/python3.9/site-packages/zigpy/zcl/clusters/general.py", line 1031, in _handle_cluster_request await self._handle_query_next_image(*args, tsn=tsn) File "/usr/local/lib/python3.9/site-packages/zigpy/zcl/clusters/general.py", line 1099, in _handle_query_next_image await self.query_next_image_response( File "/usr/local/lib/python3.9/site-packages/zigpy/device.py", line 265, in request result, msg = await self._application.request( File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 439, in request return await self._send_request( File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1401, in _send_request response = await self._send_request_raw( File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1286, in _send_request_raw response = await asyncio.shield( File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 886, in request_callback_rsp await self.request(request, **response_params) File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 842, in request self._uart.send(frame) AttributeError: 'NoneType' object has no attribute 'send'

puddly, 2022-01-24:
All of these errors occurred at exactly 3:56:04 AM.  Have you seen them since?  Did you restart HA Core or fully shut it down and start it back up?

deviantintegral, 2022-01-25:
Thanks for noticing that the first and last timestamps were the same. I was just looking at the count of messages.

I restarted yesterday to update Supervisor, and so far there's been no errors. Let's close this unless it comes up again or someone else sees this.

closed 2022-01-25 by deviantintegral