zigpy/zigpy-znp #131: python3 -m zigpy_znp.tools.energy_scan /dev/ttyUSB0 creates error

pipiche38, 2022-02-02:
I have a Domoticz user whom is reporting a strange behaviour at startup.

So I asked hime to run the here after command and that produce some zigpy stack trace.

python3 -m zigpy_znp.tools.energy_scan /dev/ttyUSB0

The guy is using a Sonoff key

```
traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in runmodule_as_main
    return runcode(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in runcode
    exec(code, run_globals)
  File "/home/joscha/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/tools/energy_scan.py", line 10, in <module>
    from zigpy_znp.tools.common import setup_parser
  File "/home/joscha/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/tools/common.py", line 7, in <module>
    import jsonschema
  File "/usr/lib/python3/dist-packages/jsonschema/init.py", line 11, in <module>
    from jsonschema.exceptions import (
  File "/usr/lib/python3/dist-packages/jsonschema/exceptions.py", line 152, in <module>
    @attr.s(hash=True)
AttributeError: module 'attr' has no attribute 's'
```

puddly, 2022-02-02:
Looks like a dependency version problem. What does `pip freeze` show?

pipiche38, 2022-02-02:
I asked the user to do 
sudo pip3 uninstall attr
sudo pip3 install attrs

and looks like it is ok

closed 2022-02-02 by pipiche38