zigpy/zigpy-znp #137: Backup of HUSBZB-1 not working

balcanuc, 2022-02-17:
### Discussed in https://github.com/zigpy/zigpy-znp/discussions/90

The recipe described on https://github.com/zigpy/zigpy-znp/discussions/90 that allows backing up a zigbee configuration from a home-assistant instance  is not working anymore. 

When I run command below
```
pip install zigpy-znp 'git+https://github.com/puddly/bellows@puddly/open-coordinator-backup'
```
```
I get output below. I think is related to a dependency reference but I'm not a python expert, so I might be wrong.

bash-5.1# pip install zigpy-znp 'git+https://github.com/puddly/bellows@puddly/open-coordinator-backup'
Collecting git+https://github.com/puddly/bellows@puddly/open-coordinator-backup
  Cloning https://github.com/puddly/bellows (to revision puddly/open-coordinator-backup) to /tmp/pip-req-build-86ee7k1s
    ERROR: Command errored out with exit status 1:
     command: /usr/local/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-86ee7k1s/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-86ee7k1s/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-agkl8k_w
         cwd: /tmp/pip-req-build-86ee7k1s/
    Complete output (28 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-86ee7k1s/setup.py", line 3, in <module>
        raise RuntimeError(
    RuntimeError:

    #######################################################################################
    This branch is obsolete!  Please test https://github.com/zigpy/zigpy-cli/pull/2 instead
    #######################################################################################

    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

JMan7777, 2022-02-25:
> #######################################################################################
>     This branch is obsolete!  Please test https://github.com/zigpy/zigpy-cli/pull/2 instead
>     #######################################################################################

@balcanuc, as written please use the new method for backup/restore [here](https://github.com/zigpy/zigpy-cli/pull/2).
I successfully migrated my ZHA coordinator from a Nortek HUSBZB-1 stick (Firmware 6.7.8) to a SONOFF ZigBee 3.0 USB Dongle Plus (Firmware CC1352P2_CC2652P_launchpad_coordinator_20220219) using the provided steps.

puddly, 2022-02-25:
It looks like @balcanuc figured it out in the end: https://github.com/zigpy/zigpy-cli/pull/2#issuecomment-1043406307. Thanks for testing things out!

closed 2022-02-25 by puddly