zigpy/zigpy-znp #141: Add asyncio.exceptions.CancelledError as accepted error for retryable_request

mdeweerd, 2022-03-03:


puddly, 2022-03-06:
I've looked a bit closer at this problem and I think the extra traceback context is causing confusion. `asyncio.CancelledError` is never raised, only `asyncio.TimeoutError`:

```Python
import asyncio
import async_timeout

async def inner():
    async with async_timeout.timeout(0.1):
        await asyncio.sleep(1)

async def main():
    try:
        await asyncio.shield(inner())
    except Exception as e:
        print(f'!!! Caught exception: {e!r}')
        raise

if __name__ == "__main__":
    asyncio.run(main())
```

Output:

```Python
!!! Caught exception: TimeoutError()
Traceback (most recent call last):
  File "/tmp/test.py", line 7, in inner
    await asyncio.sleep(1)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 605, in sleep
    return await future
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/tmp/test.py", line 18, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 641, in run_until_complete
    return future.result()
  File "/tmp/test.py", line 12, in main
    await asyncio.shield(inner())
  File "/tmp/test.py", line 6, in inner
    async with async_timeout.timeout(0.1):
  File "/tmp/venv/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/tmp/venv/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```

In short, catching `asyncio.CancelledError` within any asyncio code without a *very* good reason will not solve this problem and would introduce other more serious bugs.

closed 2022-03-06 by puddly

mdeweerd, 2022-03-06:
Hi @puddly 

Baseed on your last message, I found in python3.9 code (asyncio/tasks.py):
```python
async def sleep(delay, result=None, *, loop=None):
    """Coroutine that completes after a given time (in seconds)."""
    if delay <= 0:
        await __sleep0()
        return result

    if loop is None:
        loop = events.get_running_loop()
    else:
        warnings.warn("The loop argument is deprecated since Python 3.8, "
                      "and scheduled for removal in Python 3.10.",
                      DeprecationWarning, stacklevel=2)

    future = loop.create_future()
    h = loop.call_later(delay,
                        futures._set_result_unless_cancelled,
                        future, result)
    try:
        return await future
    finally:
        h.cancel()
```

and in futures.py:
```
    def cancel(self):
        """Cancel the future and schedule callbacks.

        If the future is already done or cancelled, return False.  Otherwise,
        change the future's state to cancelled, schedule the callbacks and
        return True.
        """
        self.__log_traceback = False
        if self._state != _PENDING:
            return False
        self._state = _CANCELLED
        self.__schedule_callbacks()
        return True

[...]

    def result(self):
        """Return the result this future represents.

        If the future has been cancelled, raises CancelledError.  If the
        future's result isn't yet available, raises InvalidStateError.  If
        the future is done and has an exception set, this exception is raised.
        """
        if self._state == _CANCELLED:
            raise exceptions.CancelledError

```

While I haven't checked in detail if those items are part of the execution path in this case, it looks like a reasonnable assumption that a timeout on a loop results in a CancelledError exception.

puddly, 2022-03-06:
The cancellation error does occur but it never leaves the timeout library, as expected: https://github.com/aio-libs/async-timeout/blob/12624c224268a0d24437739b995c601d4f47887f/async_timeout/__init__.py#L210

I'm unable to reproduce this issue myself. The only way I can see catching the cancellation error helping things is if it's preventing a higher level call from cancelling the request, which is unexpected behavior.