zigpy/zigpy-znp #147: Not all devices in the backup file

braucktoon, 2022-04-09:
Hello,

I seem to be missing 4 devices in my backup.  HA registers 20 devices including Sonoff coordinator.  In the backup file I see only 15 devices in the devices array.  I think I am missing 4 devices total in the backup.  All devices seem to be working in HA, they all respond and I only have one coordinator connected.  Steps I took are:

pip install zigpy-znp
* Disabled ZHA *
python -m zigpy_znp.tools.network_backup /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_2071f1f25645ec118ea296fd6f14af06-if00-port0 -o network_backup.json

Output looks like this:
{
    "metadata": {
        "version": 1,
        "format": "zigpy/open-coordinator-backup",
        "source": "zigpy-znp@0.7.0",
        "internal": {
            "creation_time": "2022-04-08T19:42:11-04:00",
            "zstack": {
                "version": 3.3
            }
        }
    },
    "coordinator_ieee": "00124b0024c31dc3",
    "pan_id": "054c",
    "extended_pan_id": "f19bc3ca911ece4f",
    "nwk_update_id": 0,
    "security_level": 5,
    "channel": 15,
    "channel_mask": [
        15
    ],
    "network_key": {
        "key": "e6351c3313f1c087cd80d8b7cf392e57",
        "sequence_number": 0,
        "frame_counter": 153611
    },
    "devices": [
        {
            "ieee_address": "00158d00024349b5",
            "nwk_address": "6562",
            "is_child": true
        },
        {
            "ieee_address": "00158d0002529142",
            "nwk_address": "744e",
            "is_child": true
        },
        {
            "ieee_address": "00158d0002751d0e",
            "nwk_address": "34b1",
            "is_child": false
        },
        {
            "ieee_address": "00158d0002751fcd",
            "nwk_address": "4d22",
            "is_child": true
        },
        {
            "ieee_address": "00158d000275278a",
            "nwk_address": "b41e",
            "is_child": true
        },
        {
            "ieee_address": "00158d000282d110",
            "nwk_address": "e8f4",
            "is_child": true
        },
        {
            "ieee_address": "00158d000282d477",
            "nwk_address": "b20f",
            "is_child": true
        },
        {
            "ieee_address": "00158d000282d58c",
            "nwk_address": "4cac",
            "is_child": true
        },
        {
            "ieee_address": "00158d000282f063",
            "nwk_address": "b429",
            "is_child": true
        },
        {
            "ieee_address": "00158d000283b973",
            "nwk_address": "6a15",
            "is_child": true
        },
        {
            "ieee_address": "00158d000283c4d5",
            "nwk_address": "4334",
            "is_child": true
        },
        {
            "ieee_address": "00158d0002901775",
            "nwk_address": "a1a4",
            "is_child": true
        },
        {
            "ieee_address": "00158d000313b138",
            "nwk_address": "07ea",
            "is_child": true
        },
        {
            "ieee_address": "00158d000799aa76",
            "nwk_address": "f0a6",
            "is_child": true
        },
        {
            "ieee_address": "8d000799aa769ab4",
            "nwk_address": "0015",
            "is_child": true
        }
    ],
    "stack_specific": {
        "zstack": {
            "tclk_seed": "b13619818b2b6a76b6ecfaa730d8b17d"
        }
    }

[image]

Maybe I am missing a step or did something wrong?  I would expect to see 19 devices in the JSON.

Thank you.

puddly, 2022-04-09:
The Zigbee coordinator only needs to keep track of devices that share encryption keys with it or are end devices directly connected to it. Everything else (i.e. sensors connected to other routers and other routers themselves) aren't necessary.

Realistically, you can delete all of the devices in the backup and the network will still function after a restore.

braucktoon, 2022-04-09:
Thank you for the quick response and education 😀

closed 2022-04-09 by puddly