zigpy/zigpy-znp #149: Confirm Key not reach in the right time after a Verify Key

pipiche38, 2022-04-28:
I don't know if this is the zigpy layer or the firmware which is handling the Key verification, but I'm struggling to get a specific device paired on ZNP with the zigpy layer.

This problem doesn't happen with the Zigate. 

https://github.com/Koenkk/Z-Stack-firmware/issues/365

puddly, 2022-05-02:
Zigpy doesn't control any part of the key exchange, this is all handled by the firmware.

pipiche38, 2022-05-02:
Thanks. I close that one and will follow up on the firmware. Did you see such issue in the past ?

puddly, 2022-05-05:
Can you post a complete Python debug log of the device communicating with the coordinator? zigpy-znp should be receiving the ZDO node descriptor request callback.

pipiche38, 2022-05-06:
@puddly I'm here after attaching the pairing log and the sniff  at the same time.

For now, I have put a sleep of 6 seconds after sending the Active Endpoint Request.

If you look to the sniff, we see packet 113 receiving the Node Descriptor and then immediately triggering the verification.
So if you could on the zigpy level (because I have the same issue with bellows) regulate the outgoing communication when the Node Descriptor is sent (I see here the confirm key is given in 3 seconds)

[CMS323-pairing-with-zzh.log](https://github.com/zigpy/zigpy-znp/files/8639098/CMS323-pairing-with-zzh.log)
[CMS323-Pairing-with-zzh.pcapng.zip](https://github.com/zigpy/zigpy-znp/files/8639102/CMS323-Pairing-with-zzh.pcapng.zip)

PS/ Transaction Key: 6478b48e5254773c2c159f699f823b4f

pipiche38, 2022-05-06:
@puddly find here sniff and logs based on elelabs (bellows)

[elelabs-CMS.log](https://github.com/zigpy/zigpy-znp/files/8641357/elelabs-CMS.log)
[CMS323-Pairing-Elelabs.pcapng.zip](https://github.com/zigpy/zigpy-znp/files/8641368/CMS323-Pairing-Elelabs.pcapng.zip)

Transaction Key: 0b5d7cc7f0e84de7a88de7be2c384cbb

pipiche38, 2022-05-10:
I unfortunatly do not see any Node Descriptor Request coming to the plugin layer.
I think the only we get it , is when it is following a request of the node descriptor from the plugin to the 0000 address (using the loop back)

closed 2022-12-31 by pipiche38