zigpy/zigpy-znp #156: Remone serial via tcpip

badzz, 2022-07-01:
Hi
I would like to setup something similar to this
https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_adapter.html#_1-install-ser2net
But I see that we manipulate rts and DTS lines. Will it work or I need to do something more complicated?
Thanks

Hedda, 2022-08-02:
FYI, remote connect to ser2net service with socat and works with Home Assistant's ZHA same way as Zigbee2MQTT

https://www.home-assistant.io/integrations/zha

https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_adapter.html

This type of tunneling + relay generally happens at the operating system level and is usually transparent to the application:

https://linux.die.net/man/8/ser2net

https://linux.die.net/man/1/socat

puddly, 2022-08-31:
RTS and DTS are not supported over TCP.  They are only useful for CC2530 and other hardware that connects them to RST so you can ignore them.

closed 2022-08-31 by puddly

MattWestb, 2022-08-31:
Our hacked tuya TBGW is using hardware flow control over Eth and looks working.
If the hardware handshake is not working i think the NCP is not sending anything then trying communicating with it and we is only piping the raw commands from the UART to the network and the host (bellows) so is the the firmware on the NCP not working OK (Gary cooked) or is Bellows supporting it over sockets.
(I hacked 2 boxes and one is running my IKEA controllers test setup on it and the other i have enabling tuya app for sniffing my tuya devices behavior).