zigpy/zigpy-znp #161: Backup works fine but restore fails

jds11111, 2022-07-23:
I wanted to upgrade my Sonoff Zigbee 3.0 Dongle plus to latest firmware, so am trying to utilize the backup/restore tool.  I first backed everything up, which went fine.  The JSON file all looks OK, and no errors were shown. I then updated the firmware using `cc2538-bsl.py`.  That also went fine, and no errors were shown.  

However, the backup part did not work and threw the errors:
```
traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 248, in start_network
    commissioning_rsp = await self.request_callback_rsp(
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 1022, in request_callback_rsp
    async with async_timeout.timeout(timeout):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/tools/network_restore.py", line 133, in <module>
    asyncio.run(main(sys.argv[1:]))  # pragma: no cover
  File "/usr/local/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/tools/network_restore.py", line 125, in main
    await restore_network(
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/tools/network_restore.py", line 101, in restore_network
    await znp.write_network_info(network_info=network_info, node_info=node_info)
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 365, in write_network_info
    await self.start_network()
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 286, in start_network
    raise RuntimeError(
RuntimeError: Network formation refused, RF environment is likely too noisy. Temporarily unscrew the antenna or shield the coordinator with metal until a network is formed.
```
ZHA was disabled the whole time.

puddly, 2022-07-23:
Is your coordinator plugged into a USB 2.0 extension cable and placed away from common interference sources? Can you try following the suggestions in the error message?

jds11111, 2022-07-23:
Thanks.  This time I put it in a Faraday cage (aluminum foil) and it worked!  I really did not expect it to be so sensitive.  Thanks for being patient with my stupidity.

closed 2022-07-23 by jds11111

puddly, 2022-07-23:
Glad it worked.

Note that this isn't normal behavior: you have some sort of interference problem affecting all of the channels. I would strongly suggest a USB 2.0 extension cable, as network formation failing likely will mean that your radio is being negatively affected as well when it's running, reducing your effective range significantly.

jds11111, 2022-07-23:
There is a Nortek stick nearby that I am using for Zwave that might be causing the problems.  I should probably swap that stick out with a USB-7 that I have.  Unfortunately, I think that requires re-pairing all the devices. I will get a USB 2.0 cable to be sure.