zigpy/zigpy-znp #170: Zigpy error when receiving a Malformed packet

pipiche38, 2022-09-07:
Looks like the libraries is not resilient to "Malformed Packet"

Here after is the Sniffed paquet

[image]

And here after is the message produce by the zigpy library

```
2022-09-07 20:04:01,025 WARNING :Failed to parse broken GeneralFrame(header=CommandHeader(id=0x9F, subsystem=Subsystem.ZDO, type=CommandType.AREQ), data=b'\x7F\xF1\x84\x04\x45') as <class 'zigpy_znp.types.commands.ZDO.ParentAnnceRsp.Callback'>
2022-09-07 20:04:01,029 ERROR   :Task exception was never retrieved
future: <Task finished name='Task-33' coro=<ControllerApplication.on_zdo_message() done, defined at /var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/zigbee/application.py:519> exception=ValueError('Data is too short to contain 1 bytes')>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 256, in __step
    result = coro.send(None)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/zigbee/application.py", line 527, in on_zdo_message
    args, data = list_deserialize(data, types)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/basic.py", line 668, in deserialize
    item, data = cls._item_type.deserialize(data)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/basic.py", line 694, in deserialize
    item, data = cls._item_type.deserialize(data)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/basic.py", line 147, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 1 bytes
```

puddly, 2022-09-07:
What is the expected behavior here? Nothing crashes, it's just a traceback.

MattWestb, 2022-09-07:
PA is oft being used for dosing NCPc for getting control of the host system (some SDKs have getting security patches for eliminating it) but for getting access is very tricky then you must using the serial for knocking the host system.

I think one warning shall being OK so user can see if some one is trying getting in there system.

Adminiuga, 2022-09-07:
Application should consume the exception, not the library.

pipiche38, 2022-09-07:
I don't understand what you mean by application. Here the application doesn't receive anything the stack crash is inside zigpy library, no ?

puddly, 2022-10-08:
Fixed by packet API, since ZNP forwards all packets as-is to the application with `packet_received`.

closed 2022-10-08 by puddly