zigpy/zigpy-znp #172: gnupg issues

naive-HA, 2022-09-10:
I have installed zigpy-znp and started playing with some code. I have not got very far, but my previous scripts which make use of gnupg, now, fail to run throwing the error:
'GPG' object has no attribute 'trust_keys'
Previous scripts do not make use of zigpy-znp library
I have checked my code on a clean machine, without zigpy-znp and everything works just fine
I am using gnupg 0.4.6-1
What version of gnupg are you using with zigpy-znp?
Why the issue?

puddly, 2022-09-10:
> What version of gnupg are you using with zigpy-znp?

None, zigpy-znp has no `gnupg` dependency:

```
(venv) $ pip install zigpy-znp
...
(venv) $ pip freeze | grep gnupg  # no output
```

> Why the issue?

I assume you're referring to https://github.com/vsajip/python-gnupg. That package has no dependencies so even if zigpy-znp had some packaging bug, it wouldn't matter.

Please provide a self-contained script to replicate the problem.

naive-HA, 2022-09-10:
if no dependency then clearly I need to better understand how my setup brraks when installing the library
sorry about that

closed 2022-09-10 by naive-HA