zigpy/zigpy-znp #187: Unable to push OTA to an end device

pipiche38, 2022-10-13:
We are struggling to OTA upgrade end device.
Find here attached zigpy logs.
Basically the block request response is not going out , while on routers devices it works well

> 2022-10-13 12:26:15,116 DEBUG   :Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x0E01, Relays=[0x59F4])
> 2022-10-13 12:26:15,153 DEBUG   :Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x1150), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=25, TSN=4, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x19\x04\x05\x00\x7C\x11\xC1\x11\x31\x06\x08\x23\x00\x00\x00\x00\x3F\x1E\xF1\xEE\x0B\x00\x01\x38\x00\x00\x00\x7C\x11\xC1\x11\x31\x06\x08\x23\x02\x00\x45\x42\x4C\x20\x74\x72\x61\x64\x66\x72\x69\x5F\x63\x6F\x6E\x74\x72\x6F\x6C\x6C\x65\x72\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x2C\xF6\x02\x00\x00\x00\x00\xCA\x02\x00\x00')
> 2022-10-13 12:26:15,165 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1024, SrcAddr=0x0E01, SrcEndpoint=2, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=25, SecurityUse=<Bool.false: 0>, TimeStamp=4485423, TSN=0, Data=b'\x08\x6C\x0A\x00\x00\x21\x00\x00', MacSrcAddr=0x59F4, MsgResultRadius=28)
> 2022-10-13 12:26:15,188 DEBUG   :Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x07D9, Relays=[0x05CE])
> 2022-10-13 12:26:15,206 DEBUG   :Received command: AF.DataRequestExt.Rsp(Status=<Status.MEM_ERROR: 16>)

[zigpy-errors.zip](https://github.com/zigpy/zigpy-znp/files/9775736/zigpy-errors.zip)

puddly, 2022-10-13:
> `AF.DataRequestExt.Rsp(Status=<Status.MEM_ERROR: 16>)`

What device is this?

pipiche38, 2022-10-13:
The end device is an Ikea Remote 5 Buttons. I'm getting the same with a remote Philips HUE.
And I know that @deufo had similar issue with the same error.
At the end @deufo has succeed to upgrade.

puddly, 2022-10-13:
What coordinator?

pipiche38, 2022-10-14:
Coordinator is zzh-p with firmware 20220219
48 Devices / 20 ZR + 28 ZED

pour @deufo, I think he is on a zzh mostelikely with the same firmware

closed 2022-12-31 by pipiche38