zigpy/zigpy-znp #188: :ZDO device announce: nwk=0x0000, ieee=ff:ff:ff:ff:ff:ff:ff:ff, capabilities=0

pipiche38, 2022-10-21:
I have one of our user having a strange error at start time

in short receiving this message:
```

2022-10-21 19:10:30,037 DEBUG   :Received command: ZDO.MsgCbIncoming.Callback(Src=0x0000, IsBroadcast=<Bool.true: 1>, ClusterId=19, SecurityUse=0, TSN=0, MacDst=0xFFFF, Data=b'\x00\x00\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\x00')
2022-10-21 19:10:30,037 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1026, SrcAddr=0xAC78, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=69, SecurityUse=<Bool.false: 0>, TimeStamp=10873805, TSN=0, Data=b'\x18\xFD\x0A\x00\x00\x29\xC0\x07', MacSrcAddr=0x6EB0, MsgResultRadius=28)
2022-10-21 19:10:30,037 INFO    :ZDO device announce: nwk=0x0000, ieee=ff:ff:ff:ff:ff:ff:ff:ff, capabilities=0
2022-10-21 19:10:31,053 INFO    :Device 0x0000 (ff:ff:ff:ff:ff:ff:ff:ff) joined the network
2022-10-21 19:10:31,053 DEBUG   :Received frame on uninitialized device <Device model=None manuf=None nwk=0x0000 ieee=ff:ff:ff:ff:ff:ff:ff:ff is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\x00'
2022-10-21 19:10:31,053 DEBUG   :[0x0000:zdo] ZDO request ZDOCmd.Device_annce: [0x0000, ff:ff:ff:ff:ff:ff:ff:ff, 0]
2022-10-21 19:10:34,537 DEBUG   :Received command: ZDO.EndDeviceAnnceInd.Callback(Src=0x0000, NWK=0x0000, IEEE=ff:ff:ff:ff:ff:ff:ff:ff, Capabilities=<MACCapabilities.0: 0>)
2022-10-21 19:10:34,537 DEBUG   :Command was not handled
2022-10-21 19:10:34,537 DEBUG   :Received command: ZDO.MsgCbIncoming.Callback(Src=0x0000, IsBroadcast=<Bool.true: 1>, ClusterId=19, SecurityUse=0, TSN=0, MacDst=0xFFFF, Data=b'\x00\x00\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\x00')
2022-10-21 19:10:34,537 INFO    :ZDO device announce: nwk=0x0000, ieee=ff:ff:ff:ff:ff:ff:ff:ff, capabilities=0

```

Here is a full log from the start
[znp-error.log.zip](https://github.com/zigpy/zigpy-znp/files/9841349/znp-error.log.zip)

Any paths I should investigate ?

pipiche38, 2022-10-22:
The main question is : is that normal that Zigpy is sending to above layer such device annouce ?

If yes, I'll simply drop those invalid ieee

MattWestb, 2022-10-22:
The IEEEE is on reserved one that is used in the paring posses in the 802.15.4 layer but i cant remember for witch type or paring like distributed network but its 110% being used then touch link joining device to one network.

Is the device being normal retested for paring or its it forced doing one touch link paring ??

pipiche38, 2022-10-23:
I don't know these are errors brought by a user of the zigpy stack + the zigbee for domoticz plugin
So right now, I filtered out those reserved ieee and drop the message

closed 2022-10-23 by pipiche38