zigpy/zigpy-znp #191: zigpy-packet-api integration in z4d

pipiche38, 2022-10-24:
I'm starting to integrate the new zigpy-packet-api, and I would like to thank you for keeping the backward compatibility. This is really helpfull for us taking the integration path.

I had no issue on the bellows side, but facing an issue on ZNP

```
2022-10-24 22:07:15,343 DEBUG   :Waiting for NIB to stabilize
2022-10-24 22:07:15,345 DEBUG   :Sending request: SYS.OSALNVLength.Req(Id=<OsalNvIds.NIB: 33>)
2022-10-24 22:07:15,359 DEBUG   :Received command: SYS.OSALNVLength.Rsp(ItemLen=116)
2022-10-24 22:07:15,363 DEBUG   :Sending request: SYS.OSALNVReadExt.Req(Id=<OsalNvIds.NIB: 33>, Offset=0)
                                                                                                                                                                                                                     7969,1        79%
2022-10-24 22:07:15,544 DEBUG   :Received command: AF.Register.Rsp(Status=<Status.SUCCESS: 0>)
2022-10-24 22:07:15,549 DEBUG   :Sending request: AF.Register.Req(Endpoint=8, ProfileId=260, DeviceId=2080, DeviceVersion=0, LatencyReq=<LatencyReq.NoLatencyReqs: 0>, InputClusters=[0], OutputClusters=[])
2022-10-24 22:07:15,560 DEBUG   :Received command: AF.Register.Rsp(Status=<Status.SUCCESS: 0>)
2022-10-24 22:07:15,565 DEBUG   :Sending request: AF.Register.Req(Endpoint=3, ProfileId=260, DeviceId=2080, DeviceVersion=0, LatencyReq=<LatencyReq.NoLatencyReqs: 0>, InputClusters=[0], OutputClusters=[])
2022-10-24 22:07:15,575 DEBUG   :Received command: AF.Register.Rsp(Status=<Status.SUCCESS: 0>)
2022-10-24 22:07:15,579 DEBUG   :Sending request: ZDO.MsgCallbackRegister.Req(ClusterId=65535)
2022-10-24 22:07:15,589 DEBUG   :Received command: ZDO.MsgCallbackRegister.Rsp(Status=<Status.SUCCESS: 0>)
2022-10-24 22:07:15,593 DEBUG   :[0x0000] Scheduling initialization
2022-10-24 22:07:15,596 DEBUG   :Tries remaining: 3
2022-10-24 22:07:15,598 INFO    :[0x0000] Requesting 'Node Descriptor'
2022-10-24 22:07:15,599 DEBUG   :Tries remaining: 2
2022-10-24 22:07:15,601 DEBUG   :[0x0000] Extending timeout for 0x01 request
2022-10-24 22:07:15,603 DEBUG   :Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=True, tsn=1, profile_id=0, cluster_id=<ZDOCmd.Node_Desc_req: 0x0002>, data=Serialized[b'\x01\x00\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-10-24 22:07:15,610 DEBUG   :Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=1, profile_id=0, cluster_id=<ZDOCmd.Node_Desc_req: 0x0002>, data=Serialized[b'\x01\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-10-24 22:07:15,613 WARNING :Device <ZNPCoordinator model='CC1352/CC2652, Z-Stack 3.30+ (build 20220928)' manuf='Texas Instruments' nwk=0x0000 ieee=00:12:4b:00:23:8d:ab:93 is_initialized=False> failed to initialize due to unexpected error
Traceback (most recent call last):
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/device.py", line 177, in initialize
    await self._initialize()
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/util.py", line 131, in retry
    r = await func()
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/device.py", line 202, in _initialize
    await self.get_node_descriptor()
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/device.py", line 161, in get_node_descriptor
    status, _, node_desc = await self.zdo.Node_Desc_req(
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/util.py", line 131, in retry
    r = await func()
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/device.py", line 297, in request
    await self._application.request(
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/application.py", line 654, in request
    await self.send_packet(
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/zigbee/application.py", line 1026, in send_packet
    response = await self._send_request_raw(
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/zigbee/application.py", line 879, in _send_request_raw
    self.packet_received(
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/application.py", line 839, in packet_received
    self.handle_message(
TypeError: handle_message() got an unexpected keyword argument 'dst_addressing'
```

pipiche38, 2022-10-24:
Ah we have to update the handle_message() signature that we overwrite

puddly, 2022-10-24:
Are you overloading `handle_message` in your own subclasses? This has been there for a while: https://github.com/zigbeefordomoticz/zigpy/blob/0ed5b0ed521ef3eab6df42ec10c0ce488b4c33e5/zigpy/application.py#L335

puddly, 2022-10-24:
`expect_reply, use_ieee,extended_timeout=False` should be `expect_reply=expect_reply, use_ieee=use_ieee, extended_timeout=False`.

closed 2022-10-24 by pipiche38

pipiche38, 2022-10-24:
BTW extended_timeout=False is that the previous behavior, or is that more extended_timeout=True which is the behaviour from the library before the packet-api ?

puddly, 2022-10-24:
`extended_timeout` is set to `True` by the device object when you are sending a request to an end device. It increases the interval between retries on EmberZNet. This behavior is still retained, it's just exposed via the packet API if you don't use the `Device` object.